PhracturedBlue/ESP8266MQTTMesh #1: TLS support on MQTT

lewisling, 2017-07-09:
Hi @PhracturedBlue, thanks for your contribution on MQTT over Mesh for ESP. Do you have plan to have TLS support on this project? Thanks again. Lewis

PhracturedBlue, 2017-07-09:
I didn't have plans to do so, but it looked easy, so I have started to implement it both on the mesh as well as to the MQTT broker.  The initial coding is done, but testing is needed as well as documentation on how to use it.

It will require the development branch of the esp8266 library to use though.

lewisling, 2017-07-11:
@PhracturedBlue Great to know, and looking if there is a branch out soon. Thanks again for your effort. Lewis

PhracturedBlue, 2017-07-11:
I have tested the SSL between mesh nodes, and it works as expected now.  I have not tested an SSL connection to the MQTT broker since mine isn't running with SSL enabled, but it should work.

I have documented how to use it in the HelloWorld example

lewisling, 2017-07-11:
@PhracturedBlue  Great News, I have a SSL enabled MQTT broker, and I am very happy to try it out and get back to you. Again Great contribution! Lewis

PhracturedBlue, 2017-07-11:
Try it without a fingerprint 1st.  I gave it a shot, and got a stack-trace.  If you get the same, try building this:
https://github.com/marvinroger/async-mqtt-client/tree/master/examples/FullyFeaturedSSL
(make sure to fix the variables in main.cpp)

I got the same stack trace using that code, so it looks like some issue in the MQTT library (or elsewhere).
I've submitted a bug there, though I'm not sure it is the right place:

https://github.com/marvinroger/async-mqtt-client/issues/55

It may be a configuration issue on my mqtt broker though, so I'll be interested in your result

PhracturedBlue, 2017-07-11:
And....I found the issue.  It looks like certs generated with SHA512 are problematic.
I had used this script to generate my mosquitto cert:
https://github.com/owntracks/tools/blob/master/TLS/generate-CA.sh

and that caused an issue.

When I used openssl's defaults, everything worked, so you should be good to go.

One note, I got a stacktrace when I supplied the wrong fingerprint.  I haven't debugged that yet.  It works fine with no fingerprint, or with the correct fingerprint though.

PhracturedBlue, 2017-07-11:
Hopefully all issues are now resolved.  Closing again.

lewisling, 2017-07-14:
@PhracturedBlue, a bit busy recently, let's summarize my test result: 
#1 Connection with mqtt broker works ok with both TLS and Fingerprint check;
#2 The publish works occasionally, which i still trying to figuring out last night where goes wrong, the use of AsyncMqttClient or some other reason??? As i normally use Adafruit_MQTT_Client for MQTT operation, and has test out with AsyncMqttClient's FullFeaturedSSL example works ok;
#3 I have 10+ NodeMCU with me for this test, but not yet seeing expected communication (send_message) between Node via Mesh, ESP8266_NUM_CLIENTS = 4 remains for my initial 4x Nodes test;

Some questions on top of that:
Q1: Do you see there what will be the max Node can be connected in this mesh under the available NodeMCU's RAM (128K)?
Q2: Trying to understand your logic, seem it will periodically off (actually switch) for MQTT connect via Router connected WIFI and Mesh's AP, is that a must because ESP8266 not ok for have both Mesh AP and Router connection simultaneously?

Thanks again.

PhracturedBlue, 2017-07-14:
You should not change the maximum number of nodes.  This is the # of nodes that can connect to any given node.  The ESP8266 has a hard limit on this #.  If you have more nodes, they will chain themselves together so each node has no more than 4 other connections.  this should happen automatically.

There is a practical limit of ~250 total nodes supported by the mesh today.

There should be no disconnection from the MQTT broker.  nodes can talk to other nodes and to the broker at the same time.  If you are seeing frequent reconnects, we will need to investigate that.  I have not spent much time working with the SSL library (using SSL support uses the development branch of the esp8266 library which is quite different).  I would recommend testing this client 1st and see if it is reliable:
https://github.com/marvinroger/async-mqtt-client/tree/master/examples/FullyFeaturedSSL
If that works well, the problem is in the mesh code somewhere.  if it doesn't work well, the problem is elsewhere.

lewisling, 2017-07-14:
Ok i will keep the ESP8266_NUM_CLIENTS remain 4.
I have leave the FullFeaturedSSL test over few hours for continue the publish, so far no missing any and each publish well received from my another Client. I am still debugging where make the MQTT reconnect when doing "Scheduling reconnect for 5.00 seconds from now".

PhracturedBlue, 2017-07-14:
can you please provide your serial log showing the connect/disconnect issue?

lewisling, 2017-07-14:
Here is the simplified log, i remark some of the print out to easier track the logic:

[begin] Starting Firmware 1337 : 0.1
[begin] OTA Start: 0x5e000 OTA End: 0x100000
[begin]  ==> '/bssid/5E:CF:7F:B1:90:52'
[begin] 0
[begin] Setup Complete
Setup Completed
[connect] Connecting to SSID : 'MyWiFiSSID' BSSID '70:56:81:CA:4D:DD'
[onWifiConnect] Connecting to mqtt
[connect_mqtt] Attempting MQTT connection...
[onMqttConnect] MQTT Connected
[setup_AP] Initialized AP as 'mesh_esp8266-4'  IP '192.168.4.1'
[onMqttDisconnect] Disconnected from MQTT.
[onMqttDisconnect] Bad MQTT server TCP DISCONNECTED.
[onWifiDisconnect] Disconnected from Wi-Fi: MyWiFiSSID because: 200
[schedule_connect] Scheduling reconnect for 5.00 seconds from now
[connect] Connecting to SSID : 'MyWiFiSSID' BSSID '70:56:81:CA:4D:DD'
[onWifiConnect] Connecting to mqtt
[connect_mqtt] Attempting MQTT connection...
[onMqttConnect] MQTT Connected
[setup_AP] Initialized AP as 'mesh_esp8266-4'  IP '192.168.4.1'
[onMqttDisconnect] Disconnected from MQTT.
[onMqttDisconnect] Bad MQTT server TCP DISCONNECTED.
[onWifiDisconnect] Disconnected from Wi-Fi: MyWiFiSSID because: 200
[schedule_connect] Scheduling reconnect for 5.00 seconds from now
[connect] Connecting to SSID : 'MyWiFiSSID' BSSID '70:56:81:CA:4D:DD'
[onWifiConnect] Connecting to mqtt
[connect_mqtt] Attempting MQTT connection...
[onMqttConnect] MQTT Connected
[setup_AP] Initialized AP as 'mesh_esp8266-4'  IP '192.168.4.1'
[onMqttDisconnect] Disconnected from MQTT.
[onMqttDisconnect] Bad MQTT server TCP DISCONNECTED.
[onWifiDisconnect] Disconnected from Wi-Fi: MyWiFiSSID because: 200
[schedule_connect] Scheduling reconnect for 5.00 seconds from now
[connect] Connecting to SSID : 'MyWiFiSSID' BSSID '70:56:81:CA:4D:DD'
[onWifiConnect] Connecting to mqtt
[connect_mqtt] Attempting MQTT connection...
[onMqttConnect] MQTT Connected
[setup_AP] Initialized AP as 'mesh_esp8266-4'  IP '192.168.4.1'
[onMqttDisconnect] Disconnected from MQTT.
[onMqttDisconnect] Bad MQTT server TCP DISCONNECTED.
[onWifiDisconnect] Disconnected from Wi-Fi: MyWiFiSSID because: 200
[schedule_connect] Scheduling reconnect for 5.00 seconds from now

PhracturedBlue, 2017-07-14:
2 things to try:
1)  add a `return();` at the top of void ESP8266MQTTMesh::setup_AP() {} (i.e. disable the AP mode).  This obvioulsy disables the mesh, but will ocnfirm that the transition from STA to AP mode is causing the issue
2) what channel is your WiFi AP running on?  Is it possible to try changing it to channel '1' and see if it changes anything?

We have found that AP/STA mode is not very robust on the ESP8266, and have had a lot of difficulty to make this robust.

lewisling, 2017-07-15:
@PhracturedBlue, thanks for the suggestions, tried with #1 add "return()"seems working, making the connection to broker stable. Publish, Subscribe and Keepalive work as expected.

My timecapsule remains using Channel 11.

Any suggestion or solution on switching STA/AP modes for my case, or i have done sth different make the result different with you? Thanks.

PhracturedBlue, 2017-07-15:
I have tested with my WiFi AP set to use channel 11, and my connection is stable.
Are you using platformio to build or are you building via Arduino?
In setup_AP() try to comment out `WiFi.mode(WIFI_AP_STA);`  and in begin() change `WiFi.mode(WIFI_STA);` to `WiFi.mode(WIFI_AP_STA);`

This change will prevent switching AP wifi modes.  In the past we had good luck with the MQTT connection with these settings, but they didn't work as well for the node connections.  But let's see how it goes.

Also, what is the local ip address of your Wifi router?  is it 192.168.1.xx?  If you use 192.168.4.xx for your wifi, that could be an issue.

lewisling, 2017-07-15:
@PhracturedBlue 
#1 I am using Arduino IDE, as I have some problem on using platformio;
#2 Has tried to modify setup_AP() and begin() following your instruction (the previous tested return() has been removed), the result same as before that repeat MQTT reconnection because of "TCP DISCONNECTED", and the log is same as previously i sent you;
#3 My WIFI router use 10.0.1.xx. the "4" in 192.168.4.xx, seems came from your program's subdomain, fetch from SPIFFS under /bssid/;

Lewis

lewisling, 2017-07-15:
@PhracturedBlue 

I found "4" is written from first assign_subdomain execution, where it start from i =4, and result "4" is written into SPIFFS, and later retrieved for the subdomain 192.168."4".xx, so are you expecting me to change the for loop from i=1?

    for (int i = 4; i < 256; i++) {
        if (! seen[i]) {
            File f = SPIFFS.open("/bssid/" +  WiFi.softAPmacAddress(), "w");
            if (! f) {
                dbgPrintln(EMMDBG_MSG, "Couldn't write "  + WiFi.softAPmacAddress());
                die();
            }
            f.print(i);
            f.print("\n");
            f.close();
            //Yes this is meant to be inTopic.  That allows all other nodes to see this message
            char topic[TOPIC_LEN];
            char msg[4];
            itoa(i, msg, 10);
            strlcpy(topic, inTopic, sizeof(topic));
            strlcat(topic, "bssid/", sizeof(topic));
            strlcat(topic, WiFi.softAPmacAddress().c_str(), sizeof(topic));
            dbgPrintln(EMMDBG_MQTT_EXTRA, "Publishing " + String(topic) + " == " + String(i));
            mqttClient.publish(topic, 0, true, msg);
            dbgPrintln(EMMDBG_MQTT_EXTRA, "assign_subdomain>setup_AP");
            setup_AP();
            return;
        }
    }

PhracturedBlue, 2017-07-15:
yes the 192.168.4.1 is the local gateway (the AP generated on the ESP8266).  That is expected.  The issue would be if your wfi network was also 192.168.4.1 since the ESP8266 AP can't have the same network id as your router.

Have you confirmed that you have the correct versions of the ESP8266 core (you need the development branch), ESPAsyncTCP and AsyncMQTTClient?  As I said, the code to support WIFI_AP_STA mode seems not very reliable, and I have seen differences with different versions of the core.  Since you have SSL working, I assume this is all Ok already, but you should double check everything.

I am honestly running out of ideas.  Can you check your broker and see if there are any errors there?  If you use mosquitto, maybe start it with verbose logging.

PreferHardware, 2017-07-15:
I too have regular WiFi disconnects.  Hello world will typically publish one or two mqtt messages before the WiFi resets.

Heaven knows where these messages go, though,  because the broker certainly isn't seeing them!

PreferHardware, 2017-07-16:
Two problems:
1. Unstable wifi connection
2. Serial says a message has been published, but mqtt broker doesn't get it

Both of these appear to be resolved by commenting out all WiFi.mode() calls (4 or 5?) in ESP8266MQTTMesh.cpp and having a single WiFi.mode(WIFI_AP_STA) in the setup() of the user app before the mesh.begin() call.  The alternative would be to put this in mesh.begin() itself, but I refer the better visibility that this gives.

I'm not certain, but somewhere in the back of my mind a bell is ringing that says that the WiFi.mode() is actually stored in flash (ie it's non-volatile), so it's probably not a good idea to be changing it willy-nilly anyway.  

I hope this is helpful.

PhracturedBlue, 2017-07-16:
Problem 2 is related to problem 1.  The client sends messages to the broker but never gets an acknowledgement.  This triggers a disconnect/reconnect cycle.  When mqtt is disconnected, we disconnect the wifi too and start the cycle over.

Have you tested the node connections as well?

I started with a single Wifi.mode call, but it was unreliable using a 2.3.0 code.  the development branch works differently (and for me at least, the current code works for both).  I can detect 2.3.0 vs 2.4.0 and have different behavior for each though.  Before doing that, I'd like to know that everything is working reliably for you.

PreferHardware, 2017-07-16:
I'm using the 2.3.0 core in the Arduino IDE and, initially anyway, a little ESP-01 with 1M memory.

Node connections are my next step on this journey.  I'll let you know how I get on.

PhracturedBlue, 2017-07-16:
SSL shouldn't even work with the 2.3.0 Arduino core.  I am quite surprised that you can get anything to work with it.  If you are using SSL, you really should use the development core.

PreferHardware, 2017-07-16:
I'm not using SSL at the moment, but am well aware that I may need to move to 2.4.0 if I do.  I have other reasons for wanting to stick to the stable core if I can and may even hold off on this until 2.4.0 is released if I hit a roadblock as a result.

PhracturedBlue, 2017-07-16:
I didn't realize you had jumped into lewisling's bug.
Can you please file a different issue to discuss non-TLS related problems?

closed 2017-11-28 by PhracturedBlue