PhracturedBlue/ESP8266MQTTMesh #2: WiFi ok, but MQTT fails + solution

PreferHardware, 2017-07-15:
New to this library, I installed the prerequisites and tried the hello world example.  It connected to WiFi, but failed to connect to the MQTT server, spraying the serial port with connecting...disconnected messages.

To cut a long story short, the solution was to add a line after line 72 of ESP8266MQTTMesh.cpp:

mqtt_port(mqtt_port),

PhracturedBlue, 2017-07-15:
Thanks, I've committed the needed fix.

closed 2017-07-15 by PhracturedBlue

PreferHardware, 2017-07-16:
I see you also added server and port to the dbgPrintln in mqtt_connect - this was exactly the step I took to debug this!

PhracturedBlue, 2017-07-16:
Yeah, I found this issue on Friday when I actually got a chance to test the new Builder() system. I just forgot to release it.