PhracturedBlue/ESP8266MQTTMesh #4: Tips on OTA Implementation - Improvement request

ibaranov-cp, 2017-07-29:
Hello,

I'm looking to put together an example for OTA upload over the mesh, and would happily submit a pull request with a tutorial on setting up the network and doing updates.

Would you mind sharing what your thoughts are for architecture for pushing OTA updates over the mesh network? It's mentioned in the Readme, but never really shown in the code (that I can see anyway ;)

Thanks!

PhracturedBlue, 2017-07-29:
use the utils/send_ota.py script.
you feed it the broker, node id, and a bin file and it will send the ota update.

ibaranov-cp, 2017-07-29:
Edit: Yup, I'm just blind 

Will write up a short tutorial shortly :)

ibaranov-cp, 2017-07-29:
Sending
```
python send_ota.py --bin firmware.bin --id 0x1337
```
seems to reset everything in the network.
However, after transmitting all the bytes, it caused a wdt reset on one chip, and no action on another chip.

Questions:
1) There doesn't seem to be a current way to update only a specific node. Is this a good idea for me to add?
2) Has the OTA been tested? Seems almost working currently, unless I'm missing something.

PhracturedBlue, 2017-07-29:
I haven't tested OTA recently.  It is quite possible I broke it.  I'll give it a try again soonish.

Firmware is assigned to specific firmware IDs.  all nodes with the same firmware ID will take the firmware.  So if you provide each node with its own firmware ID it could be point addressable.  My thought process was that you might have a group of temperature nodes with a specific id and a group of switch nodes with a different ID.  you would normally want to update all the temperature nodes at the same time.
We could also make it be node addressable reasonably easily.

PhracturedBlue, 2017-08-05:
I have released 0.8.4 which fixes the OTA code.  I hadn't tested it since moving to the Async implementation.

ibaranov-cp, 2017-08-07:
Hello,

I've updated to 0.8.4, however code uploading is still failing. Sometimes a wdt reset occurs on the ESP (seen through serial logging), and the device resets and recovers, but the OTA script just keeps sending bytes blindly.

PhracturedBlue, 2017-08-07:
you mean you get a reset before the data transfer is complete?  I expect you to get a stack-trace on reset after the firmware transfer is complete.
You'll need to paste the logs from your serial console for me to debug further.
If you get a stack-trace that is not related to the intentional reset after the MD5 is checked, you'll need to send me the stack-dump and your .pioenvs/nossl/firmware.elf file so that I can debug the stack-trace (I am also assuming you are debugging using the ESP8266MQTTMeshHelloWorld example.  If not, I probably need to see the code you are working on too.

vixtory09678, 2017-09-26:
Hello.
I'm trying to send OTA data, but I'm having md5 mismatch problem.
Do you have a solution?

Thanks in advance

PhracturedBlue, 2017-09-26:
Can you be more specific?  You are using the  utils/send_ota.py script?
you can edit the time.sleep() call from 'time.sleep(0.2) to time.sleep(0.5).  I have never had an issue with the current repeat time, but slowing it down may improve performance.

vixtory09678, 2017-09-27:
Hello
I tried to modify code from time.sleep (0.2) into time.sleep (0.5) in send_ota.py script
But it still does not work.

Now I use cloudmqtt broker, but I'm not sure if this broker has limited time to send the data.

vixtory09678, 2017-09-27:
Now I change the broker to my own broker and it works. 
Thank you for answer @PhracturedBlue

pathakhcst, 2017-10-11:
Hi, Is the send_ota.py is tested? I am trying to update the firmware using MQTT broker but no success.

PhracturedBlue, 2017-10-11:
that is normal.  You should check the node to see if it accepts the new firmware and reboots.  If you flash from USB and don't reboot the node afterwards, the node will not reboot properly after accepting the firmware

pathakhcst, 2017-10-11:
Means I should update the firmware from USB and reboot it. After reboot I should try the OTA using MQTT?

PhracturedBlue, 2017-10-11:
Yes.  Just the 1st time.  Anytime you install the firmware from USB, you should power-cycle the node before using OTA.

You should connect a serial port to the node to mionitor its status during OTA.

If the install AND reboot is successful, the send_ota.py script should report some status once the update is complete.

pathakhcst, 2017-10-12:
I uploaded the firmware using USB and power-off and reconnected. I have 3 ESP8266 connected to different computers with Serial Monitor. From MQTT broker I am trying to do the OTA. I fired the command
**python send_ota.py --bin firmware.bin --id 0x1337**
its shows the "_Transmitted xxxxx bytes_."
"_Completed send_"
"_Connected with result code 0_"

and it stopped. No status confirmation. None of the ESP shown any activity on serial and No reboot.
Only one ESP is connected to Wi-Fi router and other two are connected using MESH.

PhracturedBlue, 2017-10-12:
What did you set the 'intopic' to?  The send_ota.py code assumes it is set to 'esp8266-in' which is a bug.  Change the code in send-ota.py to send to the right intopic and you should see the nodes receiving the data.  I'll add a switch in the python to allow ou to change this.

pathakhcst, 2017-10-13:
When I matched the IN_TOPIC of ESP8266 and Send_Ota.py, the firmware upload using OTA is successful. Thanks PhracturedBlue.

pathakhcst, 2017-10-16:
I tried it with 6 ESP modules and couple of them got MD5 mismatch error. It was visible through serial but when they are remote we can't confirm which one is success and which is not. Because there is no status confirmation displayed at send_ota.py terminal.

crisap94, 2017-10-17:
I have been following this library since it came out i think, and there is something i don't understand quite well, i know the idea to use the id for every type of module but why @pathakhcst  had to match the topic?, is that a good solution?.

It is worth highlighting that i have not used the library before, but i plan to do it. And i have been working with many boards since 2 years ago.

By the way, really nice job here doing this @PhracturedBlue , i hope to contribute to this as soon i start the development of the project i will be part of.

PhracturedBlue, 2017-10-17:
@pathakhcst, It should be possible to improve OTA with a ping-back about bad data.  It would require a more complicated OTA protocol, and I'll need to think about it a bit.  At a minimum having each board send a pass/fail response would be an easy stop-gap which I will try to do shortly.

PhracturedBlue, 2017-10-17:
@crisap94 There is a default topic, but unfortunately the HelloWorld example doesn't use it (the other 2 examples do though).  I will probably change the HelloWorld to use the default topic as well.

pathakhcst, 2017-10-25:
When I try to do OTA to ESP which is directly connected to router its successful. But to those nodes which are indirectly connected(using Mesh network) to broker and not able to receive the firmware.bin. Even though they don't receive any publish command from broker also.

PhracturedBlue, 2017-12-04:
the current code supports direct addressing of an individual node by sending messages like`esp8266-in/ota/mesh_esp8266-4/flash`.  the update_ota script has been modified to support this via the --node option.

PhracturedBlue, 2018-03-04:
The OTA code no longer relies on sleep to ensure all nodes receive the firmware.  Instead each node responds back after each message and the send_ota script waits for all responses

closed 2018-03-04 by PhracturedBlue