PhracturedBlue/ESP8266MQTTMesh #6: Mesh connection fails

Alasknnj, 2017-07-29:
### Library versions:
- ESP8266MQTTMesh version: 0.8.1
- AsyncMQTTClient version: 0.8.1
- ESPAsyncTCP version: 1.0.1
- ESP8266 Core version: 2.4.0
- Platformio version : 3.4.0

**Description**
I've been unable to connect a node to another, through the hidden mesh network. 

In my setup I have two nodes, one is connected to the router, and the other doesn't have the router on the network list, so I could try the mesh connection. They have the HelloWorld example loaded.

The node that is not connect to the router scans for networks, tries to connect to the mesh network, but it fails with reason 201 (NO_AP_FOUND). 

Is the procedure for connecting to hidden networks the same as not hidden ones, don't they need the bssid? Also, hows does the library uses the files at /bssid/? Are they records of hidden networks that could be mesh networks? Are they only written after receiving a publish at IN_TOPIC/bssid?

PhracturedBlue, 2017-07-29:
Each node must connect directly to the mqtt broker at least once.  This registers the node with the broker and assigns it a unique ID.  the broker will broadcast all known nodes on connection, and this will be stored in the filesystem in the /bssid/ dir.  When a node sees a hidden network, it looks up the bssid to see if it has it in the /bssid directory.  If so, it will try to connect to it.
Every time a node successfully connects to either the mqtt broker or another mesh relay, it will receive an updated list of nodes, and each time a node is added to the network, its id is broadcast to all other nodes.  Thus, every node maintains a complete list of other nodes that it could theoretically connect to.

dlangamer, 2017-08-03:
By the way, is there any way for a node to receive information from the mesh without having established a first contact with the broker? Eg.: A new node that connects to the mesh network and receives the mqtt information through the other node.

In the automation system that i´m building, i would like the new node to simply get in the mesh network (which contains a node directly connected to the broker/internet) and receive the information from mqtt.

PhracturedBlue, 2017-08-03:
For a node to find other nodes, it needs to know the mesh parameters.  You could disable that such that it tries to sign into every hidden network it can see.  I would consider that insecure though.  Alternatively, you can pre-load the set of available nodes via the filesystem and then there won't be a need to connect to the broker 1st.  I don't have instructions for that, but I think you would just build the bssid/ directory with all node info in it, and then upload that to the node after the firmware.

pathakhcst, 2017-10-12:
How can we pre-load the set of available nodes via file system and what data is needed to provide like MAC_ID? We can provide the MAC ID list to each node, where should I put this list?

I am preparing the mesh network and facing difficulty to hookup with Broker on first attempt. The nodes takes more than 15 min time to connect to Mesh network and shows error 201. though it knows the name of hidden network like "mesh_esp8266-6" but doesn't connect successfully. Thanks!

PhracturedBlue, 2017-12-04:
This is now documented in docs/Filesystem.md

closed 2017-12-04 by PhracturedBlue