PhracturedBlue/ESP8266MQTTMesh #7: Compilation error

dlangamer, 2017-08-02:
Hello @PhracturedBlue,

I really liked the concept of your project and i have something similar here too. I´m using the Arduino IDE and when compile some reference errors occur (attach..). Should use PlatformIO in this case?

> Archiving built core (caching) in: C:\Users\Hero\AppData\Local\Temp\arduino_cache_530916\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_115200,FlashSize_4M3M_ff614313593e10d027a6bffb631e5a69.a
> Linking everything together...
> "C:\Users\Hero\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -w -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-LC:\Users\Hero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lib" "-LC:\Users\Hero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/ld" "-Teagle.flash.4m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130/sketch_jul26a.ino.elf" -Wl,--start-group "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\sketch\sketch_jul26a.ino.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESP8266MQTTMesh\Base64.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESP8266MQTTMesh\ESP8266MQTTMesh.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESP8266WiFi\ESP8266WiFi.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESP8266WiFi\ESP8266WiFiAP.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESP8266WiFi\ESP8266WiFiMulti.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESP8266WiFi\ESP8266WiFiSTA.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESP8266WiFi\ESP8266WiFiScan.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESP8266WiFi\WiFiClient.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESP8266WiFi\WiFiClientSecure.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESP8266WiFi\WiFiServer.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESP8266WiFi\WiFiUdp.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESPAsyncTCP\tcp_axtls.c.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESPAsyncTCP\AsyncPrinter.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESPAsyncTCP\ESPAsyncTCP.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESPAsyncTCP\ESPAsyncTCPbuffer.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\ESPAsyncTCP\SyncClient.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\Ticker\Ticker.cpp.o" "C:\Users\Hero\AppData\Local\Temp\arduino_build_4130/arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group  "-LC:\Users\Hero\AppData\Local\Temp\arduino_build_4130"
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x54): undefined reference to `AsyncMqttClientInternals::ConnAckPacket::ConnAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (bool, unsigned char)>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x58): undefined reference to `AsyncMqttClientInternals::PingRespPacket::PingRespPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void ()>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x5c): undefined reference to `AsyncMqttClientInternals::SubAckPacket::SubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short, char)>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x60): undefined reference to `AsyncMqttClientInternals::UnsubAckPacket::UnsubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x64): undefined reference to `AsyncMqttClientInternals::PublishPacket::PublishPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short)>, std::function<void (unsigned short, unsigned char)>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x68): undefined reference to `AsyncMqttClientInternals::PubRelPacket::PubRelPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x6c): undefined reference to `AsyncMqttClientInternals::PubAckPacket::PubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x70): undefined reference to `AsyncMqttClientInternals::PubRecPacket::PubRecPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x74): undefined reference to `AsyncMqttClientInternals::PubCompPacket::PubCompPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o: In function `AsyncMqttClient::_onData(AsyncClient*, char*, unsigned int)':
> 
> C:\Users\Hero\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:92: undefined reference to `AsyncMqttClientInternals::ConnAckPacket::ConnAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (bool, unsigned char)>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o: In function `std::_Function_base::_Base_manager<std::_Bind<std::_Mem_fn<void (AsyncMqttClient::*)(bool, unsigned char)> (AsyncMqttClient*, std::_Placeholder<1>, std::_Placeholder<2>)> >::_M_init_functor(std::_Any_data&, std::_Bind<std::_Mem_fn<void (AsyncMqttClient::*)(bool, unsigned char)> (AsyncMqttClient*, std::_Placeholder<1>, std::_Placeholder<2>)>&&, std::integral_constant<bool, false>)':
> 
> C:\Users\Hero\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:92: undefined reference to `AsyncMqttClientInternals::PingRespPacket::PingRespPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void ()>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o: In function `std::_Function_base::_Base_manager<std::_Bind<std::_Mem_fn<void (AsyncMqttClient::*)()> (AsyncMqttClient*)> >::_M_init_functor(std::_Any_data&, std::_Bind<std::_Mem_fn<void (AsyncMqttClient::*)()> (AsyncMqttClient*)>&&, std::integral_constant<bool, false>)':
> 
> C:\Users\Hero\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:92: undefined reference to `AsyncMqttClientInternals::SubAckPacket::SubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short, char)>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o: In function `AsyncMqttClient::_onData(AsyncClient*, char*, unsigned int)':
> 
[... cut ...] unction<void (unsigned short)>)'
> 
> C:\Users\Hero\AppData\Local\Temp\arduino_build_4130\libraries\AsyncMqttClient\AsyncMqttClient.cpp.o: In function `_Head_base<AsyncMqttClient* const, void>':
> 
> C:\Users\Hero\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:92: undefined reference to `AsyncMqttClientInternals::PubCompPacket::PubCompPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
> 
> collect2.exe: error: ld returned 1 exit status
>

PhracturedBlue, 2017-08-02:
All of these issues seem to be related to the AsyncMqttClient library.  I't looks like the library is not being compiled properly since the internal code isn't being found.  Try compiling the 'FullyFeatured' example in the AsyncMqttClient library and make syre that is ok 1st.

If you do use platformIO, it should compile fine since the dependencies are handled automatically.  But I have been able to compile with Arduino as well as long as the libraries are properly installed.

dlangamer, 2017-08-02:
In the PlatformIO worked nice! But i noticed same problem with Router Channel (like another issue posted here - constant disconnects from wifi if router is not in channel 1). Anyway works like a charm.

Thanks for help!

Best regards!

PhracturedBlue, 2017-08-02:
As mentioned, you can work around the channel issue by using the 'staging' version of the esp8266 core.  Change the platformio.ini file such that:
`platform = espressif8266_stage`
for 'env:nossl'

dlangamer, 2017-08-02:
"Staging platform" worked well. No disconnections... Thx 👍 
Another question: there a way to connecting to an external mqtt broker (hostname or external ip)?

PhracturedBlue, 2017-08-02:
There shouldn't be any difference between an external broker and a local one.
Assuming you are using the HelloWorld example, you just need to set the MQTT_SERVER and MQTT_PORT variables appropriately (which you must have already done to get anything working at all).  So I don't think I properly understand your question.  Are you wondering about domain-name lookup?  I would expect that to 'just work' but I've never tested it myself.

dlangamer, 2017-08-03:
That'a it! I have been tried connect to an external mqtt server with no success (ip or fqdn outside my network). the reported error below: 

[connect_mqtt] Attempting MQTT connection...
[onMqttDisconnect] Disconnected from MQTT.

If you want to check, I have this mqtt broker for testing purposes: vps3952.publiccloud.com.br:1883 (anonymous password allowed)

dlangamer, 2017-08-03:
Ignore last post... With Stage version work well. Thanks for all help

closed 2017-08-03 by dlangamer