PhracturedBlue/ESP8266MQTTMesh #8: Cannot get the HelloWorld example to connect to my mosquitto MQTT server

flemmingdjensen, 2017-08-02:
Make sure you have the latest version of ESP8266MQTTMesh before reporting an issue.

### Please Provide the following information:
- ESP8266MQTTMesh version: 0.8.1
- AsyncMQTTClient version: 0.8.1
- ESPAsyncTCP version: 1.0.1
- ESP8266 Core version: 1.3.1 and Stage tested

**Are you using platformio or Arduino?**
platformio

**Description of problem:**

I am trying the ESP8266MeshHelloWorld example and after uploading to a NodeMCU 1.0 board I get WifiConnection successfull but then when connecting to my mosquitto server on 1883 i see the following loop:

[connect_mqtt] Attempting MQTT connection...
[onMqttDisconnect] Disconnected from MQTT.
[connect_mqtt] Attempting MQTT connection...
[onMqttDisconnect] Disconnected from MQTT.
[connect_mqtt] Attempting MQTT connection...
[onMqttDisconnect] Disconnected from MQTT.
[connect_mqtt] Attempting MQTT connection...
[onMqttDisconnect] Disconnected from MQTT.

I have tried the AsyncMQTTExample and it works just fine connecting to my mosquitto server

I have tried both ESP Core 1.3.1 and Stage versions

Any ideas what I am doing wrong?

flemmingdjensen, 2017-08-02:
Found the problem to be that when you install the ESP8266MQTTMesh library in platformio you get version 0.8.1
where the call to mqtt_port(mqtt_port), is not called...

Can we get the platformio version updated?

PhracturedBlue, 2017-08-02:
I didn't realize it hadn't built properly.  I have updated the library to 0.8.3.  Hopefully it will show up on platformio within the next 24 hours

flemmingdjensen, 2017-08-04:
Hi, Platformio is still fetching version 0.8.1 :(
Is there really not any way to override that? How is this registry updated?

Another question... If I was to use this library in one of my projects and potentially made a product out of it that could sell... Do I then have to provide source code of my project if anybody requests this? Because of GPL V3...

BR

PhracturedBlue, 2017-08-04:
There is no way for me to force a refresh.  I requested help in the ir community page.  We'll see what happens.

As far as licensing, I'm afraid so.  Even if my library were MIT licensed, it uses the ESPAsyncTCP library which is under the LGPL.  The LGPL (from the FAQ):
(1) If you statically link against an LGPL'd library, you must also provide your application in an object (not necessarily source) format, so that a user has the opportunity to modify the library and relink the application.

So unless you are going to provide a mechanism for the user to relink an upload your library, the LGPL doesn't really help you.  As it is, that is irrelevant, since the ESP8266MQTTMesh library is actually GPLv3, you are not able to use it in commercially released products without also releasing the source code for that project.  Sorry.

PhracturedBlue, 2017-08-04:
They fixed the release.  0.8.3 is avaliable on platformio now

ibaranov-cp, 2017-08-07:
0.8.4 is up as well now, this issue can likely be closed

PhracturedBlue, 2017-08-07:
I have changed the recommendation to use the staging branch of esp8266 core, as this seems to resolve the mqtt connection issues.

closed 2017-08-07 by PhracturedBlue