PhracturedBlue/ESP8266MQTTMesh #10: restart esp

josep112, 2017-08-26:
ESP8266MQTTMesh version: 0.8.1
AsyncMQTTClient version: 0.8.1
ESPAsyncTCP version: 1.0.1
ESP8266 Core version: 2.3.0
Arduino version : 1.8.3
Description

I'm trying to test ESP8266MeshHelloWorld.ino configure the credentials but I only get restart in the esp someone could help me
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v04c11db7
~ld
⸮
Exception (28):
epc1=0x4000bf80 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1a30 end: 3fff1c80 offset: 01a0

>>>stack>>>
3fff1bd0:  0000001c 00000000 3fff0b94 401004d8  
3fff1be0:  feefeffe feefeffe 3fff2c74 40201f95  
3fff1bf0:  00000000 00000001 feefeffe feefeffe  
3fff1c00:  feefeffe feefeffe feefeffe 0001c200  
3fff1c10:  0000001c 00000000 3fff0b94 40208648  
3fff1c20:  00000000 00000000 00000000 00000000  
3fff1c30:  40206adc feefeffe feefeffe 3fff0c50  
3fff1c40:  3fffdad0 3ffef090 3fff0b94 40206bc8  
3fff1c50:  40206adc feefeffe 4021147c 40211464  
3fff1c60:  feefeffe 00000000 3fff0c48 40210544  
3fff1c70:  feefeffe feefeffe 3fff0c60 40100718  
<<<stack<<<

melogonzaga, 2017-11-21:
same problem here, some one found a solution?

josep112, 2017-11-22:
I could not solve

PhracturedBlue, 2017-11-22:
Sorry, I guess I missed this issue.  I can't debug from the info provided, but if you:
a) make sure you have the latest version
b) provide the full stack trace (like above but from the latest version)
c) provide the .elf file from platformio

I can try to find the bug.

melogonzaga, 2017-11-23:
I'm using arduino IDE

PhracturedBlue, 2017-11-23:
I don't know where the arduino ide stores the elf file but it still must create one.

melogonzaga, 2017-11-23:
tried again on platform io and i found the same problem hers the elf file

`
[firmware.elf.txt](https://github.com/PhracturedBlue/ESP8266MQTTMesh/files/1499348/firmware.elf.txt)

`

PhracturedBlue, 2017-11-23:
you also need to provide the stack-trace from the esp8266 serial port that goes with it

PhracturedBlue, 2017-12-05:
I no longer support using the 2.3.0 Arduino core.  Too many issues, too hard to debug.  Please use the 2.4.0 development branch and open a new ticket if you still have issues.

closed 2017-12-05 by PhracturedBlue

alexiszecharies, 2018-01-08:
Maybe this helps:

the boot mode:(1,7) problem is known and only happens at the first restart after serial flashing.
if you do one manual reboot by power or RST pin all will work more info see: https://github.com/esp8266/Arduino/issues/1017