PhracturedBlue/ESP8266MQTTMesh #11: get start esp8266 mesh

miladshoaei, 2017-09-05:
Make sure you have the latest version of ESP8266MQTTMesh before reporting an issue.

### Please Provide the following information:
- ESP8266MQTTMesh version:
- AsyncMQTTClient version:
- ESPAsyncTCP version:
- ESP8266 Core version:

**Are you using platformio or Arduino?**

**Description of problem:**

miladshoaei, 2017-09-05:
Hello friends
Excuse me that my English language is bad. I want to implement Mesh Networks with esp8266, what should I study to get started?
Thanks

PhracturedBlue, 2017-09-05:
take a look at examples/ESP8266MeshHelloWorld
it should show how to use the Mesh.

miladshoaei, 2017-09-12:
@PhracturedBlue 
Thanks
Is it necessary to install the firmware?

miladshoaei, 2017-09-12:
Thanks
Is it necessary to install the firmware?

PhracturedBlue, 2017-12-05:
Assuming you are still interested.  Each node in the mesh must have a firmware based on the ESP82666MQTTMesh code.  The nodes don't all need the same firmeware though.  Each node must be initially connected to the MQTT broker to register it, after which time, it can be put in service to connect to other mesh nodes as needed

closed 2017-12-05 by PhracturedBlue