PhracturedBlue/ESP8266MQTTMesh #13: Callback not calling

brunonar, 2017-09-29:
Make sure you have the latest version of ESP8266MQTTMesh before reporting an issue.

### Please Provide the following information:
- ESP8266MQTTMesh version: 0.8.4
- AsyncMQTTClient version: 0.8.1
- ESPAsyncTCP version: 1.1.0
- ESP8266 Core version: 

**Are you using platformio or Arduino?**
Yes

I'm trying the Hello World example but the callback function never is called. I receive the "hello" message via my mosquitto broker and I tried to send all kind of topics to Sonoff Basic ESP8266. I'm using the GPIO 13 as it is used on this version. I changed the helloworld code (added just one line) to send a Serial.println() whenever I receive a topic on my ESP (before match the topic), but the print never comes.

Some topics that I used:
topicin/ESPID/
ESPID/
topicout/ESPID/
topicin/mesh-id/ESPID

Could someone help me?

PhracturedBlue, 2017-09-29:
The question about Arduino vs Palatformio wasn't meant to be yes/no, but instead a 'which one'?  Please tell me which platform you are developing on.

Where (in what file) did you define the override for LED_PIN?
Is stdout working at all (i.e. do you get debug data on stdout?)
the typical command to send a message to the node would be:
esp8266-in/mesh_esp8266-1/0
or 
esp8266-in/mesh_esp8266-1/1

you can also use the following to send a message to all nodes:
esp8266-in/broadcast/0
esp8266-in/broadcast/1

If that isn't helpful, please paste a log from the node's stdout with its startup, connect and polling process.

brunonar, 2017-09-29:
I'm sorry about that, I misread. I'm developing on platformio.

**Where (in what file) did you define the override for LED_PIN?**
On credentials (#define LED_PIN = 13). When I do a forced digitalwrite (HIGH or LOW) on loop() the LED works.

**Is stdout working at all (i.e. do you get debug data on stdout?)**
Yes, both debug data and Serial.println (on setup() and loop()) is working with my terminal.

**the typical command to send a message to the node would be:**
**esp8266-in/mesh_esp8266-1/0**
**or**
**esp8266-in/mesh_esp8266-1/1**

**you can also use the following to send a message to all nodes:**
**esp8266-in/broadcast/0**
**esp8266-in/broadcast/1**

I just tried those too but nothing. 

**If that isn't helpful, please paste a log from the node's stdout with its startup, connect and polling process.**

_"[onWifiConnect] Connecting to mqtt
[connect_mqtt] Attempting MQTT connection (192.168.0.35:1883)...
[onMqttConnect] MQTT Connected
[match_bssid] Trying to match known BSSIDs for 5E:CF:7F:42:83:79
[setup_AP] Initialized AP as 'mesh_esp8266-4'  IP '192.168.4.1'
[publish] Sending: sonoffout/mesh_esp8266-4/4359033=hello from 4359033 cnt: 0
[publish] Sending: sonoffout/mesh_esp8266-4/4359033=hello from 4359033 cnt: 1
[onMqttDisconnect] Disconnected from MQTT: 0
[onWifiDisconnect] Disconnected from Wi-Fi: DOUTOR GPS because: 200
[schedule_connect] Scheduling reconnect for 5.00 seconds from now
[scan] Scanning for networks
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
[scan] Found: 7
[scan] ...
[connect] 0 * 10:FE:ED:30:ED:AE -54
[connect] 1   10:FE:ED:30:ED:AE -99999
[connect] 2    -99999
[connect] 3    -99999
[connect] 4    -99999
[connect] Connecting to SSID : 'DOUTOR GPS' BSSID '10:FE:ED:30:ED:AE'
[onWifiConnect] Connecting to mqtt
[connect_mqtt] Attempting MQTT connection (192.168.0.35:1883)...
[onMqttConnect] MQTT Connected
[match_bssid] Trying to match known BSSIDs for 5E:CF:7F:42:83:79
[setup_AP] Initialized AP as 'mesh_esp8266-4'  IP '192.168.4.1'
[publish] Sending: sonoffout/mesh_esp8266-4/4359033=hello from 4359033 cnt: 2
[publish] Sending: sonoffout/mesh_esp8266-4/4359033=hello from 4359033 cnt: 3
[onMqttDisconnect] Disconnected from MQTT: 0
[onWifiDisconnect] Disconnected from Wi-Fi: DOUTOR GPS because: 200
[schedule_connect] Scheduling reconnect for 5.00 seconds from now
..."_

The code on HelloWorld is the orginal with just this line added:
`88    Serial.println("TEST TEST");`
On this place:
```
void callback(const char *topic, const char *msg) {
  Serial.println("TEST TEST");
  if (0 == strcmp(topic, (const char*) ID.c_str())) {
 ...
```
The Serial.println is never called. I tried to change it to a `digitalWrite(LED_PIN, HIGH**/LOW**);` too but it did nothing.

I attached my credentials.h too (changing the pass and names...).
I attached the platformio.ini. I tried both stage and normal firmwares and it changed nothing.
On config.mk I tried to disable the debug and enable it again.

[image]
[image]
[image]

Thank you very very much for the response.

PhracturedBlue, 2017-09-29:
So to be clear, you tried:
mosquitto_pub -t sonoffin/mesh_esp8266-4/0 -m ""
and
mosquitto_pub -t sonoffin/mesh_esp8266-4/1 -m ""

to send commands to the node?

It looks like your node is not maintaining its connection.

Most likely you need to use 'platform=espressif8266_stage' in the non-ssl section.  This is default in the git code, I'm not sure why you aren't getting that change.

Also please:
1) disable MESH_SECURE
2) remove the build_flags (why are you overriding the MAX_PACKET_SIZE?)
3) compile without SSL

brunonar, 2017-09-30:
Thank you! Thank you very much! It worked now.

I sent the code `mosquitto_pub -t sonoffin/mesh_esp8266-4/4359033 -m "0"` and the LED changed its state! I was confusing the "topic" variable with the mosquitto topic, but when the topic "sonoffin/mesh_esp8266-4/4359033" is sent the callback function only get "4359033" as topic.
I just do not understand the "-4" at the end of "mesh_esp8266-4". The first code you did send to me has "-1" at the end. I'll study more about it later.

I changed to stage and my node is now maintaining its connection perfectly. I changed it because the default LED_PIN=1 on **Sonoff Basic** was giving wts cause 4 and I thought that this would be the cause. The git code had no fault at this time. ;)

1. I didn't need to disable the MESH_SECURE, so I don't know how it'd behave.
2. I changed the MAX_PACKET_SIZE because I saw this flag on Tasmota firmware. I thought that it would be some kind of limitation of Sonoff Basic (but was the PIN). Anyway, now I tried with both 512 and 1152 and it ran normally.
3. I didn't try the SSL until now.

Thank you very much again and for those who will hack your Sonoff Basic you just will need to change 3 things on the code:
1. Use the flag `"board_flash_mode = dout"` on `platformio.ini`
2. Use the flag `-DLED_PIN=13` on `config.mk`
3. Add `#define LED_PIN 13` at `credentials.h`

closed 2017-09-30 by brunonar

PhracturedBlue, 2017-09-30:
Glad it is working.
mesh_esp8266-4 is the name of your node.  If you get another one it will be mesh_esp8266-5.

The '4359033' is not part of the id.  This  is the subtopic that is sent to the callback (which only cares about zero and 'not zero').  
If you override MAX_PACKET_SIZE to something other than 1152, OTA updates won't work.  Otherwise it doesn't matter.

I don't recommend using SSL unless you need it.  I believe MESH_SECURE is ignored if you don't have SSL enabled.