PhracturedBlue/ESP8266MQTTMesh #14: ESP32 Compatibility

Michael-Bell, 2017-10-11:
Is there a way to run this code on a ESP32 based device?

PhracturedBlue, 2017-10-12:
I don't know.  I don't have access to such a device.  In theory it shouldn't be too hard to port, but it depnds on how similar the wifi APIs are.  But since I dont have any ESP32 devices, it is unlikely this will happen unless someone submits a patch.

PhracturedBlue, 2017-12-05:
I did some initial work on the ESP32.  I now have it compiling through platformio in the 'esp32' branch.  I still don't have an ESP32 board to actually test with though

PhracturedBlue, 2018-03-05:
ESp32 support is now part of the master branch.  Still untested though.

closed 2018-03-05 by PhracturedBlue

martinberlin, 2019-04-17:
It compiles perfectly for me on ESP32. But I cannot get passt connect. After connection the callbacks are not called like in ESP8266. If anyone has an advice I'm all ears