PhracturedBlue/ESP8266MQTTMesh #18: ESP8266MQTTMesh build problem

alexiszecharies, 2017-11-25:
Make sure you have the latest version of ESP8266MQTTMesh before reporting an issue.

### Please Provide the following information:
I am using platformio which handles the versions
- ESP8266MQTTMesh version:
- AsyncMQTTClient version:
- ESPAsyncTCP version:
- ESP8266 Core version:

**If you are you using platformio or Arduino, which one?**
Home 0.3.2·Core 3.5.0b3

**Description of problem:**
I have tried to instantiate the ESP8266MQTTMesh class inside the setup in the hello world and this does  not working by any means. I am trying to do this because I get the MQTTserverIP from the SPIFFS in the setup and can not do it outside.

enesoztrk, 2017-11-25:
You can use painlessmesh which works for me.
[https://github.com/gmag11/painlessMesh](url)

alexiszecharies, 2017-11-25:
You are really funny, but if you could explain me why there is a difference iniziating the mesh inside or outside the setup that would be great

PhracturedBlue, 2017-11-25:
Can you give an example of the code you are trying to use?  As long as the ESP8266MQTTMesh object is global, I don't see any reason why you can't delay the initialization.  I've never tried it before, but if you can give a little more info about how you've done it and what errors you see (if any) I'll take a look.
An alternative might be that I could add code to initialize from a cfg file on SPIFFS.  I believe that could be done pretty easily.

alexiszecharies, 2017-11-25:
Thanks for your quick replay. How would a global declaration be done? When I tried this: 

ESP8266MQTTMesh mesh;

I got:

src/main.cpp:41:17: error: no matching function for call to 'ESP8266MQTTMesh::ESP8266MQTTMesh()'
ESP8266MQTTMesh mesh;
^
src/main.cpp:41:17: note: candidates are:
In file included from src/main.cpp:10:0:
/home/alexis/.platformio/lib/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.h:179:5: note: ESP8266MQTTMesh::ESP8266MQTTMesh(unsigned int, const char*, 
const char**, const char*, const char*, const char*, const char*, int, int, const char*, const char*)

ESP8266MQTTMesh(unsigned int firmware_id, const char *firmware_ver,
^
/home/alexis/.platformio/lib/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.h:179:5: note:   candidate expects 11 arguments, 0 provided
/home/alexis/.platformio/lib/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.h:169:5: note: ESP8266MQTTMesh::ESP8266MQTTMesh(const char**, const char*, 
const char*, int, const char*, const char*, const char*, int, const char*, const char*, int, const char*, const char*)
ESP8266MQTTMesh(const char **networks, const char *network_password,
^
/home/alexis/.platformio/lib/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.h:169:5: note:   candidate expects 13 arguments, 0 provided
In file included from src/main.cpp:10:0:
/home/alexis/.platformio/lib/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.h:60:7: note: ESP8266MQTTMesh::ESP8266MQTTMesh(const ESP8266MQTTMesh&)
class ESP8266MQTTMesh {
^
/home/alexis/.platformio/lib/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.h:60:7: note:   candidate expects 1 argument, 0 provided
/home/alexis/.platformio/lib/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.h:60:7: note: ESP8266MQTTMesh::ESP8266MQTTMesh(ESP8266MQTTMesh&&)
/home/alexis/.platformio/lib/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.h:60:7: note:   candidate expects 1 argument, 0 provided

So I tried to do:
const char*  networks[]       = { \
             "ssid 1",     \
             "ssid 2", \
             "",            \
             };
 ESP8266MQTTMesh mesh = ESP8266MQTTMesh::Builder(networks, "", "", 1111); 

with this outcome:

src/main.cpp:36:71: error: conversion from 'ESP8266MQTTMesh::Builder' to non-scalar type 'ESP8266MQTTMesh' requested
ESP8266MQTTMesh mesh = ESP8266MQTTMesh::Builder(networks, "", "", 1111); 

So went for  ESP8266MQTTMesh mesh = ESP8266MQTTMesh::Builder(networks, "", "", 1111).build(); 

void setup() {
  Serial.begin(115200);
  delay(15000);

  pinMode(LED_PIN,OUTPUT);
  /*if(getBrokerFromSPIFFS()!=""){
    mqtt_server=getBrokerFromSPIFFS().c_str();
  }else{
    mqtt_server=mqtt_default_server;
  }
  Serial.print("El server es");
  Serial.println(mqtt_server);
  */
  mesh = ESP8266MQTTMesh::Builder(networks, network_password, mqtt_default_server, mqtt_port)
  .setVersion(FIRMWARE_VER, FIRMWARE_ID)
  .setMeshPassword("")
  .setBaseSSID("192.168.1.1")
  .setMeshPort(1234)
  .setTopic("", "")
  .build();

  Serial.println("hi");
  Serial.println(NETWORK_PASSWORD);
  Serial.println(mqtt_server);
  mesh.setCallback(callback);
  mesh.begin();
}

Where the build is successful, but in the esp serial monitor i get : 

3fff3090:  00000000 00000000 00000000 00000000
3fff30a0:  00000000 00000000 00000000 00000000
3fff30b0:  00000000 00000000 00000000 00000000
3fff30c0:  00000000 00000000 00000000 00000000
3fff30d0:  00000000 00000000 00000000 00000000
3fff30e0:  00000000 00000000 00000000 00000000
3fff30f0:  00000000 00000000 00000000 00000000
3fff3100:  00000000 00000000 00000000 00000000
3fff3110:  00000000 00000000 00000000 00000000
3fff3120:  00000000 00000000 00000000 00000000
3fff3130:  00000000 00000000 00000000 00000000
3fff3140:  00000000 00000000 00000000 00000000
3fff3150:  00000000 00000000 00000000 00000000
3fff3160:  00000000 00000000 00000000 00000000
3fff3�

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v02f0c112
~ld

If i do it the right way by assigning the mesh instance a new ESP8266MQTTMesh::Builder like this:

void setup() {
  Serial.begin(115200);
  delay(15000);

  pinMode(LED_PIN,OUTPUT);
  /*if(getBrokerFromSPIFFS()!=""){
    mqtt_server=getBrokerFromSPIFFS().c_str();
  }else{
    mqtt_server=mqtt_default_server;
  }
  Serial.print("El server es");
  Serial.println(mqtt_server);
  */
  mesh = ESP8266MQTTMesh::Builder(networks, network_password, mqtt_default_server, mqtt_port)
  .setVersion(FIRMWARE_VER, FIRMWARE_ID)
  .setMeshPassword("")
  .setBaseSSID("192.168.1.1")
  .setMeshPort(1234)
  .setTopic("", "")
  .build(); //deleted the ESP8266MQTTMesh before mesh

  Serial.println("hi");
  Serial.println(NETWORK_PASSWORD);
  Serial.println(mqtt_server);
  mesh.setCallback(callback);
  mesh.begin();
}
 
I get this error:

src/main.cpp:58:8: error: use of deleted function 'ESP8266MQTTMesh& ESP8266MQTTMesh::operator=(ESP8266MQTTMesh&&)'
mesh = ESP8266MQTTMesh::Builder(networks, network_password, mqtt_default_server, mqtt_port)
^
In file included from src/main.cpp:10:0:
/home/alexis/.platformio/lib/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.h:60:7: note: 'ESP8266MQTTMesh& ESP8266MQTTMesh::operator=(ESP8266MQTTMesh&
&)' is implicitly deleted because the default definition would be ill-formed:
class ESP8266MQTTMesh {
^
/home/alexis/.platformio/lib/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.h:60:7: error: non-static const member 'const unsigned int ESP8266MQTTMesh:
:firmware_id', can't use default assignment operator
/home/alexis/.platformio/lib/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.h:60:7: error: non-static const member 'const int ESP8266MQTTMesh::mesh_por
t', can't use default assignment operator

which is the right way to do the assignment?

It would be great if you did this that you said: to initialize from a cfg file on SPIFFS!

PhracturedBlue, 2017-11-26:
This turns out to be a bit harder than I would have thought.  I added a new function to the mesh-builder 'buildptr()'
now you should be able to do:
```
ESP8266MQTTMesh *mesh;

void setup() {
  ...
  mesh = ESP8266MQTTMesh::Builder(networks, network_password, mqtt_server, mqtt_port)
                       .setVersion(FIRMWARE_VER, FIRMWARE_ID)
                       .setMeshPassword(mesh_password)
                       .buildptr();
```

One note:  Make sure all string parameters passed to the builder are global.  The ESP8266MQTTMesh object does not copy the char * objects, so you need to ensure they are persistent.

Note that I haven't released a new version with this change yet.  you'll need to pull the latest code from github directly.

alexiszecharies, 2017-11-28:
It is working now! Thank you very much! We are using your library for our final project at university

closed 2017-11-28 by PhracturedBlue

alexiszecharies, 2017-11-29:
I tried to set the callback i used to use and it is not working, i will open another issue