PhracturedBlue/ESP8266MQTTMesh #19: Callback calls with lastest commit restarts ESP

alexiszecharies, 2017-11-29:
Continuing this issue https://github.com/PhracturedBlue/ESP8266MQTTMesh/issues/18

**Description of problem:**

Despite succeeding to delay the broker IP set, when I tried to set the callback as i used to before this fix and testing it in the ESP I am getting this error:

Exception (9):
epc1=0x4020dfb5 epc2=0x00000000 epc3=0x00000000 excvaddr=0xfe090016 depc=0x00000000

ctx: sys 
sp: 3ffffaa0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffc40:  3fff03cc 000006da 000006da 3fff03cc  
3ffffc50:  00000010 3ffffce0 00000000 4010053d  
3ffffc60:  00000034 000036d0 000006da 3fff1f18  
3ffffc70:  00000030 00000001 3ffe863e 402111cc  
3ffffc80:  00000010 fe090016 00000000 4020e080  
3ffffc90:  3fff362c 00000047 3ffef1f4 40201651  
3ffffca0:  3fff03cc 3fff1f18 3fff1c48 402094fc  
3ffffcb0:  3fff362c 3fff1c44 00000001 40209648  
3ffffcc0:  3ffe90e2 3fff1c44 3ffffcf8 3ffffcf8  
3ffffcd0:  3fff362c 3fff1c44 3fff1f18 4020c19f  
3ffffce0:  00000000 00000000 00000000 00000000  
3ffffcf0:  00000000 00000000 00000000 00000000  
3ffffd00:  00000000 00000000 00000000 00000000  
3ffffd10:  ffffffff 00000000 3ffe9b01 00000008  
3ffffd20:  40213a32 3ffed120 3fff0ccc 00000000  
3ffffd30:  00000000 3fff173c 00000000 4020c1dd  
3ffffd40:  00000000 00000000 00000020 40201e70  
3ffffd50:  3f000000 40212e87 3ffed120 40208b87  
3ffffd60:  3f000000 3fff172c 00000000 402118c2  
3ffffd70:  00000002 00000000 00000000 4020f8c4  
3ffffd80:  00000000 40222edc 3fff427e 3fff4244  
3ffffd90:  3fff1444 3fff1584 40208b64 4020c1bc  
3ffffda0:  3f000000 000001e8 000001e8 00000002  
3ffffdb0:  3fff174c 3fff362c 00000000 3ffedb30  
3ffffdc0:  401009c2 00000001 00000000 00000000  
3ffffdd0:  3ffea2d6 00000026 3fff4244 40211781  
3ffffde0:  00000000 00000000 00000000 00000000  
3ffffdf0:  00000005 00000000 00000020 40210c14  
3ffffe00:  00000000 00000000 00000000 00000000  
3ffffe10:  40101b41 00000026 3fff4244 40210cb9  
3ffffe20:  00007fff 00000026 3fff1ce4 4020f6b9  
3ffffe30:  3fff1b04 00000000 4020ea84 402116f0  
3ffffe40:  3fff13fc 05bb82b2 4020ea3c 40211748  
3ffffe50:  00000026 3ffea230 3ffea230 00000001  
3ffffe60:  3fff1e51 3fff4244 4000050c 3fffc278  
3ffffe70:  40229a8d 05bb82b2 00000000 3fff1d38  
3ffffe80:  00000001 3fff13e4 3fff1ce4 4020f6ec  
3ffffe90:  3ffea23c 3fff0ce4 3fff1ce4 4020e2e8  
3ffffea0:  3fff1584 00000030 0000000f ffffffff  
3ffffeb0:  3fff0af0 3fff0af8 00000006 3fff09b8  
3ffffec0:  3fff1584 3fff09c0 3fff09bc 4020e33b  
3ffffed0:  3fff1584 3fff09c0 3fff09bc 40226d10  
3ffffee0:  00000026 6a01a8c0 00000018 00000026  
3ffffef0:  00000018 00000000 3ffec20d 40107404  
3fffff00:  40220000 00000000 0000007d 3fff0af4  
3fffff10:  3ffea2ae 3fff0af8 3fff13e4 40224101  
3fffff20:  3fff08c0 3fff144c 3fff144c 3ffeea40  
3fffff30:  00000000 3fff13e4 0000001c 3fff144c  
3fffff40:  3ffea2a0 00000000 3fff13e4 402234f9  
3fffff50:  3201a8c0 00000074 00000000 00000041  
3fffff60:  00000002 0000001a 40213343 3ffed3f0  
3fffff70:  3ffea280 3fffdcc0 3ffe99f0 3ffe99f0  
3fffff80:  402132b6 3ffed3f0 00000000 3fff14d4  
3fffff90:  3fffdc80 00000000 3fff13e4 4021c287  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ethets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

I am referencing the callback  at the top as all variables void callback(const char *topic, const char *msg);
 
And then implementing it below:

void callback(const char *topic, const char *msg) { 
    // topic is the command and msg the payload
    Serial.println(topic);
    Serial.println(msg);
}

This used to work before the fix in the issue. How should I declare the callback to make it global if it is not?

PhracturedBlue, 2017-11-29:
Please attach the full code you are using there are too many unknowns for me to be able to tell what is wrong from this.

alexiszecharies, 2017-11-29:
I reduced it to this, same mistake found  

`#include <Arduino.h>
#include "credentials.h"
#include <ESP8266MQTTMesh.h>
#include <ESP8266WiFi.h>

#define      FIRMWARE_ID        0x1337
#define      FIRMWARE_VER       "0.1"
const char*  networks[]       = NETWORK_LIST;
const char*  network_password = NETWORK_PASSWORD;
const char*  mesh_password    = MESH_PASSWORD;
const char*  base_ssid        = BASE_SSID;
const char*  mqtt_server      = MQTT_SERVER;
const int    mqtt_port        = MQTT_PORT;
const int    mesh_port        = MESH_PORT;
const int    LED_PIN          = LED_BUILTIN;
const int    RELAY_PIN        = RELAY_BUILTIN;
#if ASYNC_TCP_SSL_ENABLED
const uint8_t *mqtt_fingerprint = MQTT_FINGERPRINT;
bool         mqtt_secure      = MQTT_SECURE;
bool         mesh_secure      = MESH_SECURE;
#endif

String ID  = String(ESP.getChipId());

const char *out_topic   =   OUT_TOPIC;
const char *in_topic    =   IN_TOPIC;
unsigned long previousMillis = 0;
const long interval = 1000;
int   cnt = 0;

ESP8266MQTTMesh *mesh;

void callback(const char *topic, const char *msg);

void setup() {
  Serial.begin(115200);
  delay(10000);
  Serial.println("hi");
  Serial.println(NETWORK_PASSWORD);
  Serial.println(MQTT_SERVER);
  mesh = ESP8266MQTTMesh::Builder(networks, network_password, mqtt_server, mqtt_port)
                        .setVersion(FIRMWARE_VER, FIRMWARE_ID)
                        .setMeshPassword(mesh_password)
                        .setBaseSSID(base_ssid)
                        .setMeshPort(mesh_port)
                        .setTopic(in_topic, out_topic)
                        .buildptr();
  mesh->setCallback(callback);
  mesh->begin();
  pinMode(LED_BUILTIN, OUTPUT);
  // always use this to "mount" the filesystem

}

void loop() {
  unsigned long currentMillis = millis();
  if (! mesh->connected()){
    return;
  }
  if (currentMillis - previousMillis >= interval) {
    String cntStr = String(cnt);
    String msg = "hello from " + ID + " cnt: " + cntStr;
    mesh->publish(ID.c_str(), msg.c_str());
    previousMillis = currentMillis;
    cnt++;
  }

}

void callback(const char *topic, const char *msg) { //entered callback because last digit in /device_in/mesh_esp8266-X matched
  // topic is the command and msg the payload
  //reference in tasmota messages for command and payload https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
  Serial.println(topic);
  Serial.println(msg);

}
`
credentials.h:
#define NETWORK_LIST { \
             "PFCdud3",     \
             "ssid 2", \
             "",         \
             }// this need to be ended by an empty string
#define NETWORK_PASSWORD ""
#define MESH_PASSWORD    "pfcdud3_devic3s"
#define BASE_SSID        "mesh_esp8266-"
#define MQTT_DEFAULT_SERVER      "192.168.1.50"
#define MQTT_PORT        1883
#define MESH_PORT        1884
#define OUT_TOPIC        "/device_out/"
#define IN_TOPIC         "/device_in/"
#define LED_BUILTIN      13
#define RELAY_BUILTIN    12
#define BUTTON_BUILTIN    0

alexiszecharies, 2017-11-29:
The broker call is mosquitto_pub -d -t /device_in/mesh_esp8266-4/LedState -m ""

PhracturedBlue, 2017-11-29:
Thanks.  I don't see an obvious issue.  I need to run it on some hardware to see what is going on.
I'll try to get back to you within 24hrs.

If you'd like to try to do some debug on your own, I added a script in utils/dump_stacktrace.py
you run it as:
utils/dump_stacktrace.py <path to .elf file> <path to stacktrace file>

It assumes linux and platformio, and that you've build the firmware after adding '-g' to the build_flags, so I'm not sure if it will be useful to you.

PhracturedBlue, 2017-11-29:
If you use arduino (or even if you use platformio and just have arduino installed), this tool can do the same task:
https://github.com/me-no-dev/EspExceptionDecoder

PhracturedBlue, 2017-11-30:
Can you remove the 'delay(10000)' and see if it works?
I ran your code and I got a stacktrace during connection.  I need to look at that to understand why.  However, once I removed the delay() function the code worked as expected, and responded to mosquitto messages.
If it doesn't work,
please provide a full log from your serial terminal from the very beginning.  Also what is your build platform?  In the previous ticket you wrote 'Home 0.3.2·Core 3.5.0b3'  I don't know what that means.

PhracturedBlue, 2017-11-30:
I found two causes of crashes.  Basically when we dynamically create the ESP8266MQTTMesh object, not all memory is initialized to 0 which is different than when it is statically initialized.  Hopefully with these fixes, you will have better luck.  Let me know how it goes.

alexiszecharies, 2017-12-01:
I have done several tests and it is working! with and without delay. 

Now I am testing if the mesh is working as spected and I can not fully understand the subdomain assigination. I have 3 ESPs with the mesh code and when subscribing to the broker I see this:

Client mosqsub/868-raspberrypi received PUBLISH (d0, q0, r0, m0, '/device_out/mesh_esp8266-6/11823366', ... (26 bytes))
hello from 11823366 cnt: 3
Client mosqsub/868-raspberrypi received PUBLISH (d0, q0, r0, m0, '/device_out/mesh_esp8266-4/1000341', ... (26 bytes))
hello from 1000341 cnt: 28
Client mosqsub/868-raspberrypi received PUBLISH (d0, q0, r0, m0, '/device_out/mesh_esp8266-4/1138851', ... (27 bytes))
hello from 1138851 cnt: 126

But when I see the domain assignation method in your code I understand this number should be unique. When sending orders from broker this number is read by ESPs to know if the order is for them, right? 

Another thing i do not understand is how to make the ESPs have a new subdomain number. I deleted mosquito db and formatted SPIFFS and did not work.

PhracturedBlue, 2017-12-01:
the subdomains are stored in 2 locations as you noted:
1) the mqtt broker has a persistent value which it broadcasts on connection
2) each node stores its value in SPIFFS

The easiest way to change a node's value is to reassign it on the broker (with the node connected) and then reboot the node

for instance:
```
mosquitto_pub -r -t /device-in/bssid/92:F6:52:F2:51:2F -m ''5'
```
I don't know why you were unable to reset the value, but I would not delete the mosquitto database (that will break all nodes).  If you want to remove a single node:
```
mosquitto_pub -r -t /device-in/bssid/92:F6:52:F2:51:2F -n
```
You would also need to remove the file from the SPIFFS filesystem.  I am unfamiliar with the formatting process and whether it would work, but you can download the filesystem as a directory, modify it and restore it back to the device

alexiszecharies, 2017-12-01:
I did what you told me and now the broker is publishing this messages at the begining as spected:

pi@raspberrypi:~ $ sudo mosquitto_sub -d -t /#
Client mosqsub/1640-raspberryp sending CONNECT
Client mosqsub/1640-raspberryp received CONNACK
Client mosqsub/1640-raspberryp sending SUBSCRIBE (Mid: 1, Topic: /#, QoS: 0)
Client mosqsub/1640-raspberryp received SUBACK
Subscribed (mid: 1): 0
Client mosqsub/1640-raspberryp received PUBLISH (d0, q0, r1, m0, '/device-in/bssid/2E:3A:E8:0F:43:95', ... (1 bytes)) 5
Client mosqsub/1640-raspberryp received PUBLISH (d0, q0, r1, m0, '/device-in/bssid/5E:CF:7F:BD:24:5D', ... (1 bytes)) 4
Client mosqsub/1640-raspberryp received PUBLISH (d0, q0, r1, m0, '/device-in/bssid/5E:CF:7F:BD:69:06', ... (1 bytes)) 8

So now I need to delete the file from SPIFFS, platformio gives the option to operate the filesystem as a directory? Or who does that job? We have been deleting files uploading code to specifically do it which is very ineffective. Thank you very much for your big help!

PhracturedBlue, 2017-12-02:
You should not need to delete anything from the SPIFFS.  Just rebooting the node should be enough to change its subdomain.  Note that updating the subdomain is a 2 step process.
The 1st time the node connects, it receives its new subdomain, but it may not apply the new subdomain until the 2nd time the node is rebooted.  It all depends on when the node receives the update from the broker.

If you are having issues, then post the serial log of the node booting.

PhracturedBlue, 2017-12-02:
I did some testing and rebooting the node shouldn't be needed.
```
2017-12-01 18:08:18.489233   esp8266-out/mesh_esp8266-5/10498856                                = b'hello from 10498856 cnt: 7'
2017-12-01 18:08:18.822702   esp8266-in/bssid/5E:CF:7F:A0:33:28                                = b'111'
2017-12-01 18:08:23.489493   esp8266-out/mesh_esp8266-111/10498856                                = b'hello from 10498856 cnt: 8'
```
Here you can send me changing the subdomain while the node is running.
Note that I just pushed an update that fixes a bug when the node count > 10.  You probably want that change.

alexiszecharies, 2017-12-02:
It has not worked yet I have tried:

        mosquitto_pub -r -t /esp8266-in/bssid/5E:CF:7F:BD:69:06 -m '8'
        mosquitto_pub -r -t /device-in/bssid/5E:CF:7F:BD:69:06 -m '8'
        mosquitto_pub -r -t /device_in/bssid/5E:CF:7F:BD:69:06 -m '8'
        mosquitto_pub -r -t /esp8266_in/bssid/5E:CF:7F:BD:69:06 -m '8'

since I do not understand if I have to put device or esp 8266 before "in" because in the broker I receive messages as Client mosqsub/3347-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/mesh_esp8266-4/1138851', ... (27 bytes)) that have the  word "device" and "_" instead of esp8266-out.

Maybe I am getting the MAC address wrong? I get it from this phrase in the ESP8266 (Sonoff) from the serial monitor:
 
[match_bssid] Trying to match known BSSIDs for 5E:CF:7F:B4:69:06

As requested I post the serial log of the node booting:

[begin] Starting Firmware 1337 : 0.1
[begin] OTA Start: 0x4a000 OTA End: 0x6b000
[begin]  ==> '/bssid/2E:3A:E8:0F:43:95'
[begin]  ==> '/bssid/5E:CF:7F:BD:24:5D'
[begin]  ==> '/bssid/5E:CF:7F:B4:69:06'
[begin]  ==> '/bssid/5E:CF:7F:BD:69:06'
[begin] 0
[begin] Setup Complete
[scan] Scanning for networks
[onWifiDisconnect] Disconnected from Wi-Fi:  because: 8
[schedule_connect] Scheduling reconnect for 5.00 seconds from now
[scan] Found: 0
Not Connected to mesh
Not Connected to mesh
Not Connected to mesh
Not Connected to mesh
Not Connected to mesh
[scan] Scanning for networks
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
Not Connected to mesh
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
Not Connected to mesh
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
Not Connected to mesh
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
[scan] Found: 12
[scan] Found SSID: 'Lender' BSSID '8C:E0:81:7F:09:3E'
[scan] Did not match SSID list
[scan] Found SSID: '' BSSID '2E:3A:E8:11:60:A3'
[match_bssid] Trying to match known BSSIDs for 2E:3A:E8:11:60:A3
[scan] Failed to match BSSID
[scan] Found SSID: '' BSSID '2E:3A:E8:0F:43:95'
[match_bssid] Trying to match known BSSIDs for 2E:3A:E8:0F:43:95
[scan] RSSI: -12
[scan] Found SSID: '' BSSID 'FA:8F:CA:90:A7:BF'
[match_bssid] Trying to match known BSSIDs for FA:8F:CA:90:A7:BF
[scan] Failed to match BSSID
[scan] Found SSID: 'PFCdud3' BSSID '10:7B:44:E3:D1:10'
[scan] Matched
[scan] RSSI: -18
[scan] Found SSID: 'Alter ego' BSSID '64:70:02:79:4E:38'
[scan] Did not match SSID list
[scan] Found SSID: ' -Mourglia' BSSID '14:60:80:3C:1E:55'
[scan] Did not match SSID list
[scan] Found SSID: 'Cobre2' BSSID '98:FC:11:AC:20:5B'
[scan] Did not match SSID list
[scan] Found SSID: 'NEW_HOME_GUEST' BSSID 'C0:25:67:25:79:D1'
[scan] Did not match SSID list
[scan] Found SSID: 'NEWHOME' BSSID 'C0:25:67:25:79:D0'
[scan] Did not match SSID list
[scan] Found SSID: 'Peru' BSSID '54:22:F8:93:4D:76'
[scan] Did not match SSID list
[scan] Found SSID: '' BSSID 'FA:8F:CA:7C:EC:28'
[match_bssid] Trying to match known BSSIDs for FA:8F:CA:7C:EC:28
[scan] Failed to match BSSID
[connect] 0 * 10:7B:44:E3:D1:10 -18
[connect] 1   2E:3A:E8:0F:43:95 -12
[connect] Connecting to SSID : 'PFCdud3' BSSID '10:7B:44:E3:D1:10'
Not Connected to mesh
Not Connected to mesh
Not Connected to mesh
[onWifiConnect] Connecting to mqtt
[connect_mqtt] Attempting MQTT connection (192.168.1.50:1883)...
[onMqttConnect] MQTT Connected
[match_bssid] Trying to match known BSSIDs for 5E:CF:7F:B4:69:06
[setup_AP] Initialized AP as 'mesh_esp8266-6'  IP '192.168.6.1'
[publish] Sending: /device_out/mesh_esp8266-6/11823366=hello from 11823366 cnt: 0
Subscribe acknowledged.
  packetId: 1
  qos: 0
[onMqttMessage] Message arrived [/device_in/bssid/5E:CF:7F:BD:69:06] '8'
[publish] Sending: /device_out/mesh_esp8266-6/11823366=hello from 11823366 cnt: 1
[publish] Sending: /device_out/mesh_esp8266-6/11823366=hello from 11823366 cnt: 2

PhracturedBlue, 2017-12-02:
I dont think you have the right MAC address.  I don't think the current MAC address gets printed out on the consol (though you can print it out yourself via:
```
Serial.println(WiFi.softAPmacAddress().c_str())
```
I normally figure out the MAC address from the broker:
When I connect to the broker I see something like this:
```
2017-12-02 12:58:07.744868   esp8266-in/bssid/92:F6:52:F2:51:2F                                = b'3'
2017-12-02 12:58:07.745407   esp8266-in/bssid/9C:3D:CF:F1:CB:D6                                = b'3'
2017-12-02 12:58:07.745994   esp8266-in/bssid/5E:CF:7F:A0:33:EB                                = b'4'
2017-12-02 12:58:07.746525   esp8266-in/bssid/5E:CF:7F:A0:33:28                                = b'5'
2017-12-02 12:58:07.747078   esp8266-in/bssid/5E:CF:7F:A0:2F:9E                                = b'6'
2017-12-02 12:58:07.747593   esp8266-in/bssid/5E:CF:7F:0D:40:F2                                = b'7'
2017-12-02 12:58:07.748266   esp8266-in/bssid/5E:CF:7F:0D:44:57                                = b'8'
2017-12-02 12:58:07.748788   esp8266-in/bssid/5E:CF:7F:11:2F:46                                = b'9'
```
That is the MAC to subdomain mapping.
the prefix will be '/device-in/' for you since that is what you defined.

You would want something like:
```
mosquitto_pub -r -t /device-in/bssid/5E:CF:xx:xx:xx:xx -m '5'
```

alexiszecharies, 2017-12-02:
I added these lines in the code : 

  Serial.print("My MAC address is: ");
  Serial.println(WiFi.softAPmacAddress().c_str());

And got this result:

My MAC address is: 5E:CF:7F:B4:69:06, which is the same as in the first message from where I got the MAC in the previous post:  [match_bssid] Trying to match known BSSIDs for 5E:CF:7F:B4:69:06

I deleted all nodes from db now. When i do: sudo mosquitto_sub -d -t /#

This is what I get after restarting the ESPs:

Client mosqsub/3711-raspberryp sending CONNECT
Client mosqsub/3711-raspberryp received CONNACK
Client mosqsub/3711-raspberryp sending SUBSCRIBE (Mid: 1, Topic: /#, QoS: 0)
Client mosqsub/3711-raspberryp received SUBACK
Subscribed (mid: 1): 0
Client mosqsub/3711-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/mesh_esp8266-6/11823366', ... (26 bytes))
hello from 11823366 cnt: 0
Client mosqsub/3711-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/mesh_esp8266-4/1000341', ... (25 bytes))
hello from 1000341 cnt: 0
Client mosqsub/3711-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/mesh_esp8266-4/1138851', ... (25 bytes))
hello from 1138851 cnt: 0
Client mosqsub/3711-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/mesh_esp8266-6/11823366', ... (26 bytes))
hello from 11823366 cnt: 1
Client mosqsub/3711-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/mesh_esp8266-4/1000341', ... (25 bytes))
hello from 1000341 cnt: 1

It seems they are not storing nor checking the mosquitto db to get their subdomain. Could you tell me how to see the ESP filesystem in order to delete all files and test again?

PhracturedBlue, 2017-12-02:
the easiest thing is probably to do:
```
esptool erase_flash
```
and then re-upload your firmware.

alexiszecharies, 2017-12-03:
I flashed 3 ESP and put my code in them, when i initialized the third one it got the samesub domain as the second one. I leave the initalization log of the first two, tell me if you need the same for the third one because i did not get to the beginning of it.

[begin] Starting Firmware 1337 : 0.1
[begin] OTA Start: 0x4a000 OTA End: 0x6b000
[begin] 0
[begin] Setup Complete
[scan] Scanning for networks
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
Not Connected to mesh
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
Not Connected to mesh
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
Not Connected to mesh
[schedule_connect] Scheduling reconnect for 0.50 seconds from now
[scan] Found: 15
[scan] Found SSID: 'LENDER' BSSID '8C:E0:81:7F:09:3E'
[scan] Did not match SSID list
[scan] Found SSID: 'AntelGydAj' BSSID '08:3F:BC:D9:2A:9A'
[scan] Did not match SSID list
[scan] Found SSID: 'ZALMA' BSSID '8C:E0:81:7D:5A:CB'
[scan] Did not match SSID list
[scan] Found SSID: 'PFCdud3' BSSID '10:7B:44:E3:D1:10'
[scan] Matched
[scan] RSSI: -32
[scan] Found SSID: 'Alter ego' BSSID '64:70:02:79:4E:38'
[scan] Did not match SSID list
[scan] Found SSID: '' BSSID 'FA:8F:CA:90:A7:BF'
[match_bssid] Trying to match known BSSIDs for FA:8F:CA:90:A7:BF
[scan] Failed to match BSSID
[scan] Found SSID: 'Alter ego 2' BSSID 'C4:E9:84:D2:16:64'
[scan] Did not match SSID list
[scan] Found SSID: ' -Mourglia' BSSID '14:60:80:3C:1E:55'
[scan] Did not match SSID list
[scan] Found SSID: 'Nino Wi-Fi' BSSID 'DC:A4:CA:EC:F2:A0'
[scan] Did not match SSID list
[scan] Found SSID: 'Peru' BSSID '54:22:F8:93:4D:76'
[scan] Did not match SSID list
[scan] Found SSID: 'Nino Guests' BSSID 'DE:A0:F2:EC:CA:A1'
[scan] Did not match SSID list
[scan] Found SSID: 'Cobre2' BSSID '98:FC:11:AC:20:5B'
[scan] Did not match SSID list
[scan] Found SSID: '' BSSID 'FA:8F:CA:7C:EC:28'
[match_bssid] Trying to match known BSSIDs for FA:8F:CA:7C:EC:28
[scan] Failed to match BSSID
[scan] Found SSID: 'TROUVILLE' BSSID '8C:E0:81:7E:18:94'
[scan] Did not match SSID list
[scan] Found SSID: 'francisco1' BSSID '8C:E0:81:7D:E5:BD'
[scan] Did not match SSID list
[connect] 0 * 10:7B:44:E3:D1:10 -32
[connect] Connecting to SSID : 'PFCdud3' BSSID '10:7B:44:E3:D1:10'
Not Connected to mesh
Not Connected to mesh
Not Connected to mesh
[onWifiConnect] Connecting to mqtt
[connect_mqtt] Attempting MQTT connection (192.168.1.50:1883)...
[onMqttConnect] MQTT Connected
[match_bssid] Trying to match known BSSIDs for 5E:CF:7F:B4:69:06
Subscribe acknowledged.
  packetId: 1
  qos: 0
[onMqttMessage] Message arrived [/device_in/bssid/5E:CF:7F:B4:69:06] '4'
[read_subdomain] Failed to read /bssid/5E:CF:7F:B4:69:06
[setup_AP] Initialized AP as 'mesh_esp8266-4'  IP '192.168.4.1'
[publish] Sending: /device_out/mesh_esp8266-4/new_device={"_id":11823366,"type":1}
[publish] Sending: /device_out/mesh_esp8266-4/11823366=hello from 11823366 cnt: 0
[publish] Sending: /device_out/mesh_esp8266-4/new_device={"_id":11823366,"type":1}
[publish] Sending: /device_out/mesh_esp8266-4/11823366=hello from 11823366 cnt: 1
[publish] Sending: /device_out/mesh_esp8266-4/new_device={"_id":11823366,"type":1}
[publish] Sending: /device_out/mesh_esp8266-4/11823366=hello from 11823366 cnt: 2
[... 211 lines cut ...]
[scan] Found SSID: 'DIRECT-yT-BRAVIA' BSSID 'DA:5D:E2:21:85:C1'
[scan] Did not match SSID list
[scan] Found SSID: 'AntelGydAj' BSSID '08:3F:BC:D9:2A:9A'
[scan] Did not match SSID list
[connect] 0 * 10:7B:44:E3:D1:10 -22
[connect] 1   5E:CF:7F:B4:69:06 -10
[connect] Connecting to SSID : 'PFCdud3' BSSID '10:7B:44:E3:D1:10'
Not Connected to mesh
Not Connected to mesh
Not Connected to mesh
[onWifiConnect] Connecting to mqtt
[connect_mqtt] Attempting MQTT connection (192.168.1.50:1883)...
[onMqttConnect] MQTT Connected
[match_bssid] Trying to match known BSSIDs for 2E:3A:E8:11:60:A3
[setup_AP] Initialized AP as 'mesh_esp8266-5'  IP '192.168.5.1'
[publish] Sending: /device_out/mesh_esp8266-5/1138851=hello from 1138851 cnt: 18
[publish] Sending: /device_out/mesh_esp8266-5/1138851=hello from 1138851 cnt: 19
[publish] Sending: /device_out/mesh_esp8266-5/1138851=hello from 1138851 cnt: 20
[publish] Sending: /device_out/mesh_esp8266-5/1138851=hello from 1138851 cnt: 21
[publish] Sending: /device_out/mesh_esp8266-5/1138851=hello from 1138851 cnt: 22
[publish] Sending: /device_out/mesh_esp8266-5/1138851=hello from 1138851 cnt: 23
[publish] Sending: /device_out/mesh_esp8266-5/1138851=hello from 1138851 cnt: 24
[publish] Sending: /device_out/mesh_esp8266-5/1138851=hello from 1138851 cnt: 25

_______________________________________________________________________________

alexiszecharies, 2017-12-03:
Changing the subdomain number did not work in this case either

PhracturedBlue, 2017-12-03:
yes I need to see the 3rd device logs.  These 2 devices worked properly.  There is nothing wrong with the logs.

Make sure you remove the bad MAC from your broker for the 3rd node, BUT DO NOT REMOVE THE DATA FOR THE 1st TWO NODES!
Also wipe the node and reflash to be sure.

If you want me to debug why changing the subdomain doesn't work, you need to capture the log from the node from the beginning AND include your attempt to change the subdomain).  I just looked at your logs, and I think the reason it didn't work before is because the MAC of the node was:
`5E:CF:7F:B4:69:06` but the change command was:
```
[onMqttMessage] Message arrived [/device_in/bssid/5E:CF:7F:BD:69:06] '8'
```
those are not the same mac address (one is 'B4' the other is 'BD')

alexiszecharies, 2017-12-03:
I noticed the  'B4' and the other 'BD' issue, I will try to do the test again later, but i think i found out the problem behind de subdomain initialization problem. Look at this broker received messages:

Messages from the first ESP to register in the broker:

_________________________________________________________________________________

Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/1000341', ... (25 bytes))
hello from 1000341 cnt: 4
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/1000341', ... (25 bytes))
hello from 1000341 cnt: 5
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/1000341', ... (25 bytes))
hello from 1000341 cnt: 6
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/1000341', ... (25 bytes))
hello from 1000341 cnt: 7
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/1000341', ... (25 bytes))
hello from 1000341 cnt: 8
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/1000341', ... (25 bytes))
hello from 1000341 cnt: 9
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_in/bssid/2E:3A:E8:0F:43:95', ... (1 bytes))
4
Client mosqsub/1896-raspberryp sending PINGREQ
Client mosqsub/1896-raspberryp received PINGRESP
Client mosqsub/1896-raspberryp sending PINGREQ
Client mosqsub/1896-raspberryp received PINGRESP
Client mosqsub/1896-raspberryp sending PINGREQ
Client mosqsub/1896-raspberryp received PINGRESP
Client mosqsub/1896-raspberryp sending PINGREQ
Client mosqsub/1896-raspberryp received PINGRESP
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/11823366', ... (26 bytes))
hello from 11823366 cnt: 0
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/11823366', ... (26 bytes))
hello from 11823366 cnt: 1
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/11823366', ... (26 bytes))
hello from 11823366 cnt: 2
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/11823366', ... (26 bytes))
hello from 11823366 cnt: 3
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/11823366', ... (26 bytes))
hello from 11823366 cnt: 4
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/11823366', ... (26 bytes))
hello from 11823366 cnt: 5
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/11823366', ... (26 bytes))
hello from 11823366 cnt: 6
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/11823366', ... (26 bytes))
hello from 11823366 cnt: 7
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/11823366', ... (26 bytes))
hello from 11823366 cnt: 8
Client mosqsub/1896-raspberryp received PUBLISH (d0, q0, r0, m0, '/device_out/11823366', ... (26 bytes))
hello from 11823366 cnt: 9
Client mosqsub/1896-raspberryp sending PINGREQ
Client mosqsub/1896-raspberryp received PINGRESP

__________________________________________________________________________________

Messages from the second ESP (disconnected first one from power cord) :

pi@raspberrypi:~ $ mosquitto_sub -d -t /#
Client mosqsub/1901-raspberryp sending CONNECT
Client mosqsub/1901-raspberryp received CONNACK
Client mosqsub/1901-raspberryp sending SUBSCRIBE (Mid: 1, Topic: /#, QoS: 0)
Client mosqsub/1901-raspberryp received SUBACK
Subscribed (mid: 1): 0
Client mosqsub/1901-raspberryp received PUBLISH (d0, q0, r1, m0, '/device_in/bssid/2E:3A:E8:0F:43:95', ... (1 bytes))
4
Client mosqsub/1901-raspberryp sending PINGREQ
Client mosqsub/1901-raspberryp received PINGRESP
__________________________________________________________________________________

For some reason it did not manage to send the information, and what is worse, this is the initilization log of the second ESP (should not be publishing if it is not connected):

__________________________________________________________________________________
[... 348 lines cut ...]
  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
    if ( mesh.connected()){
      if(!SPIFFS.exists("/signed_in_broker/")){
        //sign_up_to_broker();
        previousMillis = currentMillis;
      }
      String cntStr = String(cnt);
      String msg = "hello from " + ID + " cnt: " + cntStr;
      mesh.publish(ID.c_str(), msg.c_str());
      cnt++;
    } else {
        //return;
        Serial.println("Not Connected to mesh");
    }
    previousMillis = currentMillis;
  }

}

void callback(const char *topic, const char *msg) { 
  Serial.println(topic);
  Serial.println(msg);
}

alexiszecharies, 2017-12-03:
I broker was installed like this:

sudo apt-get install mosquitto

PhracturedBlue, 2017-12-03:
I know that the mesh.connected indicates a connection before a node is assigned.  That is a bug, but not a major one, and I think unrelated to the issue.

I have asked a couple of times, and am still not sure:  What platform do you use to build this code?  platformio?  arduion IDE?  something else?

In your logs I keep seeing the node connecting and disconnecting.  That plus the fact that you are seeing publish messages with no reception on the broker side makes me think you have the wrong version of the espressiff8266 library.  You need to have the development version of this library, otherwise connections are unreliable.

PhracturedBlue, 2017-12-03:
I have checked in an update that should enforce that you have the latest version of the esp8266 library.  If in doubt, grab the latest version.

alexiszecharies, 2017-12-03:
I use PlatformIO, version 3.5.0rc5. On the global setup of platformio I have:
AsyncMqttClient 0.8.1; 
ESPAsyncTCP 1.1.3;
The latest commit of your library.

I dont know where the espresiff8266 library you are talking about is imported to check the version.

Know when trying to compile I get : 
Compiling .pioenvs/esp01/lib/AsyncMqttClient_ID346/AsyncMqttClien
t.o
Compiling .pioenvs/esp01/lib/ESP8266MQTTMesh/ESP8266MQTTMesh.o
/home/alexis/.platformio/lib/ESP8266MQTTMesh/src/ESP8266MQTTMesh.
cpp:46:2: error: #error "This version of the ESP8266 library is n
ot supported"
#error "This version of the ESP8266 library is not supported"

As you mentioned. Would you be so kind as to tell me what should I install and where? because in that log I understand that is the version of your library that is not updated and I have the lastest commit.

PhracturedBlue, 2017-12-03:
Make sure your platformio.json has this:
```
platform = espressif8266_stage
```
You can see an example in the ESP8266MeshHelloWorld example

alexiszecharies, 2017-12-03:
This is my platformio.ini file. Can remember how I ended up with this configuration because I started working with the ESP8266MeshHelloWorld example. 

[env:esp01]
platform = espressif8266
framework = arduino
board = esp01
#  lib_deps = ESP8266MQTTMesh
#  board_f_cpu = 80000000L
#  board_f_flash = 40000000L
#  upload_resetmethod = ck
board_flash_mode = dout
#  upload_speed = 115200

shajek, 2017-12-03:
I have same error, if i use " platform = espressif8266_stage " 

> PlatformManager: Installing espressif8266_stage
> Error: Detected unknown package 'espressif8266_stage'

if i use " platform = https://github.com/platformio/platform-espressif8266.git#feature/stage "

> C:\Users\Meiton\.platformio\lib\ESP8266MQTTMesh_ID1747\src\ESP8266MQTTMesh.cpp:46:2: error: #error "This version of the ESP8266 library is not supported"
> #error "This version of the ESP8266 library is not supported"

alexiszecharies, 2017-12-03:
I tried uncommenting that line but still got the #error "This version of the ESP8266 library is not supported"
error

shajek, 2017-12-03:
i cannot get it work in PlatformIO, but in Arduino is everything OK. But how work callback ? I have serial.print inside, but i didnt see anything on Serial Monitor... I publish to esp8266-in/MY_NODE_ID .... in Serial Monitor i can see

> [onMqttMessage] Message arrived [esp8266-in/3048040] '1'

but nothing from my callback function doesnt work (i didnt see anything)

alexiszecharies, 2017-12-03:
Shajek:
In the broker I put : 
mosquitto_sub -d -t /# to see what is going on and the in another terminal something like:

mosquitto_pub -d -t  /device_in/mesh_esp8266-4/LedPower -m "1" and that should make the node code enter in the callback

PhracturedBlue, 2017-12-03:
Instructions on using the staging platform:
http://docs.platformio.org/en/stable/platforms/espressif8266.html#using-arduino-framework-with-staging-version
I can't help with arduino.
@shajek : Note that @alexiszecharies has a non-standard configuration.
For you, you want to do:
```
mosquitto_pub -d -t esp8266-in/mesh_esp8266-4/3048040 -m "1"
```
Make sure the subdomain (the '4' part) is correct.

shajek, 2017-12-03:
Ok, thanks, callback work great. But i have small problem with that part with "mesh_esp8266-x"... I dont care if is Node connected to mesh or directly to WiFi, i want care only about uniqe ID, or how i can easily find Node without know where it is connected?

PS: i have still problem with "espressif8266_stage"

i have

> [env:d1_mini]
> platform = espressif8266_stage
> board = d1_mini
> framework = arduino

but still error:

> > Executing task: platformio.exe run <
> 
> [12/03/17 19:45:44] Processing d1_mini (platform: espressif8266_stage; board: d1_mini; framework: arduino)
> -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> PlatformManager: Installing espressif8266_stage
> Error: Detected unknown package 'espressif8266_stage'
> The terminal process terminated with exit code: 1

alexiszecharies, 2017-12-03:
@PhracturedBlue take a look at this https://github.com/platformio/platform-espressif8266/issues/61 I had that same bug happen to me for the last 2 hours when trying to complile your hello world. I think you should change it to what it says in that issue. Moreover, now that i dont have that error i still get this error:

.piolibdeps/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.cpp:46:2: error: #error "This version of the ESP8266 library is not supported"
#error "This version of the ESP8266 library is not supported"
^

I believe I have the lastest version. This is the hello world config I am using now:

[platformio]
env_default = nossl

[env:ssl]
platform = espressif8266
framework = arduino
board = esp01_1m
#build_flags = -DASYNC_TCP_SSL_ENABLED=1 -DGATEWAY_ID=10499051 -DLED_PIN=2 -g
build_flags = -DASYNC_TCP_SSL_ENABLED=1
lib_deps = ESP8266MQTTMesh

[env:nossl]
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
framework = arduino
board = esp01_1m
lib_deps = ESP8266MQTTMesh
#build_flags = -DLED_PIN=2 -g

alexiszecharies, 2017-12-03:
@shajek look at what I just posted. We are having the same issue.

PhracturedBlue, 2017-12-03:
Ok, apparently they changed the build system and I must still have the old one.  I'll refresh and see what is going on

shajek, 2017-12-03:
Yes, we have same issue :) can i ask you , where do you get your build flags ? Or i miss something? Because i dont find no documentation except readme, i am intrested in "gateway_id"

PhracturedBlue, 2017-12-03:
don't use 'GATEWAY_ID'.  It is for debug only, and is used to force a specific node to act as a gateway to the WiFi network.  I.e.e it is useful when you have a bunch of nodes in the same room, and you wnat to enforce a specific topology for testing.  It is not otherwise useful.

shajek, 2017-12-03:
Ok, but as i wrote above. I have mesh of sensors in area and i want read from each Nodes (i know where is specific Node with specific ID) but problem is that i dont know where are they connected (mesh_esp8266-X) (yes, in subscribe i can use wild card) and bigger problem Publish some information to each Node because i dont know they mesh numbers

PhracturedBlue, 2017-12-03:
@shajek I don't understand.  You say you know their ID but not where they are connected.  What does that mean?
I can think of 3 ways that we can identify a node:
The ChipID: ESP.getChipId()
The MAC address: WiFi.softAPmacAddress()
and the subdomain (mesh_esp8266-X)
The broker maps the MAC address o the subdomain.  We don't use the ChipID for anything so I don't store it, but you can send:
```
mosquitto-pub -t esp8266-in/fw/broadcast -n
```
Which will tell each node to send back its firmware ID.  I'll add ChipID to that since it is easy enough to do, but I'm not sure that is what you are looking for.

PhracturedBlue, 2017-12-03:
@alexiszecharies 
Try using this isntead:
```
platform = https://github.com/platformio/platform-espressif8266.git#feature/2.4.0-rc2
```
I'm having some issues updating my platformio so I haven't tested it yet myself.

shajek, 2017-12-03:
> [12/03/17 20:54:56] Processing d1_mini (platform: https://github.com/platformio/platform-espressif8266.git#feature/2.4.0-rc2; board: d1_mini; framework: arduino)
> -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> Collected 30 compatible libraries
> Looking for dependencies...
> Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
> |-- <ESP8266MQTTMesh> v0.8.7
> |   |-- <ESPAsyncTCP> v1.1.3
> |   |   |-- <ESP8266WiFi> v1.0
> |   |-- <AsyncMqttClient> v0.8.1
> |   |   |-- <ESPAsyncTCP> v1.1.3
> |   |   |   |-- <ESP8266WiFi> v1.0
> |   |-- <ESP8266WiFi> v1.0
> |   |-- <Ticker> v1.0
> |-- <ESP8266WiFi> v1.0
> Compiling .pioenvs\d1_mini\src\main.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\cont_util.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_eboot_command.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_flash_utils.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_flash_utils.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_i2s.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_main.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_noniso.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_phy.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_postmortem.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_si2c.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_timer.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_wiring.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_wiring_analog.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_wiring_digital.o
> In file included from C:\Users\Meiton\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiClient.h:25:0,
> from C:\Users\Meiton\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39,
> from src\main.cpp:2:
> c:\users\meiton\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
> C:\Users\Meiton\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
> #define max(a,b) ((a)>(b)?(a):(b))
> ^
> c:\users\meiton\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_bvector.h: In member function 'std::vector<bool, _Alloc>::size_type std::vector<bool, _Alloc>::_M_check_len(std::vector<bool, _Alloc>::size_type, const char*) const':
> C:\Users\Meiton\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
> #define max(a,b) ((a)>(b)?(a):(b))
> ^
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_wiring_pulse.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_wiring_pwm.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\core_esp8266_wiring_shift.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\debug.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\heap.o
> Compiling .pioenvs\d1_mini\FrameworkArduino\libb64\cdecode.o
> *** [.pioenvs\d1_mini\src\main.o] Error 1
> =============================================================================================== [ERROR] Took 3.53 seconds ===============================================================================================
> The terminal process terminated with exit code: 1

alexiszecharies, 2017-12-03:
@shajek  I use this flags sudo pio run -t upload -e nossl for the hello world. I got them from watching the platformio terminal when compiling. I have to be in the folder which has the platformio.ini file. This is the path I have /home/alexis/ESP8266MQTTMesh/examples/ESP8266MeshHelloWorld

shajek, 2017-12-03:
@PhracturedBlue sorry for bad description. Problem is this : I dont know where my node connect because they can move and signal maybe variabile. I dont know if Node connect to mesh or direct to Wifi. I know, that first run of node must be directly connected to Wifi. Number X in mesh_8266-X change how ? One uniqe ID have one mesh_8266-X ? or can be more nodes with mesh_8266-X ?

> mesh_8266-1/123456
> mesh_8266-1/987654
> mesh_8266-1/123789

or 

> mesh_8266-1/123456
> mesh_8266-2/987654
> mesh_8266-3/123789

i say, that i know wich ESP ID is where physically and i dont know where they connected and i want write web application to assign to bunch of nodes something called "Zone1/2/3".  Because i map UniqeID to virtualnumber which i assigned to zone for me would be best publish to  esp8266-in/+/UNIQE-ID , because i can read it esp8266-out/+/UniqeID

alexiszecharies, 2017-12-03:
The problem you are showing in the last post I had it in some occasion. I was comparing something I should not I think. I would adivece to try to compile the hello world example first.

alexiszecharies, 2017-12-03:
@PhracturedBlue I tried with that line, got the same versioning error. #error "This version of the ESP8266 library is not supported"

shajek, 2017-12-03:
sorry, my post about that nodes is offtopic of this issue... yes with that line i have still error

> ompiling .pioenvs\nossl\lib\ESP8266MQTTMesh_ID1747\ESP8266MQTTMesh.o
> Archiving .pioenvs\nossl\lib\libTicker.a
> .piolibdeps\ESP8266MQTTMesh_ID1747\src\ESP8266MQTTMesh.cpp:46:2: error: #error "This version of the ESP8266 library is not supported"
> #error "This version of the ESP8266 library is not supported"
> ^
> *** [.pioenvs\nossl\lib\ESP8266MQTTMesh_ID1747\ESP8266MQTTMesh.o] Error 1
> =============================================================================================== [ERROR] Took 9.60 seconds ===============================================================================================
> 
> ======================================================================================================= [SUMMARY] =======================================================================================================
> Environment ssl         [SKIP]
> Environment nossl       [ERROR]
> =============================================================================================== [ERROR] Took 9.61 seconds ===============================================================================================
> The terminal process terminated with exit code: 1

i ask here, becasue i dont want open issue, because that is not issue... only my little problem... please read it again, and if you can help with that problem ... help me (i upload it and edit in Arduino becasue platformio is broken)

PhracturedBlue, 2017-12-03:
@shajek  p;ease move this to a different issue.  I cannot address both your issue and @alexiszecharies issue in the same ticket.  It is too confusing.

shajek, 2017-12-03:
ok :) but i still cannot compile in PlatformIO, same issue as @alexiszecharies

PhracturedBlue, 2017-12-03:
yes, and I will address that part of the issue here.

PhracturedBlue, 2017-12-03:
@alexiszecharies, @shajek 
I have updated the build so it should build using feature/stage (and not using 2.3.0)
Please try it now and let me know if it works.  I haven't done any real testing using this version of the arduino-core, so I hope it still works properly.

alexiszecharies, 2017-12-03:
I cloned the repo and same result. Deleted ESP8266MQTTMesh from global config and added it again with no luck.

PhracturedBlue, 2017-12-03:
'Same result' means you have this error:
```
This version of the ESP8266 library is not supported
```
?

Look in `.piolibdeps/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMesh.cpp` (from wherever your platformio.ini is, and make sure it contains:
```
#ifndef pgm_read_with_offset //Requires Arduino core 2.4.0
    #error "This version of the ESP8266 library is not supported"
#endif
````
If it doesn't have that, you don't have the right library.  If it does have that you still don't have a compatible espressif8266 library.

I did a clean platformio install with features/stage and it does compile for me now.  I had issues with the
```
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
```
line, and ended up needing this instead:
```
platform = espressif8266@4f59cef
```
I don't know why.  platformio seems to make it difficult to point at the libraries I want to use

alexiszecharies, 2017-12-04:
'Same result' means you have this error: yes.

I could make it compile!!!!!!!!!! I really do not understand the difference between .piolibdeps and .platformio/lib/ where aparently the code was being updated. Will look into this later. I will start with the tests  with my code again. Thank you very much!!!!!!!!

alexiszecharies, 2017-12-04:
I am having this problem with my code 

.pioenvs/nossl/lib/libESP8266MQTTMesh_ID1747.a(ESP8266MQTTMesh.o): In function `strlcat':
ESP8266MQTTMesh.cpp:(.text.strlcat+0x8): multiple definition of `strlcat'
/home/alexis/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-strlcat.
o):/Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/string/../../../.././newlib/libc/string/strlcat.c:49: first defined here
collect2: error: ld returned 1 exit status

Gonna check it out later. Thank you for your help!

PhracturedBlue, 2017-12-04:
That specific issue should now be fixed.  I have an alternate implementation of strlcat in the mesh code.  It would collide with the libc variant.  I've renamed mine to prevent that error.  But please open new issues for things unrelated to the connectivity issue this thread has been about.

alexiszecharies, 2017-12-04:
Ok, note taken. Today I had little time to do some testing, but after flashing the 3 ESPs and putting my software I could see that the are no disconnections  as before and they got a different number for their subdomain in no time. It looks really promising.

PhracturedBlue, 2017-12-04:
I'm closing this issue for now.  If you see problems with nodes not properly communicating with the broker, please reopen or file a new ticket.

closed 2017-12-04 by PhracturedBlue