PhracturedBlue/ESP8266MQTTMesh #20: Publish to node without knowing mesh number

shajek, 2017-12-03:
Make sure you have the latest version of ESP8266MQTTMesh before reporting an issue.

### Please Provide the following information:
- ESP8266MQTTMesh version: 0.8.7
- AsyncMQTTClient version: 0.8.1
- ESPAsyncTCP version: 1.1.3
- ESP8266 Core version: 2.4.0rc2

**If you are you using platformio or Arduino, which one?**
Arduino 1.8.5

**Description of problem:**

 I dont know where my node connect because they can move and signal maybe variabile. I dont know if Node connect to mesh or direct to Wifi. I know, that first run of node must be directly connected to Wifi. Number X in mesh_8266-X change how ? One uniqe ID have one mesh_8266-X ? or can be more nodes with mesh_8266-X ?

mesh_8266-1/123456
mesh_8266-1/987654
mesh_8266-1/123789
or

mesh_8266-1/123456
mesh_8266-2/987654
mesh_8266-3/123789
i say, that i know wich ESP ID is where physically and i dont know where they connected and i acctualy write web application that assign to bunch of nodes "virtual numbers" to something called "Zone1/2/3". Because i map UniqeID to virtualnumbers which are assigned to zone. **For me would be best publish to esp8266-in/+/UNIQE-ID** , because i can read it esp8266-out/+/UniqeID

PhracturedBlue, 2017-12-03:
I think you may misunderstand how the mesh works.
Each node must connect via wiFi at least one time.  This is how it gets its identifier and learns about other nodes in the mesh.  After that, the next time it reboots it will try to connect to the strongest available AP, whether that be WiFi or another mesh node.

the 'mesh_8266-4' is a unique ID.  It doesn't matter where the node connects, it will always have an identifier with that name (assigned the first time it boots up)

If you want to send a message to all nodes, you can send:
esp8266-in/broadcast/<xxx>
And all connected nodes will receive that message.

PhracturedBlue, 2018-03-04:
This issue has been made irrelevant with the 1.0 code release.  Nodes no longer need to connect to the broker once before trying to connect to other nodes on the mesh.

closed 2018-03-04 by PhracturedBlue