PhracturedBlue/ESP8266MQTTMesh #21: How proper use SSL between Node and Broker

shajek, 2017-12-04:
- ESP8266MQTTMesh version:0.8.8
- AsyncMQTTClient version:0.8.1
- ESPAsyncTCP version:1.1.3
- ESP8266 Core version:2.4.0rc2

Latest PlatformIO

Hello, how to use proper SSL ?

I have properly installed Mosquito and defined my letsencrypt certificate
when i use:

> mosquitto_sub -h mqtt.MYPAGE.com -t "#" -p 8883 --cafile /etc/ssl/certs/ca-bundle.crt -u "someuser" -P "somepassword"

> mosquitto_pub -h mqtt.MYPAGE.com -t test -m "hello again" -p 8883 --cafile /etc/ssl/certs/ca-bundle.crt -u "someuser" -P "somepassword"

everything is OK

i generate SSL fingerprint in this format:

> const uint8_t MQTT_FINGERPRINT[] = {0x92,0xd3,0x90,0x57,0x56,0xd9,0x02,0x29,0x45,0x0f,0x1a,0x3c,0xb6,0x77,0x78,0xdb,0x78,0xe3,0x1c,0x02};

and of cource enabled SSL:

> #define MQTT_SECURE      true

everything compiled and uploaded OK

but node show only this:

> [onMqttDisconnect] Disconnected from MQTT: 5
> [connect_mqtt] Attempting MQTT connection (something:8883)...
> [onMqttDisconnect] Disconnected from MQTT: 0
> [connect_mqtt] Attempting MQTT connection (something:8883)...
> [onMqttDisconnect] Disconnected from MQTT: 5
> [connect_mqtt] Attempting MQTT connection (something:8883)...
> [onMqttDisconnect] Disconnected from MQTT: 0
> [connect_mqtt] Attempting MQTT connection (something:8883)...

and after few attempts show stacktrace...

and my broker show this:

> 1512420001: Socket error on client <unknown>, disconnecting.
> 1512420001: New connection from 1.2.3.4 on port 8883.
> 1512420002: Socket error on client <unknown>, disconnecting.
> 1512420002: New connection from 1.2.3.4 on port 8883.
> 1512420002: Socket error on client <unknown>, disconnecting.
> 1512420002: New connection from 1.2.3.4 on port 8883.

PhracturedBlue, 2017-12-04:
To clarify:
1) you generated the fingerprint using: `utils/get_mqtt_fingerprint.py` ?
2) you set MQTT_SECURE = true
3) you  set MESH_SECUURE = false

If so, please post the results of get_mqtt_fingerprint.py in case it says anything useful.

You  might try subscribing to the broker with 'mosuitto_sub -d' to see a bit more info about what is going on.

shajek, 2017-12-04:
yes, i used `utils/get_mqtt_fingerprint.py` from yours 0.8.8 version
yes, i set MESH to false and MQTT to true

result of get_mqtt:
`INFO: Found signature algorithm: rsa_sha256
const uint8_t MQTT_FINGERPRINT[] = {0x92,0xd3,0x90,0x57,0x56,0xd9,0x02,0x29,0x45,0x0f,0x1a,0x3c,0xb6,0x77,0x78,0xdb,0x78,0xe3,0x1c,0x02};`

mosuitto_sub -d , show nothnig, because Node doesnt connect as server log say:
`1512423654: Socket error on client <unknown>, disconnecting.
1512423654: New connection from SOME_IP on port 8883.
`

PhracturedBlue, 2017-12-04:
You also compiled using -DASYNC_TCP_SSL_ENABLED=1 right?  This needs to be done from the compile step, and cannot be set in a header file.

Maybe your broker has some other debug logs that may provide more info about why the connection isn't going through.

shajek, 2017-12-04:
yes, i use complete your example with

> [env:ssl]
> platform = espressif8266@4f59cef
> framework = arduino
> board = d1_mini
> build_flags = -DASYNC_TCP_SSL_ENABLED=1
> lib_deps = ESP8266MQTTMesh

and you are right, i have new message from debug (i use mosquitto)
`1512424258: Client connection from MyIP failed: error:140760FC:SSL routines:SSL23_GET_CLIENT_HELLO:unknown protocol.`

PhracturedBlue, 2017-12-04:
and you built using: `pio run -e ssl -t upload`?
There aren't a whole lot af parameters available here, and it really looks like the Mesh is trying to use the secure port with an unsecure connection.  You could try changing the port back to 1883 (but leave everything else the same) and see if it connects to an unsecured broker.  That would tell us whether the mesh is even trying to use SSL.

shajek, 2017-12-05:
Yes i built it with that flags. I today again make clean install of Platformio, manually add your 0.8.8 (because platformio have 0.8.7), again paste my fingerprint, set flags, and for now everything works :) thank you for help!
PS: Platformio on Windows is terrible, it gave me 543267436 error during downloading all depencies for esp8266 and some of that i must install manually .......

PhracturedBlue, 2017-12-05:
I have a love/hate relationship with platformio.  I like the library management, and cmdline utilities.  I don't like the GUI or the difficulty of debugging anything.  Ah well.  Glad it is working now.

closed 2017-12-05 by PhracturedBlue