PhracturedBlue/ESP8266MQTTMesh #22: Connection from node to node when the broker is not involved

alexiszecharies, 2017-12-09:
I tested the mesh feature and it works great. I was wondering how is the logic behind the mesh. When a node losses connection from broker and some time goes through it looks for known bssids and then publishes a message? That would not make sense because nodes can not become brokers. I dont get what protocol the code does to send messages between nodes.

PhracturedBlue, 2017-12-09:
You might look at this image:
https://github.com/PhracturedBlue/ESP8266MQTTMesh/issues/3#issuecomment-319101592

A given node can connect to the broker directly (though an AP) or to another node (which will act as a relay to the broker).  Every node can act as a relay for up to 4 other nodes.
If a node loses its upstream connection (to the broker or a relay node), it will immediately disconnect all downstream nodes (those using this node as a relay).  This has a cascading effect, so if your wiFi AP goes down, all nodes will disconnect.  A node only allows connections to itself once it has an upstream connection.

The node behavior is to listen to the upstream connection (either the broker or another node).  It will locally try to process each message AND it will pass that message to all the nodes connected to itself.   The node also listens to its connected nodes, and will forward all packets upstream (generally without inspecting the packets).  this means that nodes can't really talk to each other, instead they need to talk through the broker, with any intermediate nodes acting as a relay.

The socket protocol used when a node is acting like a relay is just a single TCP packet which contains a topic and message.

alexiszecharies, 2017-12-21:
Understood, thanks

closed 2017-12-21 by alexiszecharies

johnmay1, 2017-12-28:
Hi @PhracturedBlue ,

I did a couple of test for mesh connection. 

In my first scenario, i have placed the one espA near by wifi router, so it's connected directly to mqtt broker. I have placed the second espB first nearby wifi router , so it can connect with mqtt broker directly. Both espA and espB working as expected in this scanario. Now i have moved espB far from wifi router but within reach of espA, so it can connect with mqtt broker via espA using mesh network, which is not working. Am I doing something wrong? 

In my second scenario, i have defined GATEWAY_ID as chipId of espA, so espA is direclty connected with mqtt broker and espB should connect with mqtt broker using espA mesh network. In this scanario espB is connected with espA mesh network but i am not seeing any message from espB on mqtt broker. 

I am using HelloWorld example for both cases.

PhracturedBlue, 2017-12-28:
If you are using GATEWAY_ID, then set it to the chipid of espA and load the same firmware onto both espA and ESPB AFTER you have at least once associated both espA and ESPB to the broker.  The idea of GATEWAY_ID is to prevent all nodes that do not match the GATEWAY_ID from trying to connect via your AP.  I.e. the GATEWAY_ID is only useful for preventing a node from connecting to the AP, it does not force a node to connect to the AP.  It allows you to test the mesh without needing to move the nodes far away.

That said, I don't see anything wrong with your setup.  You will need to provide serial logs from espA and espB during the connection to see what is going on.

johnmay1, 2017-12-28:
> AFTER you have at least once associated both espA and ESPB to the broker.

How i can associate espB with a broker ?

PhracturedBlue, 2017-12-29:
You said you did already.  install the stock 'helloWorld' without setting GAEWAYID and allow it to connect to your AP.  It will be assigned a subdomain (some number between 4 and 255) which you will see as a persistent variable on the broker.

johnmay1, 2017-12-29:
> You said you did already.

Yes,i did. I thought it will overwrite the SPIFF files if i upload sketch with GATEWAY_ID. So confused how it will associate with broker.  Yes, now i am able see data on broker from espB via espA.