PhracturedBlue/ESP8266MQTTMesh #23: Connect node to hidden SSID WiFI

shajek, 2017-12-09:
Is any way to connect Node to wifi with hidden SSID ?

PhracturedBlue, 2017-12-10:
It isn't currently possible.  The code could probably be enhanced to allow specifying the WiFi AP by BSSID rather than by name.  I'll think about how that could be done.

PhracturedBlue, 2017-12-10:
It turns out this was pretty easy.  I have implemented hidden network support now, though it is untested.
instead of the name, instead specify the MAC address of your access Point.

It will loo something like: `AA:BB:CC:DD:EE:FF`

shajek, 2017-12-10:
I'm going to test it. Thank you. Do you think it is possible to connect via SSID OR BSSID ? That would be great and so universal :)

shajek, 2017-12-10:
```
[connect] 0 * 00:8C:54:F4:ED:71 -68
[connect] Connecting to SSID : '00:8C:54:F4:ED:71' BSSID '00:8C:54:F4:ED:71'
[onWifiDisconnect] Disconnected from Wi-Fi: 00:8C:54:F4:ED:71 because: 201
```
I have unsecured hidden Wifi... but, when i connect to wifi on PC or smartphone, it ask me SSID name .... why here i must not write it ?

PhracturedBlue, 2017-12-10:
i probably should have looked into this more 1st.  I just checked in a new solution, It is a bit more complicated.
in platformio.ini add `build_flags = -DUSE_EXTENDED_NETWORKS=1`
then instead of using a list of strings for the networks use this structure:
```
typedef struct {
    const char *ssid;
    const char *bssid;
    bool hidden;
} wifi_conn;
```
example:
```
const wifi_conn networks[] = {
    {"ssid", NULL, false},
    {"hidden_ssid","XX:XX:XX:XX:XX:XX", true},
    NULL
};
```
The final 'NULL' is important.
All hidden networks need both ssid and bssid.  If bssid is specified for a non-hidden network, it will limit the match to that specific AP.

I only compile tested it as I don't have any hidden networks to play with, but let me know how it goes.

PhracturedBlue, 2018-03-04:
Extended networks are now default and the old network list is no longer supported in version 1.0

closed 2018-03-04 by PhracturedBlue