PhracturedBlue/ESP8266MQTTMesh #25: Connect to WiFi without scan

shajek, 2017-12-15:
Hello,
is there any option to stop scanning (maybe only first time) and only connect to predefined Wifi ? I'm trying to make battery module with this library and this proces take too much time.

PhracturedBlue, 2017-12-15:
No, but in ESP8266MQTTMesh::begin() there is a call to 'WiFi.disconnect()'  If you remove that, I believe the node will automatically attempt to reconnect to the previous connection.  If you are trying to use deepsleep it won't work though, because the async callback will notice the callback and push for a new scan.  This dove-tails into your other question, which is that we need to add specific code to properly de-initialize and re-initialize when doing a deepsleep.

shajek, 2018-02-26:
Hello, me again. I have another question wich is similiar to this thread. Is there some way to speed up booting ? Currently is there 11s cca to first message... 11s vs 9s of booting time to firt message is for me 12day in battery life 
Thank you

PhracturedBlue, 2018-02-26:
I don't have any great ideas, but if you profile your stratup times, perhaps we could find some fat to trim