PhracturedBlue/ESP8266MQTTMesh #26: Use of deepsleep

shajek, 2017-12-15:
Hello again,
if i try use DeepSleep, sometime its work sometime not ....
```

if (0 == strcmp(topic, (const char*) sleep.c_str())) {
       val = atoi(msg);
       //Serial.printf("i am in IF for slep");
        if(val > 0) {
            ESP.deepSleep(val * 1000000);
            delay(500); 
        }
    }
```

if i have uncommented Serial.print, deepsleep work correctly (delay on the end is the same story, without it deepsleep crash)
but if i dont have Serial.print or i have some delay deepsleep crash again

```
[onMqttMessage] Message arrived [esp8266-in/mesh_esp8266-1/sleep] '60'
[onMqttDisconnect] Disconnected from MQTT: 0

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```

PhracturedBlue, 2017-12-15:
i don't think you can sleep with a mesh node.  If you shut off the wireless, then you can't act as a relay, and you will miss all messages not just for yourself but for your child nodes.
I suppose we could define a new type of node that can only act as an endpoint and does not act as a relay.  that node would likely need to shutdown its connections before entering sleep and then wake up and reconnect.  That is an expensive process, but could be done for nodes that only send data.  It doesn't seem like it would be that much work, but I haven't really looked yet

shajek, 2017-12-15:
Yop, i think like you. This "battery node" cannot act like a relay (maybe yes, when broker send to all battery nodes same sleep time). But yep, is useless to starting AP for meshing ... because time and energy...
From serial output you see, that message
`[onMqttDisconnect] Disconnected from MQTT: 0`
Problem, is, that right now, deepsleep working, but very unstable ... 2/10 sleeps crash .. but 8/10 works good

PhracturedBlue, 2017-12-15:
your crash report isn't helpful, since there is no backtrace to debug.
I am only guessing, but to make deepsleep work we probably need to take down any mqtt/socket connections before entering deep-sleep.
But because of the asynchronous nature of the code you probably need something like:
```
void loop() {
  if (sleep_ready) {
    sleep_ready = false;
    ESP.deepsleep(...);
  }
  if (sleep) {
   sleep = false;
   WiFi.disconnect();
   sleep_ready = true;
  }
}
```
As I said though, that is just a guess and is untested.  Also, it would be better to have an ability to flush all messages before shutdown, since it is possible for the network layer to cache transmissions, and if you want to wakeup, do something, then go back to sleep we want to ensure that any messages are actually transmitted during 'do-something'