PhracturedBlue/ESP8266MQTTMesh #27: Number of nodes supported

johnmay1, 2017-12-22:
Hi @PhracturedBlue ,

Is there any limitation on number of nodes in mesh network ? 

Regards,
John

PhracturedBlue, 2017-12-22:
The current max # of nodes is 252.  Raising this to 999 would not be difficult.  even more than that is possible with some minor changes.

johnmay1, 2017-12-22:
> The current max # of nodes is 252. Raising this to 999 would not be difficult. even more than that is with with some minor changes.
That's great.

So we can connect 252 ESP-01 appx. ? 
Are you seeing any limitation due to onboard RAM memory of ESP-01 ?

PhracturedBlue, 2017-12-22:
let me clarify.
The entire mesh can have a total of 252 ESP-01 nodes.  However, any given node can only support 4 children, so the depth of the mesh may be very deep if you try to approach that limit.  The 4-node limit is dictated by the esp8266 core, and was set using core version 2.3.  It is possible that the limit has been raised in the v2.4 core (which is required by the ESP8266MQTTMesh for stability reasons).  I haven't done any experimentation with that.  If there was no core limitation, we could probably have each node support 20 other nodes easily and maybe more.

johnmay1, 2017-12-23:
Cool... Thanks for clarification.

closed 2017-12-23 by johnmay1