PhracturedBlue/ESP8266MQTTMesh #28: Retain Messages

johnmay1, 2018-01-04:
How to publish retain messages to mqtt broker ? 

I can't see any parameter available in "mesh.publish" to retain messages on mqtt broker.

PhracturedBlue, 2018-01-04:
it isn't currently supported.  Supporting it will require changing the mesh protocol, so I need to think about it a bit to decide how best to do that.

johnmay1, 2018-01-04:
ok, Is it on the roadmap?

PhracturedBlue, 2018-01-05:
Yes, I am working on it now.  I should have something available in a couple days

johnmay1, 2018-01-05:
Awsome... Thanks.

PhracturedBlue, 2018-01-05:
I have checked in a new feature that implements this.  The mesh protocol changed, so all nodes need to be recompiled.  Further testing is needed before I release a new version with these changes.

johnmay1, 2018-01-05:
oh wow... that's so fast. I will test it sometime today. 

Thanks.

johnmay1, 2018-01-13:
Got some time to test today. Mesh node connected without any issue but "callback" is not being called on mesh node. A message is correctly forwarded from the gateway node to mesh node.
Mesh node is reporting : `[handle_client_data] Received: msg from 192.168.41.1 on STA`
I think it should call "onMessage" or "parse_message" method to further process the method.

Also, I have to try couple of time to get it work and seeing duplicate files in "bssid" directory with different values. It might be a good idea to put some checking in to avoid duplicate files.

Noticed that bssid published only for the first time when it assigns a subdomian. Do you think it will be helpful just to publish self bssid whenever it joins the network. Just a thought.

John

PhracturedBlue, 2018-01-13:
If you get that message, than 'idx' == 0 in handle_client_data.  That should cause parse_message() to get called.  If you aren't seeing your callback called, it likely means that something is wrong with the message format.   You could add some debug into parse_message() to see what is going on.  I can also try to help if you report what the message is (it shuld be on the next line after the 'Received' line (--> '...') and your node configuration.

I don't think SPIFFS should allow duplicate files.  I'd need more info on exactly what you are seeing there.

My expectation is that nodes do not change their node-id so it shouldn't be necessary to resend on every connection.  We heavily rely on the broker to store the node map so if we can't trust it we have bigger issues.  If you are having specific issues that would make transmitting the bssid/subdomain necessary, please file a separate ticket for that.

PhracturedBlue, 2018-01-14:
Yes that looks like a bug.  data sent from the node to the mqtt broker will have that extra character.  Data sent back from the broker should not.

PhracturedBlue, 2018-01-14:
I just checked in what is hopefully a fix for this.

johnmay1, 2018-01-14:
Thanks but i can't see any commit. Is it on master branch ?

PhracturedBlue, 2018-01-14:
Sorry.  It helps if I do a 'git push' too.  Try it now

johnmay1, 2018-01-14:
great... will test it. you might need to fix this  "# We only send the msgType upstream" in mean time.

PhracturedBlue, 2018-01-14:
ugh...sorry.  1st thing in the morning here.

johnmay1, 2018-01-14:
np... topic is removed now... 

[handle_client_data] --> '={"on":1}'

PhracturedBlue, 2018-01-14:
I fixed that too.  If it is still not working, I'll test it on real hardware.  I just don't have anything ready to go at the moment.

johnmay1, 2018-01-14:
It's working but mesh node is keep disconnected from gateway node with following,

[onTimeout] Got timeout  192.168.41.1: 5113
[onDisconnect] Disconnected from mesh
[onWifiDisconnect] Disconnected from Wi-Fi:  because: 8

PhracturedBlue, 2018-01-14:
That is a classic symptom of having the wrong esp8266-core version.  can you check your core and see what version you have?  2.4.0 was recently released and should work properly (the snapshots of 2.4.0 did work), but I haven't tested it.  2.3.0 had the issue you are seeing

johnmay1, 2018-01-14:
I didn't see that issue before and don't know how to check the core version. I am building code using platformio and it's pointing to "platform = https://github.com/platformio/platform-espressif8266.git#feature/stage" in my .io file. 
I tried after sometime looks like it's working fine now.

Thanks for quick response and fixing the bug.

closed 2018-01-14 by johnmay1