PhracturedBlue/ESP8266MQTTMesh #29: WiFi Manager

alexiszecharies, 2018-01-08:
Is there a posibility to integrate this mesh library with something like https://github.com/tzapu/WiFiManager to have the possibility to load credentialls from a web server?

PhracturedBlue, 2018-01-08:
With the latest changes, the network setup can be deferred, so it should be possible already.
Basically, you need to store the credentials in SPIFFS and manage them yourself.  If they don't exist, load the Wifi manager.  otherwise, initialize the mesh.
I think direct integration of these two projects is beyond the scope of the mesh network code.

alexiszecharies, 2018-01-09:
I am having some trouble passing the networks variable to de mesh builder since it asks for a cont char**

.piolibdeps/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMeshBuilder.h:30:5: note: ESP8266MQTTMesh::Builder::Builder(const char**, const char*, const char*, int)
Builder(const char   **networks,
^

.piolibdeps/ESP8266MQTTMesh_ID1747/src/ESP8266MQTTMeshBuilder.h:30:5: note:   no known conversion for argument 1 from 'const char*' to 'const char**'

I am trying with:

ssid="{"+ ssid + "," + "" + "}";
networks[]=ssid.c_str();

Why does the networks variable need to be a const char** ?

PhracturedBlue, 2018-01-09:
the networks is a null terminated string list:
```
const char *networks[] = {"network1", "network2", NULL};
```
Using a c++ string may not be a good idea, because I don't know that you can gaurantee the immutability of the c_str, but you could try:

```
const char *networks[] = { ssid.c_str(), NULL};
```
or:
```
const char * networks[2];
networks[0] = ssid.c_str;
networks[1] = NULL;
```

alexiszecharies, 2018-01-10:
Thank you. The integration is pretty easy as you said.

closed 2018-01-10 by alexiszecharies