PhracturedBlue/ESP8266MQTTMesh #30: Receiving message size

johnmay1, 2018-01-08:
Hi @PhracturedBlue,

What is the size limit for receiving message ? 
I was looking at [Memory management](https://github.com/marvinroger/async-mqtt-client/blob/bbc8465504c37045f8c403a9ff91300471d00486/docs/3.-Memory-management.md) . As per their document max receive size is about 1460 bytes per call. 

How it's handled here? 

Regards,
John

PhracturedBlue, 2018-01-08:
The maximum supported mqtt size is 1152 bytes.  I believe that includes both the topic and the message.

johnmay1, 2018-01-14:
Yes, i saw that message buffer size if fixed size on header file. We can support unlimited message size if we can somehow pass "size_t len, size_t index, size_t total" parameters of "onMqqtMessage" to mesh node "onData" method. 
Do you think something is possible to implement?

PhracturedBlue, 2018-01-15:
it is possible but would require dynamic memory allocation to allow nodes to repeat arbitrary message sizes.  I spent a lot of effort to have no dynamic memory allocation in this library, as it tends to be a poor choice for embedded applications.  What is your need that you expect such large messages?

johnmay1, 2018-01-16:
@PhracturedBlue Understood. I don't have hard requirement for large message. Just wondering if can remove such limitation  from software side.

closed 2018-03-04 by PhracturedBlue