PhracturedBlue/ESP8266MQTTMesh #31: How to upload pre-configured setting from windows

johnmay1, 2018-01-17:
Hi @PhracturedBlue ,
I am trying to upload pre-configured settings by following doc https://github.com/PhracturedBlue/ESP8266MQTTMesh/blob/master/docs/Filesystem.md

As ":" is special character for windows file system and it won't allow me to create file with special character. 

Is it still possible to upload such configuration from windows ?

John

johnmay1, 2018-01-27:
@PhracturedBlue  did you get a chance to look into it ?

PhracturedBlue, 2018-01-27:
I honestly don't think you'll be able to do this on Windows.
From what I've read, the ':' cannot be part of a filename in windows.
We could replace ':' with '.' when looking up the file names in the code, but I really don't see much ROI in that change as it will break the existing filesystem layout.
One possible fix would be to write a python script to build the spiffs filesystem from a text file rather than from a directory.  I may look into that, but it isn't high on my priority list at the moment.

johnmay1, 2018-01-28:
hi @PhracturedBlue ... thanks for look into it. Even i don't think it's priority. Just wanted to know if there is any alternative you are using here. 
By the way, do you documented roadmap anywhere ? 
Have you look at this library https://github.com/marvinroger/homie-esp8266/ ?
John

PhracturedBlue, 2018-01-28:
Someone brought it to my attention recently, but I wasn't aware of it when I wrote this library.
I do all of my development on Linux where ':' is a legal filename component.  If you direly need this functionality, you might consider a linux virtualmachine to generate the filesystem.

I do not have any roadmap as this library meets my current needs as it is.  I will entertain requests from users though.

PhracturedBlue, 2018-03-04:
I have released the 1.0 version of this code, and there is no longer any need to pre-load the nodes with info about other nodes.

closed 2018-03-04 by PhracturedBlue