PhracturedBlue/ESP8266MQTTMesh #35: GATEWAY_ID configuration

C-monC, 2018-02-11:
- ESP8266MQTTMesh version: 0.8.8
- AsyncMQTTClient version: 0.8.2
- ESPAsyncTCP version: 1.0.1
- ESP8266 Core version: 2.4
Using Platform.io

How does one configure the GATEWAY_ID so that not all the nodes have to be directly connected to the AP.

PhracturedBlue, 2018-02-11:
That is not the purpose of GATEWAY_ID.  The GATEWAY_ID variable is a debug only setting that prevents a node from attempting to connect to the AP so that the mesh topology can be tested within range of an AP.
Every node must connect at least once to the AP to get its configuration (or the filesystem must be pre-populated) since the list of valid nodes is stored on the mqtt broker.  Once it has connected once, the next time it boots, it will connect to the nearest mesh node.

C-monC, 2018-02-11:
Ah I see, thank you for the quick reply.

closed 2018-02-11 by C-monC