PhracturedBlue/ESP8266MQTTMesh #37: Mesh not wokring

C-monC, 2018-02-15:
-ESP8266MQTTMesh version: 0.8.8
-AsyncMQTTClient version: 0.8.2
-ESPAsyncTCP version: 1.0.1
-ESP8266 Core version: 2.4
-Using Platform.io
-Using the HelloWorld example

When I connected all the esp's to the same AP as my broker, everything works and I receive messages from all the esp's. 

However, when I set GATEWAY_ID in my platformio.ini file to test with the AP still in range, the esp's still connect to the AP. I have connected all the esp's to the broker at least once, and have restarted them after connecting them to the broker on the AP with the GATEWAY_ID already set, but they continue connecting to the AP directly.

I've added GATEWAY_ID to my platformio.ini file, as shown below:

```c++
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino

GATEWAY_ID = 2862868
```

Am I doing something wrong in my implementation?

PhracturedBlue, 2018-02-15:
you need t use a line like this in your platformio.ini:
build_flags = -DGATEWAY_ID=2862868

C-monC, 2018-02-17:
Thank you for the quick response. Unfortunately I am still not having any success.

I have our broker and GATEWAY_ID node (tried with and without the build_flags option enabled on this gateway node) connected directly to the AP and running, then I flashed the other esp's without the GATEWAY_ID flag enabled (to register it with the broker at least once) and everything works. But after flashing the nodes with the GATEWAY_ID flag enabled (all nodes except the gateway node), then I stop receiving messages.  

When I flash all of the nodes with the same code (GATEWAY_ID flag enabled), then only the gateway node connects to the broker.

My updated platformio.ini looks like this for the non-gateway nodes:
```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags = -DGATEWAY_ID=2862868
``` 

All esp's have been registered with the broker at least once and they're all within range of one another. Is it maybe the fact that I am re-flashing the nodes after their initial connection to the broker? 

I am using the chipID that is sent with the messages in the HelloWorld example (`String ID  = String(ESP.getChipId());`) to identify the gateway node.

PhracturedBlue, 2018-02-17:
I don't see anything wrong with your setup.
You can flash the same firmware onto the gateway node too (that is the whole point of GATEWAY_ID)  only the node with GATEWAY_ID should connect to the broker, the others should connect to the gateway node (or each other)

To ensure that the issue isn't the GATEWAY_ID parameter remove the GATEWAY_ID define and take one of your nodes and change the wifi network name to a non-existent name.  That should force the node to connect to the mesh (since it can no longer see the wifi-AP).  Make sure that node can also communicate with the broker.

Reflashing does not wipe the file-system, so it should not cause any issue as long as the node has registered with the broker at least once.

If this still doesn't work, connect a serial cable to one of the non-gateway nodes and send a log of what is going on.

C-monC, 2018-02-23:
I couldn't get it working, so I'm attaching log file retrieved after connecting a serial cable to specific nodes during my attempts.

I again flashed all the nodes without the GATEWAY_ID flag so that they could all connect to the broker and get registered (as seen in the `gateway_disabled.txt` log). 
Then I flashed all the nodes with the GATEWAY_ID flag enabled, and as before, only the GATEWAY_ID node connected with the broker, while the other nodes fail to find the GATEWAY_ID node AP and continue scanning endlessly (as seen in the `gateway_disabled.txt` log).

I have also added the GATEWAY_ID nodes' log for extra measure (the `gateway_node.txt)` log. I have not yet inspected the source code, but from the log file of the gateway node, it looks like the mesh AP created is not named with the node chipID. I'm not sure if this is the reason why the non-gateway nodes can't find the GATEWAY_ID nodes' AP.

changed_ssid.txt:
- https://pastebin.com/dqDvPp7h 

gateway_disabled.txt:
- https://pastebin.com/Lh0NbKyx

gateway_node.txt:
 - https://pastebin.com/EBrJ8nZz

gateway_enabled.txt
- https://pastebin.com/7u4vmE1Y

Thank you for your help.

PhracturedBlue, 2018-02-26:
your changed_ssid.txt and gateway_enabled.txt both show that the gateway AP could not be seen.  I don't know why that would be the case, since it is seen in the gateway_disabled.txt example so clearly the gateway is broadcasting at least sometimes.

In gateway_disabled.txt I do see that it attempted to connect to the gateway after failing to connect to the AP.  The return code was '201' which is 'WIFI_DISCONNECT_REASON_NO_AP_FOUND' which implies the gateway wasn't ready yet or wasn't configured properly.

Your gateway_node.txt shows the gateway AP is intitalized, so things look ok from that side.

I would be interested to see gateway_enabled configuration retried a couple times w/logs to see if the gateway is ever identified.  To be fair, I haven't run a full set of tests since 2.4.0 was released, maybe something wrong there.  I probably wont' have time to look at this right away, but I'll try running some tests on my side as soon as I have a chance.

PhracturedBlue, 2018-03-04:
Note that if you continue trying to work in this, you should upgrade to the new 1.0 code branch.  The mesh no longer requires connecting to the broker once and nodes no longer store any info about other nodes.  I am not sure if your current issues will go away, but further debug should be done on the latest code stream.

closed 2018-03-04 by PhracturedBlue