PhracturedBlue/ESP8266MQTTMesh #38: [request] Integrate WifiManager

kvn1351, 2018-02-19:
Hi @PhracturedBlue, would it be possible to integrate https://github.com/tzapu/WiFiManager into this?

It would be awesome if we could set the wifi credentials on one device and let it share those over the mesh to all other clients.

I've been looking into it a bit, but this fairly outside my comfort zone - being a mobile/fullstack dev.

Thanks a lot in advance!

alexiszecharies, 2018-02-19:
Hi @kvn1351 . We had the same idea. It is pretty staightforward. I opened a this issue https://github.com/PhracturedBlue/ESP8266MQTTMesh/issues/29. 

If @PhracturedBlue wants to I could contribute with a example.

PhracturedBlue, 2018-02-19:
I'm always open to more examples/documentation

kvn1351, 2018-02-19:
@alexiszecharies @PhracturedBlue I'd have to add some logic to share the ssid though. What would be the most elegant solution?

kvn1351, 2018-02-19:
@PhracturedBlue I personally think that it would make much more sense that only one node needs to have its wifi credentials set. What's your take on it?

alexiszecharies, 2018-02-19:
Our solution was the following. We have a flask server in a RPi which also has mosquitto running. This is the central entity Which knows all the data of the mesh. On a first installation the mobile device has a interace to connect to the RPi by writting the RPi local Pi and by doing this stores the data (mesh data included) in localstorage.  Another interface makes the device first configuration. When the client clicks there, the device starts to send ajax requests with the MQTT network SSID an password, and the interface tells the client to switch networks to the esp generated network. When the esp recives the ajax it stores the mesh data in SPIFFS and runs the loop.

alexiszecharies, 2018-02-19:
Thinking about your idea, the nodes that do not have the credentials need at least the password the ESPs use to communicate between each other. Nevertheless, the nodes need to have been connected to the broker at least one time to the broker in order to communicate between each other as @PhracturedBlue has stated in another issue. This would need to be changed so that your idea could be possible. I think there is another library called Painlessmesh which has this feature.

kvn1351, 2018-02-19:
That's actually an MQTT requirement. All clients have to connect at least once to receive updates from the broker (IIRC). So I guess that the only way of achieving this would be by first using HTTP to set the ssid.

PhracturedBlue, 2018-02-20:
I've been thinking about an alternative implementation of the mesh where every node has the same (hidden) id (rather than having each node have a unique id as provided by the broker).  It would be slightly less secure, but it would mean that nodes could self-assemble without 1st connecting to the broker.

The big problem is that it is a large rewrite of the connection protocol and I don't really have the time for it right now.

PhracturedBlue, 2018-03-04:
I have re-implemented the mesh connection algorithm such that it no longer requires pre-connecting to the broker once.  Nodes no longer store info about other nodes, and can locate other mesh nodes on their own.

closed 2018-03-04 by PhracturedBlue