PhracturedBlue/ESP8266MQTTMesh #39: v1.0.0

shajek, 2018-03-03:
Congratulation to 1.0.0 version :)

I see lot of rewrite, not only about SPIFFS but about identification and mesh comunication. 
Can you explain in short how it work for now ?
i see, that only broker store IDs.
but what happend when i start node without wifi range in mesh ?
what happend when i change ID of node on broker ? (i last version Node adopt new ID and instantly change it).
Can you please write all new stuff ? or what you change :)

thank you for your work

PhracturedBlue, 2018-03-04:
1.0.0 is not actually released yet.  I am still doing testing for it.  When it is ready, I will update the documentation and create a release so it is available on platformio

PhracturedBlue, 2018-03-04:
grr...I see that platformio actually detected the change.  I thought they waited for the actual tag.  That is annoying.  The current 1.0.0 is still not functional.

PhracturedBlue, 2018-03-04:
I just finished the initial verification and bumped the rev to 1.0.1.  The documentation is updated.  If you find any of it unclear please let me know.

The storage of the node info on the broker is now only for reference and is no longer used by the mesh ( maybe I'll add a configuration option to disable the RETAIN argument).  The nodes will now self-assemble even if they have never connected to the broker before.  Also note that SPIFFS is no longer included as there is no need for it in the code anymore.

closed 2018-03-04 by PhracturedBlue