PhracturedBlue/ESP8266MQTTMesh #41: WDT reset on wifi connect

C-monC, 2018-03-06:
- ESP8266MQTTMesh version: 1.0.0
- ESP8266 12-E
- On platformio

When the setup up finishes and the esp8266 matches the SSID the hardware wdt reset is activated. I have implemented ESP.wdtDisable() (in setup) to give it more time to connect but still get the same problem.

Serial monitor output: https://pastebin.com/TxpqwVqJ

I have tried multiple AP's and double checked the login credentials but get the same error for all.

PhracturedBlue, 2018-03-06:
Does this happen with 1.0.2?  1.0.0 was never meant to be released, and it was quite broken.

C-monC, 2018-03-06:
Sorry, I meant 1.0.2( just re-downloaded files to be sure). I noticed even with the AP off the esp still returns 
```[match_networks] Comparing pi``` and if I boot the pi nothing changes. 

I have also tried with my home network as the AP and my pc as the MQTT broker

PhracturedBlue, 2018-03-06:
I assume you are using the helloworld example?  please post your credentials.h

C-monC, 2018-03-07:
Yes, I'm using the helloworld example

credentials.h: https://pastebin.com/xF29kZGU

PhracturedBlue, 2018-03-07:
Your network list needs to have a null element at the end

C-monC, 2018-03-07:
Thank you for your help. That was the problem

closed 2018-03-07 by C-monC

PhracturedBlue, 2018-03-07:
I should probably use a more C++ way of defining those so you don't need the null element like a std::vector, or maybe even just add it into the builder.  I'll think about it