PhracturedBlue/ESP8266MQTTMesh #42: Error in send_ota.py

shajek, 2018-03-17:
In your latest send_ota.py is error when i run it with Python 3.6.3

```
Updating firmware on the following nodes:
        805de
Traceback (most recent call last):
  File "ota.py", line 201, in <module>
    main()
  File "ota.py", line 197, in main
    send_firmware(client, data, [args.node] if args.node else [])
  File "ota.py", line 97, in send_firmware
    client.publish("{}{}".format(send_topic, str(pos)), b64d)
  File "D:\WinPython-64bit-3.6.3.0Qt5\python-3.6.3.amd64\lib\site-packages\paho\
mqtt\client.py", line 871, in publish
    raise TypeError('payload must be a string, bytearray, int, float or None.')
TypeError: payload must be a string, bytearray, int, float or None.
```

when i in send firmware function change in client.publish b64d to  b64d.decode(), it will stuck on 
```
Updating firmware on the following nodes:
	805de
1 node(s) missed the message, retrying
1 node(s) missed the message and no retires left
1 node(s) missed the message and no retires left
```
i have latest version of library on that device wich i update

shajek, 2018-03-17:
finally is error that b64d must be decoded. I try it modified with this fix, and it works on Python 3.6.3

PhracturedBlue, 2018-03-17:
That is weird.  I would expect an implicit conversion from a bytes object to a bytearray.  I also don't see this issue on my system using 3.6.3 in linux.

Does it also work with this:
b64d = bytearray(base64.b64encode(d))

shajek, 2018-03-17:
```
# -*- coding: utf-8 -*-
#!/usr/bin/python3

from threading import Thread
import tkinter as tk
from tkinter import ttk
from tkinter import filedialog
import paho.mqtt.client as mqtt
import os
import sys
import argparse
import datetime
import hashlib
import base64
import time
import ssl
import re
import queue

class GUI(Thread):

    def __init__(self):
        super(GUI, self).__init__()
        self.start()
        
    def run(self):
        self.client = mqtt.Client()
        self.q = queue.Queue();
        self.data = None
        self.send_topic = None
        self.master = tk.Tk()
        self.master.title("OTA upgrader")
        self.master.resizable(False, False)
        self.server = dict()
        self.port = dict()
        self.user = dict()
        self.password = dict()
        self.ssl = dict()
        self.topic = dict()
        self.intopic = dict()
        self.outtopic = dict()
        self.node = dict()
        self.firmware = dict()
        self.file = dict()

        self.frame = tk.Frame(self.master)
        self.frame.grid(row=0, column=0)
        self.textview = tk.Text(self.frame)
        self.textview.grid(row=0, column=0, columnspan=2, rowspan=12, sticky=tk.W+tk.E+tk.N+tk.S, padx=5, pady=5)

        self.server["var"] = tk.StringVar()
        self.server["text"] = tk.Label(self.frame, text="server")
        self.server["text"].grid(row=0, column=2, padx=5, sticky=tk.E)
        self.server["entry"] = tk.Entry(self.frame, textvariable=self.server["var"])
        self.server["entry"].grid(row=0, column=3, padx=5)
        self.server["value"] = str()

        self.port["var"] = tk.StringVar()
        self.port["text"] = tk.Label(self.frame, text="port")
        self.port["text"].grid(row=0, column=4, padx=5, sticky=tk.E)
        self.port["entry"] = tk.Entry(self.frame, textvariable=self.port["var"])
        self.port["entry"].grid(row=0, column=5, padx=5)
        self.port["value"] = int()

        self.user["var"] = tk.StringVar()
        self.user["text"] = tk.Label(self.frame, text="username")
        self.user["text"].grid(row=1, column=2, padx=5, sticky=tk.E)
        self.user["entry"] = tk.Entry(self.frame, textvariable=self.user["var"])
        self.user["entry"].grid(row=1, column=3, padx=5)
[... 295 lines cut ...]
        if self.radio.get():
            self.send_topic = "{}/ota/{}/".format(self.intopic["value"], self.firmware["var"].get()) #firmware
        else:
            node_number=self.node["list_id"][self.node["var"].get()]
            print (node_number)
            self.send_topic = "{}/ota/{}/".format(self.intopic["value"], node_number) #node
            
        print(self.send_topic)
        self.client.loop_start()
        self.master.update_idletasks()
        self.send_firmware([node_number] if node_number else [])
        print("succesfull send firmware send")        

################################################################
#######################  OTA upgrader   ########################
########################### shajek ########################
#######################      2018       ########################
################################################################
if __name__ == "__main__":
    apl = GUI()

```
This is first draft, for now i dont handle some exceptions around input parameters, but if you enter host, optional username and password and topic (i have standard esp8266-), and click connect, if is all succesfull it will fill up combobox when you click of node radiobutton (or click on firmware and then on node and choose one), then you select your firmware bin file and then click Send to device
and here is problem, i successful send erase command, but then i stuck on "wait for" commmand, as you see i add few prints and i see that when i am in wait for i dont get any message
can you look at it ? it is basically dressed up your code

shajek, 2018-03-18:
```
# -*- coding: utf-8 -*-
#!/usr/bin/python3

from threading import Thread
import tkinter as tk
from tkinter import ttk
from tkinter import filedialog
import paho.mqtt.client as mqtt
import base64, hashlib, queue, re, time, ssl

class GUI(Thread):
    def __init__(self):
        super(GUI, self).__init__()
        self.start()
        
    def run(self):
        self.master = tk.Tk()
        self.master.title("OTA upgrader")
        self.master.resizable(False, False)
        self.server = dict()
        self.port = dict()
        self.user = dict()
        self.password = dict()
        self.ssl = dict()
        self.topic = dict()
        self.intopic = dict()
        self.outtopic = dict()
        self.node = dict()
        self.firmware = dict()
        self.file = dict()
        self.frame = tk.Frame(self.master)
        self.frame.grid(row=0, column=0)
        self.textview = tk.Text(self.frame)
        self.textview.grid(row=0, column=0, columnspan=2, rowspan=12, sticky=tk.W+tk.E+tk.N+tk.S, padx=5, pady=5)
        self.server["var"] = tk.StringVar()
        self.server["text"] = tk.Label(self.frame, text="server")
        self.server["text"].grid(row=0, column=2, padx=5, sticky=tk.E)
        self.server["entry"] = tk.Entry(self.frame, textvariable=self.server["var"])
        self.server["entry"].grid(row=0, column=3, padx=5)
        self.server["value"] = str()
        self.port["var"] = tk.StringVar()
        self.port["text"] = tk.Label(self.frame, text="port")
        self.port["text"].grid(row=0, column=4, padx=5, sticky=tk.E)
        self.port["entry"] = tk.Entry(self.frame, textvariable=self.port["var"])
        self.port["entry"].grid(row=0, column=5, padx=5)
        self.port["value"] = int()
        self.user["var"] = tk.StringVar()
        self.user["text"] = tk.Label(self.frame, text="username")
        self.user["text"].grid(row=1, column=2, padx=5, sticky=tk.E)
        self.user["entry"] = tk.Entry(self.frame, textvariable=self.user["var"])
        self.user["entry"].grid(row=1, column=3, padx=5)
        self.user["value"] = str()
        self.password["var"] = tk.StringVar()
        self.password["text"] = tk.Label(self.frame, text="password")
        self.password["text"].grid(row=1, column=4, padx=5, sticky=tk.E)
        self.password["entry"] = tk.Entry(self.frame, show="*", textvariable=self.password["var"])
        self.password["entry"].grid(row=1, column=5, padx=5)
        self.password["value"] = str()
        self.ssl["var"] = tk.IntVar()
        self.ssl["check"] = tk.Checkbutton(self.frame, text="SSL", variable=self.ssl["var"])
        self.ssl["check"].grid(row=2, column=2, padx=5)
        self.topic["var"] = tk.StringVar()
        self.topic["text"] = tk.Label(self.frame, text="topic")
        self.topic["text"].grid(row=3, column=2, padx=5, sticky=tk.E)
        self.topic["entry"] = tk.Entry(self.frame, textvariable=self.topic["var"])
        self.topic["entry"].grid(row=3, column=3, columnspan=3, padx=5, pady=5, sticky=tk.W+tk.E)
        self.topic["value"] = str()
        self.intopic["var"] = tk.StringVar()
        self.intopic["text"] = tk.Label(self.frame, text="intopic")
[... 268 lines cut ...]
            err = True
        if seen[node][2] != b'MD5 Passed':
            print("Node {} did not pass final MD5 check: {}".format(node, seen[node][2]))
            err = True
    if err:
        return
    print("Checksum verified. Flashing and rebooting now...")
    client.publish("{}flash".format(send_topic), "")

################################################################
#######################  OTA upgrader   ########################
####################### Stanislav Hajek ########################
#######################      2018       ########################
################################################################
if __name__ == "__main__":
    global client, data, send_topic, topic, inTopic, outTopic
    client, q, data, send_topic, inTopic, outTopic = mqtt.Client(), queue.Queue(), None, None, None, None
    apl = GUI()
    
     
```

Hello, me again. It is definitely wierd problem in mqtt paho and your functions. special to wait_for where blocks any kind of comunicatiom. I for now use you untouched code and i fill it with GUI from that prints you can see where it stuck. I will upgrade graphics, but for now doesnt work intention of this whole thing. I see that you prepare some nodes array, maybe for selective node update ? I will add popup windows where would be tickboxes how many we have bssid retain messages on broker. 
Please look at it and write some notes or decision or whatever :)
Thank you

shajek, 2018-03-18:
```
# -*- coding: utf-8 -*-
#!/usr/bin/python3

from threading import Thread
import tkinter as tk
from tkinter import ttk
from tkinter import filedialog
import paho.mqtt.client as mqtt
import base64, hashlib, queue, re, time, ssl

class GUI(Thread):
    def __init__(self):
        super(GUI, self).__init__()
        self.start()
        
    def run(self):
        self.master = tk.Tk()
        self.master.title("OTA upgrader")
        self.master.resizable(False, False)
        self.server = dict()
        self.port = dict()
        self.user = dict()
        self.password = dict()
        self.ssl = dict()
        self.topic = dict()
        self.intopic = dict()
        self.outtopic = dict()
        self.node = dict()
        self.firmware = dict()
        self.file = dict()
        self.frame = tk.Frame(self.master)
        self.frame.grid(row=0, column=0)
        self.textview = tk.Text(self.frame)
        self.textview.grid(row=0, column=0, columnspan=2, rowspan=12, sticky=tk.W+tk.E+tk.N+tk.S, padx=5, pady=5)
        self.server["var"] = tk.StringVar()
        self.server["text"] = tk.Label(self.frame, text="server")
        self.server["text"].grid(row=0, column=2, padx=5, sticky=tk.E)
        self.server["entry"] = tk.Entry(self.frame, textvariable=self.server["var"])
        self.server["entry"].grid(row=0, column=3, padx=5)
        self.server["value"] = str()
        self.port["var"] = tk.StringVar()
        self.port["text"] = tk.Label(self.frame, text="port")
        self.port["text"].grid(row=0, column=4, padx=5, sticky=tk.E)
        self.port["entry"] = tk.Entry(self.frame, textvariable=self.port["var"])
        self.port["entry"].grid(row=0, column=5, padx=5)
        self.port["value"] = int()
        self.user["var"] = tk.StringVar()
        self.user["text"] = tk.Label(self.frame, text="username")
        self.user["text"].grid(row=1, column=2, padx=5, sticky=tk.E)
        self.user["entry"] = tk.Entry(self.frame, textvariable=self.user["var"])
        self.user["entry"].grid(row=1, column=3, padx=5)
        self.user["value"] = str()
        self.password["var"] = tk.StringVar()
        self.password["text"] = tk.Label(self.frame, text="password")
        self.password["text"].grid(row=1, column=4, padx=5, sticky=tk.E)
        self.password["entry"] = tk.Entry(self.frame, show="*", textvariable=self.password["var"])
        self.password["entry"].grid(row=1, column=5, padx=5)
        self.password["value"] = str()
        self.ssl["var"] = tk.IntVar()
        self.ssl["check"] = tk.Checkbutton(self.frame, text="SSL", variable=self.ssl["var"])
        self.ssl["check"].grid(row=2, column=2, padx=5)
        self.topic["var"] = tk.StringVar()
        self.topic["text"] = tk.Label(self.frame, text="topic")
        self.topic["text"].grid(row=3, column=2, padx=5, sticky=tk.E)
        self.topic["entry"] = tk.Entry(self.frame, textvariable=self.topic["var"])
        self.topic["entry"].grid(row=3, column=3, columnspan=3, padx=5, pady=5, sticky=tk.W+tk.E)
        self.topic["value"] = str()
        self.intopic["var"] = tk.StringVar()
        self.intopic["text"] = tk.Label(self.frame, text="intopic")
[... 269 lines cut ...]
        if seen[node][2] != b'MD5 Passed':
            print("Node {} did not pass final MD5 check: {}".format(node, seen[node][2]))
            err = True
    if err:
        return
    print("Checksum verified. Flashing and rebooting now...")
    client.publish("{}flash".format(send_topic), "")

################################################################
#######################  OTA upgrader   ########################
####################### Stanislav Hajek ########################
#######################      2018       ########################
################################################################
if __name__ == "__main__":
    global client, data, send_topic, topic, inTopic, outTopic
    client, q, data, send_topic, inTopic, outTopic = mqtt.Client(), queue.Queue(), None, None, None, None
    apl = GUI()
    
      
     
```

Hello, me again. It is definitely wierd problem in mqtt paho and your functions. special to wait_for where blocks any kind of comunicatiom. I for now use you untouched code and i fill it with GUI from that prints you can see where it stuck. I will upgrade graphics, but for now doesnt work intention of this whole thing. I see that you prepare some nodes array, maybe for selective node update ? I will add popup windows where would be tickboxes how many we have bssid retain messages on broker. 
Please look at it and write some notes or decision or whatever :)
Thank you

PhracturedBlue, 2018-03-18:
I won't have time to actually try it for a little while, but I don't think using the sendbin function is a great way to do this for an interactive gui.  really the send_firmware should either be asynchronous, or at least be sent to its own thread and then use a queue to send messages back to the GUI about progress.