PhracturedBlue/ESP8266MQTTMesh #43: Subscribe to topic.

austonpramodh, 2018-03-22:
Make sure you have the latest version of ESP8266MQTTMesh before reporting an issue.

### Please Provide the following information:
- ESP8266MQTTMesh version:
- AsyncMQTTClient version:
- ESPAsyncTCP version:
- ESP8266 Core version:

**If you are you using platformio or Arduino, which one?
I am Using Arduino.
**Description of problem:**
How do I subscribe to a topic and use call back function? Please Help

PhracturedBlue, 2018-03-22:
You can't subscribe to an arbitrary topic.  The default topic that is subscribed to is 'esp8266-in/' though this can be changed during initialization with the setTopic() build helper.
The specified callback will be called whenever a topic of 'esp8266-in/<nodeid>/' is received (or if 'esp8266-in/broadcast' is received)

<nodeid> is the node's chipid in hex (not zero padded)..  ie if your chipid was 123456 the expected topic would be 'esp8266-in/1e240/'

austonpramodh, 2018-03-22:
Thanks, man... I check everything and it worked... Thank You.

austonpramodh, 2018-03-22:
Hey, Is there any way to remove the prefix of subscribing and publish? or at least remove the chipID leaving only ESP8266-out? so that I can inform broker that this device is connected.

PhracturedBlue, 2018-03-23:
Each node will leave a retained message on the broker notifying that it is available.
The only way that nodes are addressable is either by the chipid or via 'broadcast'.  Nodes can send retain messages through the publish API, but they will always contain the chipid

closed 2018-03-27 by PhracturedBlue

austonpramodh, 2018-03-27:
Hey, M sorry I forgot to thank you. And I figured out how to change the
topics. thank you so much...

austonpramodh, 2020-11-02:
Hey,

I just signed the petition "IT Minister Shri Ravishankar Prasad: 90 days of
Notice period in IT and ITes is harming industry and needs to be 30 days."
and wanted to see if you could help by adding your name.

Our goal is to reach 7,500 signatures and we need more support. You can
read more and sign the petition here:

http://chng.it/X7zNsGY6SZ

Thanks!
Auston

austonpramodh, 2020-11-02:
Hi there,
Please ignore the last email you received from "austonpramodh@gmail.com".
It was a mistake. My sincere apologies.

Thank you,
-- 

Regards,
Auston Pramodh Barboza
www.auston.dev