PhracturedBlue/ESP8266MQTTMesh #44: Doubt in RSSI Sorting

jassi00713, 2018-03-27:
Make sure you have the latest version of ESP8266MQTTMesh before reporting an issue.

### Please Provide the following information:
- ESP8266MQTTMesh version:
- AsyncMQTTClient version:
- ESPAsyncTCP version:
- ESP8266 Core version:

**If you are you using platformio or Arduino, which one?**

**Description of problem:**

The following code has been taken from  scan() function.
I have doubts in the two statements marked **BOLDLY** .

//sort by RSSI
        ap_t *ap_last = NULL;
        for(ap_ptr = ap; ap_ptr != NULL; ap_last = ap_ptr, ap_ptr = ap_ptr->next) {
            if(network_idx >= 0) {
                //AP is Wifi AP
                if (ap_ptr->ssid_idx != NETWORK_MESH_NODE && rssi <= ap_ptr->rssi) {
                   continue;
                }
            } else {
                //AP is mesh node
                if (ap_ptr->ssid_idx != NETWORK_MESH_NODE || rssi <= ap_ptr->rssi) {
                   continue;
                }
            }
            **next_ap->next = ap_ptr->next;
            ap_ptr->next = next_ap;**
            break;
        }
        if (! ap_ptr) {
            //Didn't insert this AP, so add it to the end;
            next_ap->next = NULL;
            if (ap_last) {
                ap_last->next = next_ap;
            } else {
                ap = next_ap;                 // Itself is the only ap_t
            }
        }

Shouldn't these statements be:
next_ap->next = ap_ptr;
ap_last->next = next_ap;

PhracturedBlue, 2018-03-27:
It is a good find, but the fix isn't quite right since it doesn't work if we want to insert before the 1st element.  The new code should work properly though

closed 2018-03-27 by PhracturedBlue

jassi00713, 2018-03-30:
No problem. You have done great work with this library.