PhracturedBlue/ESP8266MQTTMesh #47: Doubt regarding reducing socket timeout

jassi00713, 2018-03-30:
When i suddenly take out or switch off the mesh Gateway, the nodes still keep showing a valid connection for sometime.

How to reduce that time?
Keep Alive time or something like that needs to be reduced

PhracturedBlue, 2018-03-30:
Do you mean that the nodes don't detect that the gateway node is no longer present, and so don't try to find an alternate path?

This appears to be an issue with the ESP8266 core code, and I'm not sure it is fixable.  As per this message from a year ago, it doesn't seem to be configurable:
https://github.com/esp8266/Arduino/issues/3279#issuecomment-303388329

A workaround may be to send a 'ping' every so often and reset if it isn't returned.  If the 2.4.0 core didn't address this, that may be the only option.

jassi00713, 2018-03-31:
hmm
It seems that changing the keep alive time is not allowed in the SDK.
Well, we will have to settle with sending the pings which is not a bad option.