PhracturedBlue/ESP8266MQTTMesh #51: OTA not going through

benjaminraja, 2018-05-04:
Hi, I am trying to update a single node with 

`python send_ota.py --bin newfirmware.bin --node 30592d --id 0x1337`

Here is the error it throws. I am not sure if it is something to do with python; So, pardon me for my ignorance in python.  Besides, it seems to suggest that there is no response from the nodes, but there has been a response from the node, I've attached the same at the bottom of the message. Any help to get it to work highly appreciated. Thanks in advance.

File: newfirmware.bin
Sending to topic: esp8266-in/ota/0x1337/
Listening to topic: esp8266-out
Connected with result code 0
Erasing...
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/paho_mqtt-1.3.1-py2.7.egg/paho/mqtt/client.py", line 2650, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/local/lib/python2.7/dist-packages/paho_mqtt-1.3.1-py2.7.egg/paho/mqtt/client.py", line 1481, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/usr/local/lib/python2.7/dist-packages/paho_mqtt-1.3.1-py2.7.egg/paho/mqtt/client.py", line 1003, in loop
    rc = self.loop_read(max_packets)
  File "/usr/local/lib/python2.7/dist-packages/paho_mqtt-1.3.1-py2.7.egg/paho/mqtt/client.py", line 1284, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python2.7/dist-packages/paho_mqtt-1.3.1-py2.7.egg/paho/mqtt/client.py", line 1849, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python2.7/dist-packages/paho_mqtt-1.3.1-py2.7.egg/paho/mqtt/client.py", line 2305, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python2.7/dist-packages/paho_mqtt-1.3.1-py2.7.egg/paho/mqtt/client.py", line 2500, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python2.7/dist-packages/paho_mqtt-1.3.1-py2.7.egg/paho/mqtt/client.py", line 2647, in _handle_on_message
    self.on_message(self, self._userdata, message)
  File "send_ota.py", line 47, in on_message
    if regex(r'/([0-9a-f]+)/ota/erase$', msg.topic, match):
  File "send_ota.py", line 25, in regex
    group.clear()
AttributeError: 'list' object has no attribute 'clear'

1 node(s) missed the message
No nodes responded to erase.  Aborting

At the same time, the serial output on the node is as follows...
=============================================
[onMqttMessage] Message arrived [esp8266-in/ota/0x1337/start] 'md5:JC7prFWnpRz91dgR6U0T0A==,len:323456'
[handle_ota] OTA cmd 0x1337/start Length: 39
[handle_ota] OTA Start
[parse_ota_info] Key/Value: md5:JC7prFWnpRz91dgR6U0T0A==
[parse_ota_info] Key: md5 Value: JC7prFWnpRz91dgR6U0T0A==
[parse_ota_info] Key/Value: len:323456
[parse_ota_info] Key: len Value: 323456
[handle_ota] -> md5:JC7prFWnpRz91dgR6U0T0A==,len:323456
[handle_ota] Erasing 79 sectors
LmacRxBlk:0
LmacRxBlk:0
[erase_sector] Erase complete in 2.747650 seconds
[publish] Sending: esp8266-out/30592d/ota/erase=2747650

nikost999, 2018-05-16:
Hi
I have the same exact problem. I have 3 nodes, python script returns this:

> Sending to topic: esp8266-in/ota/0x1337/
Listening to topic: esp8266-out
Connected with result code 0
Erasing...
Updating firmware on the following nodes:
        2c7714
        165cef
        da3dbe
Got unexpected md5 for node 2c7714 at 0x0
         b'19a95c7c365a9b964ca6f9ccf065431b' (expected: b'fb5322bdc18b2f95b9ffb5f099f65672')

at the same time, the nodes report this in serial:

> [publish] Sending: esp8266-out/165cef/1465583=hello from 1465583 cnt: 112
[onMqttMessage] Message arrived [esp8266-in/ota/0x1337/start] 'md5:Z4PBGTiFgeuxyFVfX+nIEQ==,len:305456'
[handle_ota] OTA cmd 0x1337/start Length: 39
[handle_ota] OTA Start
[parse_ota_info] Key/Value: md5:Z4PBGTiFgeuxyFVfX+nIEQ==
[parse_ota_info] Key: md5 Value: Z4PBGTiFgeuxyFVfX+nIEQ==
[parse_ota_info] Key/Value: len:305456
[parse_ota_info] Key: len Value: 305456
[handle_ota] -> md5:Z4PBGTiFgeuxyFVfX+nIEQ==,len:305456
[handle_ota] Erasing 75 sectors
[erase_sector] Erase complete in 3.419353 seconds
[publish] Sending: esp8266-out/165cef/ota/erase=3419353
[publish] Sending: esp8266-out/165cef/1465583=hello from 1465583 cnt: 113
[publish] Sending: esp8266-out/165cef/1465583=hello from 1465583 cnt: 114
[onMqttMessage] Message arrived [esp8266-in/ota/0x1337/0] '6QECQJzyEEAA8BBAaAUAABAQAABQ9RBAHEsAQMwkAED///8///8PQP9/EED///9f8P//PwAQAAAc4gBAAEoAQExKAEAABwBgAAAAgOgrAEDwMABAoC8AQLcdwQQAEgBgABAA63wSAGASwdAJsfmh/QEpTzhPIeb/KiNLPwxEAeb/wAAAjFIMEoYHAAAAIeH/KQ8oDygCKS8oDygSKT84H4svAd7/wAAADAIdDwix+KESwTAN8AAAABLBwAnx+eH9ASmPKI8i0hApDzLPECgPQqAIAdH/wAAAjEIMEkYuAAAoD4siKQ8MAikfBiUADAIpbwwCKX8yzxgoDwyEAcf/wAAAjDIMIsYjACgPiyIpDyhvKT8MAiJPCCg/McH/JzMEDBIiTwgoPzG//yezDCg/Mb7/JzMEDBIiTwgoPzG8/yezBSKgASJPCDIPCAwSICMwICB0jJIofzgPKiMpD8YGADg/SH8oDwGs/8AAAIxCDDJGCQAAKH84DyojKQ8o'
[handle_ota] OTA cmd 0x1337/0 Length: 508
[handle_ota] Got 381 bytes FW @ 0
[handle_ota] Wrote 381 bytes in 0.004225 seconds
[publish] Sending: esp8266-out/165cef/ota/md5/0=19a95c7c365a9b964ca6f9ccf065431b
[onMqttMessage] Message arrived [esp8266-in/ota/0x1337/0] 'HxsiKR8iDxE4HyezAsbX/xGo/yhfPQKgAwAMAh0PCPH44RLBQA3wEsGwAmETwmES0mER4mEQ+fH9ASmPOZ9Jry0B7QIojyAgBMxiKJ8gIAQWQgAMEoYvACGX/yk/KD8LIilPKD+NAgwJgC0l0NkR0NIg0MgRKD9tAgwHYC0l0LcRsLIg0KYRKD/7IiAkQcAiESARwC0BLQIpXzivKD8qIwsyKD8gIGAgIxApDyiPKR8onykvBhYAKC84PwGA/8AAAAF//8AAAIxCDCIGEgAAOF8oH0g/AXH/wAAAjEIMMgYNAAA4XygvSD8Bdv/AAACMQgxCBggAADgfKD8qIykfOC8oPyojKS84Dyg/ICPAKQ8oD1YS+gwCHQ4dDwIhE8IhEtIhEeIhEPjxEsFQDfAAkqCgkBHAAmEn8mEmEPEgIqAJKQ8ioABF2v9LL8UZAMzShSYATAIBXv/AAACGBAAAfPIpLwwCIm8DIqB+AVn/wAAAKC9mElIioGMBVf/AAAAioHAB'
[handle_ota] OTA cmd 0x1337/0 Length: 516
[handle_ota] Got 387 bytes FW @ 0
[handle_ota] Wrote 387 bytes in 0.004095 seconds
[publish] Sending: esp8266-out/165cef/ota/md5/0=1d57808925ccb704f6294bc74c1cc75e
[publish] Sending: esp8266-out/165cef/1465583=hello from 1465583 cnt: 115

If its of any help, commenting out the MD5 check in the python script returns this:
> Sending to topic: esp8266-in/ota/0x1337/
Listening to topic: esp8266-out
Connected with result code 0
Erasing...
Updating firmware on the following nodes:
        2c7714
        165cef
        da3dbe
Got unexpected address 0x0 (expected: 0x768) from node 2c7714

nikost999, 2018-05-16:
Ok, I solved it myself.
It seems there's something wrong with Mosquitto on windows (or its config, although its a vanilla install).
On ubuntu + mosquitto it works fine.
It seems mosquitto on windows breaks up the message into segments of 381 + 387 bytes, instead of sending a single message of 768 bytes (as can be seen in the log above).

benjaminraja, 2018-05-17:
@nikost999, thanks for narrowing down to the root cause. I too just realised mosquito is indeed splitting messages. I am running it on a Pi. Since I have to use only Pi, I am now trying to change the broker to emqtt. Let me try and post the results. May be useful to someone else.

benjaminraja, 2018-05-17:
@PhracturedBlue Hi, I am still struggling to solve this issue. Any help would be greatly appreciated. OTA message from send_ota.py is split into 2 messages. I tried listening to topic "esp8266-in/ota/0x1337/#", the message is delivered as a single payload, but when it reaches the ESP node, it comes as 2 different messages, with each containing part of the original payload. Could you pls throw some light? Thanks much in advance.

benjaminraja, 2018-05-18:
Folks, When I compiled my code with **PlatformIO IDE** instead of **Arduino IDE**, everything got solved, now OTA works fantastically. Great job @PhracturedBlue .

closed 2018-05-18 by benjaminraja

c-, 2018-06-02:
I ran into this and decided to do some digging.

The root of the problem is that the onMqttMessage callback doesn't handle the index/total buffering as described in https://github.com/marvinroger/async-mqtt-client/issues/98

If you use the v2 Lower Memory lwIP variant, TCP/IP buffers are so small (mss 536) that the OTA chunks fragment. If you use the v2 Higher Bandwidth option (mss 1460) then it works fine. I assume that's what PlatformIO defaults to.

You *can* make onMqttMessage() handle fragmentation quite easily:

```
diff --git a/src/ESP8266MQTTMesh.cpp b/src/ESP8266MQTTMesh.cpp
index be659f4..a12ece8 100644
--- a/src/ESP8266MQTTMesh.cpp
+++ b/src/ESP8266MQTTMesh.cpp
@@ -1008,11 +1008,20 @@ void ESP8266MQTTMesh::onMqttUnsubscribe(uint16_t packetId) {
 }
 
 void ESP8266MQTTMesh::onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {
-  memcpy(inbuffer[0], payload, len);
-  inbuffer[0][len]= 0;
-  dbgPrintln(EMMDBG_MQTT_EXTRA, "Message arrived [" + String(topic) + "] '" + String(inbuffer[0]) + "'");
-  broadcast_message(topic, inbuffer[0]);
-  parse_message(topic, inbuffer[0]);
+  if( total >= MQTT_MAX_PACKET_SIZE ) {
+    dbgPrintln(EMMDBG_WIFI, "Ignoring oversize " + String(total) + " byte packet");
+	 return;
+  }
+  memcpy(&inbuffer[0][index], payload, len);
+  inbuffer[0][index+len]= 0;
+  if( index+len == total ) {
+	  dbgPrintln(EMMDBG_MQTT_EXTRA, "Message arrived [" + String(topic) + "] '" + String(inbuffer[0]) + "'");
+	  broadcast_message(topic, inbuffer[0]);
+	  parse_message(topic, inbuffer[0]);
+  } else {
+	  dbgPrintln(EMMDBG_MQTT_EXTRA, "Partial message [" + String(index+len)
+	  + " of " + String(total) + "]");
+  }
 }
 
 void ESP8266MQTTMesh::onMqttPublish(uint16_t packetId) {
```

*However* I don't know if it'll work in a true mesh situation because broadcast_message() then needs to get these "oversize" messages out, and I'm guessing that's not going to go well without more work.

A quick and dirty fix is to send smaller chunks. This will work with either lwIP, at the expense of way more packets.

```
diff --git a/utils/send_ota.py b/utils/send_ota.py
index 7d67011..4ff4923 100755
--- a/utils/send_ota.py
+++ b/utils/send_ota.py
@@ -90,9 +90,9 @@ def send_firmware(client, data, nodes):
     print("Updating firmware on the following nodes:\n\t{}".format("\n\t".join(nodes)))
     pos = 0
     while len(data):
-        d = data[0:768]
+        d = data[0:256]
         b64d = base64.b64encode(d)
-        data = data[768:]
+        data = data[256:]
         topic = "{}{}".format(send_topic, str(pos))
         client.publish(topic, b64d)
         expected_md5 = hashlib.md5(d).hexdigest().encode('utf-8')
@@ -120,7 +120,7 @@ def send_firmware(client, data, nodes):
                 print("\t {} (expected: {})".format(md5, expected_md5))
                 return
         pos += len(d)
-        if pos % (768 * 13) == 0:
+        if pos % (256 * 13) == 0:
             print("Transmitted %d bytes" % (pos))
     print("Completed send")
     client.publish("{}check".format(send_topic), "")
```

Another approach is just:

```
diff --git a/src/ESP8266MQTTMesh.h b/src/ESP8266MQTTMesh.h
index 58fb6bc..2e7ab3d 100644
--- a/src/ESP8266MQTTMesh.h
+++ b/src/ESP8266MQTTMesh.h
@@ -10,6 +10,10 @@
         #error "Must define MQTT_MAX_PACKET_SIZE >= 1152"
     #endif
     #define HAS_OTA 1
+
+        #if TCP_MSS < MQTT_MAX_PACKET_SIZE
+               #error "TCP MSS must be >1152 bytes. Use the Higher Bandwidth lwIP"
+        #endif
 #else
     #define HAS_OTA 0
 #endif
```

bfruiz04, 2019-05-18:
the solution is change in arduino tools settings v2 higher Bandwidth

Transmitted 359424 bytes
Transmitted 369408 bytes
Completed send
Checksum verified. Flashing and rebooting now...

simone1999, 2019-07-25:
just for everyone searching for it, to activate the higher Bandwith mode, you have to write following in your Platformio Config File:
build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH