PhracturedBlue/ESP8266MQTTMesh #53: Disconnect event.reason 201

stepanovdg, 2018-05-17:
Make sure you have the latest version of ESP8266MQTTMesh before reporting an issue.

### Please Provide the following information:
- ESP8266MQTTMesh version:1.0.4
- AsyncMQTTClient version:0.8.2
- ESPAsyncTCP version: 1.1.3 or 1.0.1
- ESP8266 Core version:

**If you are you using platformio or Arduino, which one?**
platformio

**Description of problem:**
Using example code - just configured mqtt server and networks. 
And esp32 proceed trying to connect to existing NETGEAR02 router
But always getting 201 reason. Such loop can last for an half an hour of retries or more.
Reboot, reset, reupload - do not help. 

Very similar to here https://github.com/espressif/esp-idf/issues/724

[schedule_connect] Scheduling reconnect for 0.50 seconds from now
[scan] Found: 13
[scan] Found SSID: 'NETGEAR02' BSSID '24:44:27:B5:C8:28' RSSI: -54
[match_networks] Comparing NETGEAR02
[match_networks] Matched
[scan] Found SSID: 'HUAWEI-QJ7k' BSSID '24:44:27:AF:E5:08' RSSI: -65
[match_networks] Comparing NETGEAR02
[match_networks] Comparing NETGEAR03
[match_networks] Comparing EPBYMINW3529
[scan] Did not match SSID list
[scan] Found SSID: 'HUAWEI-8ck4' BSSID '24:44:27:B5:FC:28' RSSI: -65
[match_networks] Comparing NETGEAR02
[match_networks] Comparing NETGEAR03
[match_networks] Comparing EPBYMINW3529
[scan] Did not match SSID list
[scan] Found SSID: '107' BSSID '70:72:3C:06:80:FC' RSSI: -67
[match_networks] Comparing NETGEAR02
[match_networks] Comparing NETGEAR03
[match_networks] Comparing EPBYMINW3529
[scan] Did not match SSID list
[scan] Found SSID: 'Krival' BSSID '24:44:27:B4:2C:54' RSSI: -77
[match_networks] Comparing NETGEAR02
[match_networks] Comparing NETGEAR03
[match_networks] Comparing EPBYMINW3529
[scan] Did not match SSID list
[scan] Found SSID: 'HUAWEI-pVu7' BSSID '24:44:27:99:CD:B0' RSSI: -81
[match_networks] Comparing NETGEAR02
[match_networks] Comparing NETGEAR03
[match_networks] Comparing EPBYMINW3529
[scan] Did not match SSID list
[scan] Found SSID: 'Wi-fi (90)' BSSID '48:D5:39:68:52:6C' RSSI: -84
[match_networks] Comparing NETGEAR02
[match_networks] Comparing NETGEAR03
[match_networks] Comparing EPBYMINW3529
[scan] Did not match SSID list
[scan] Found SSID: 'HUAWEI-7jtU' BSSID '24:44:27:B4:33:D8' RSSI: -88
[match_networks] Comparing NETGEAR02
[match_networks] Comparing NETGEAR03
[match_networks] Comparing EPBYMINW3529
[scan] Did not match SSID list
[scan] Found SSID: 'office_rat_f_off_guest' BSSID '02:7B:EF:60:35:2C' RSSI: -89
[match_networks] Comparing NETGEAR02
[match_networks] Comparing NETGEAR03
[match_networks] Comparing EPBYMINW3529
[scan] Did not match SSID list
[scan] Found SSID: 'office_rat_f_off' BSSID '10:7B:EF:60:35:2C' RSSI: -91
[match_networks] Comparing NETGEAR02
[match_networks] Comparing NETGEAR03
[match_networks] Comparing EPBYMINW3529
[scan] Did not match SSID list
[scan] Found SSID: 'MIH' BSSID 'AC:61:75:9E:3E:88' RSSI: -91
[match_networks] Comparing NETGEAR02
[match_networks] Comparing NETGEAR03
[match_networks] Comparing EPBYMINW3529
[scan] Did not match SSID list
[scan] Found SSID: 'HUAWEI-cG4f' BSSID '24:44:27:99:C6:0C' RSSI: -94
[match_networks] Comparing NETGEAR02
[match_networks] Comparing NETGEAR03
[match_networks] Comparing EPBYMINW3529
[scan] Did not match SSID list
[scan] Found SSID: 'HUAWEI-q5Jf' BSSID 'AC:61:75:85:11:AC' RSSI: -96
[match_networks] Comparing NETGEAR02
[match_networks] Comparing NETGEAR03
[match_networks] Comparing EPBYMINW3529
[scan] Did not match SSID list
[connect] 0 * 24:44:27:B5:C8:28 -54
[connect] Connecting to SSID : 'NETGEAR02' BSSID '24:44:27:B5:C8:28'
[onWifiDisconnect] Disconnected from Wi-Fi: NETGEAR02 because: 201
[schedule_connect] Scheduling reconnect for 5.00 seconds from now
[scan] Scanning for networks

stepanovdg, 2018-05-17:
Yes I also tried other networks (hotspot on androif phone and hotspot on fedora notebook). 
The same random time unwiiling to connect. After some time it succeds to connect. AP is working as I have already connected to the same network orange pi with mqtt server on it.

stepanovdg, 2018-05-17:
Also may be you can advise about OTA update - does any additional coding should be done to receive firmware.bin on nodes site. I tried HelloWorld example. And in debug I only got message (after using send_ota.py with correct arguments) [onMqttMessage] Message arrived [ESP32-in/ota/0x1337/start] 'md5:bbBaO0Ys5G+tq27rjgsoxQ==,len:586848' and that is all. Nothing begins to starting download or to start installing and reboot. Should I manually add callback for OTA?

simone1999, 2018-05-21:
For me OTA works perfectly fine, but 2 month ago or so we found a Bug in the OTA Python script, which should run now, but maybe there is still a little error. Are you Updating over the Node ID or the Node's Software ID (sry I don't know the exact names at the Moment)? Maybe just try the other way you tried.

simone1999, 2018-05-21:
it is called id for the Software Versions ID and node for the direct Node ID.

stepanovdg, 2018-05-21:
I think that I tried both ways. First was used without node name:
python3 send_ota.py --bin /esp32MqttOTA/.pioenvs/esp32dev/firmware.bin --id 0x1337 --broker 192.168.100.53 --user user --password password --intopic ESP32-in --outtopic ESP32-out 

Second was the same but with node id  there. As I could see from serial port - message about ota update received on my node - but has no effect (likle no callback called or smth). 

I used simple hello project. So as I understood OTA possibility should be included in library itself. Or I should register additional listener/callback for such kind of messages?

stepanovdg, 2018-05-21:
Thank you for help. Currently for me the most problem is 201 exception after already connected node go to reboot. After it takes about half an hour to connect again. Used different AP`s to check. All other devices connects fine. I googled such issue and seen some changes in esp-idf core for such problem. Just I wasn`t able to find place in ESP8266MQTTMesh library to check if has already fix or not.

stepanovdg, 2018-05-28:
#if  ! defined(ESP8266MESHMQTT_DISABLE_OTA) && ! defined(ESP32)
    //By default we support OTA
    #if ! defined(MQTT_MAX_PACKET_SIZE) || MQTT_MAX_PACKET_SIZE < (1024+128)
        #error "Must define MQTT_MAX_PACKET_SIZE >= 1152"
    #endif
    #define HAS_OTA 1
#else
    #define HAS_OTA 0
#endif 

Why OTA is disabled for ESP32?