PhracturedBlue/ESP8266MQTTMesh #56: ssl enabled mqtt crashing when a node connects to gateway

jassi00713, 2018-07-15:
I am using platformio
i have set the ASYNC_TCP_SSL_ENABLED Flag

Then i have kept the following settings in the main file:
MESH_SECURE      false
MQTT_SECURE      true

Hence, i am using ssl for MQTT only and not within the mesh.

I followed all the steps to start ssl mqtt and it worked.
My gateway makes a successful secure connection to my broker and all looks fine.

But when i turn on the other non-gateway nodes, they search the gateway ACCESS POINT and attempt connection.
As soon as i get onAPConnect on the gateway, the mqtt disconnects ie i get onMqttDisconnect callback.

Why is that??

without ssl everything works well

You have done a great work on this library. Kindly look into this issue.

jassi00713, 2018-07-16:
hey ... any updates on this issue?

PhracturedBlue, 2018-07-16:
I have been fighting with issues with the mesh nodes disconnecting with limited success.  I think this is related to issues running in combined AP/STA mode.  I have not noticed that it is specifically related to using SSL though.  With the current code, it is difficult to tell whether a connection is reset due to an MQTT disconnect or an AP disconnect.  Unfortunately I have not as yet made any significant progress with this issue.  It sounds like yours may be a little different (I've been having this issue without SSL).  I will try to reproduce, but I cannot say how quickly I'll be able to get to it

jassi00713, 2018-07-17:
i have no issue withoutssl

jassi00713, 2018-07-17:
i face it with ssl ... and i have done some debugging .... while ssl enabled .... on connection of a node .... the mqtt on gateway disconnects with reason 0 i.e. TCP loss

jassi00713, 2018-07-17:
code this be a RAM issue ? ... maybe because of ssl ... more data is in RAM ... the code size is certainly big when ssl enabled

jassi00713, 2018-07-17:
one big major point .... i need to correct myself ..... the mqtt connection doesn't go instantly .... when a node connects to the gateway .... everything is fine.

The MQTT is up, the gateway AP is up, the node is also connected to it and the tcp packets are going through perfectly between the gateway and the node.

Problem arises when after all this ... i send a mqtt publish or i do a mqtt subscribe ... it then crashes.

MQTT Publish and MQTT Subscribe works fine before a node connects to the gateway. But after a node connects, MQTT Publish or MQTT SUbscribe causes MQTT Disconnect.

Can you deduce something from this?

PhracturedBlue, 2018-07-17:
Now I don't understand.  a node does not subscribe to the mqtt broker, only the gateway can subscribe to messages, and this only happens when it 1st connects.  I do not believe it will even allow nodes to connect until it has subscribed to the broker.  Nodes can publish data of course, but I am unclear whether it is the leaf-node publishing or the gateway publishing (or both) that causes the disconnect