PhracturedBlue/ESP8266MQTTMesh #59: Gateway and mesh crash when internet goes offline

dlangamer, 2018-07-28:
- ESP8266MQTTMesh version: 1.0.4
- AsyncMQTTClient version: 0.8.2
- ESPAsyncTCP version: 1.1.0
- ESP8266 Core version: 2.4.1

IDE Plataformio 0.17.4

Hi!

I have noticed a strange problem in the mesh network. When the internet connection goes offline, the radio gateway crashes and restarts with wdt error. Some network nodes connected via mesh also restart with a similar error (gateway direct connected nodes).

Any idea how to solve this? Is it a known problem?

Log below

Thx

**Error on gateway when internet goes offline**
`[publish] Sending: esp8266-out/65670/415344=hello from 415344 cnt: 172
[onData] Got data from 192.168.26.100
[onMqttDisconnect] Disconnected from MQTT: 0
[onDisconnect] Disconnected from AP
[onDisconnect] Disconnected unknown client

Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x400326b0 depc=0x00000000

ctx: sys
sp: 3fffeab0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffec50:  3fff0eac 000005ab 000005ab 4010020c
3fffec60:  3fff2f54 00000005 3fffeca0 401006dc
3fffec70:  3fff2f54 00000005 3fff0cb0 4020a54c
3fffec80:  3ffe90a8 00000005 3ffef378 4020a568
3fffec90:  3fff3094 00000005 3ffef378 40208c11
3fffeca0:  00000000 00000000 00000000 00000000
3fffecb0:  3fff0eac 000001fa 000001fa 4010020c
3fffecc0:  3fff0ba8 3fff2f54 3fff2f94 401006dc
3fffecd0:  3fff0ba8 00000000 3fff2f54 4020a9ec
3fffece0:  3fff0ba8 00000000 3fff2f54 402063c2
3fffecf0:  00000000 3fff2f54 3fff2f54 4020c030
3fffed00:  3fff0eac 3fff2f54 3fff2f54 402030ad
3fffed10:  00000000 3fffed5c 00000000 402070a0
3fffed20:  3ffef2a8 00000000 3fff0cb0 4020a54c
3fffed30:  00000000 3fffed5c 3ffef2a8 3fff1d04
3fffed40:  00000000 3fffed5c 3ffef2a8 4020875c
3fffed50:  00000000 00000000 00000000 00000000
3fffed60:  00000000 00000000 00000000 00000000
3fffed70:  00000000 00000000 00000004 4020a990
3fffed80:  3fff1cf4 00000000 3ffef398 4020878a
3fffed90:  3ffef498 00000000 3ffef398 40204273
...`

**Error on a node direct connected to a Gateway:**
`[handle_client_data] --> 'esp8266-out/6eb3a/453434=hello from 453434 cnt: 14190'
[publish] Sending: esp8266-out/1c0e2c/1838636=hello from 1838636 cnt: 21
[onError] Got error on 0.0.0.0: -8
[onDisconnect] Disconnected from mesh
[onDisconnect] Disconnected from AP
[onDisconnect] Disconnected unknown client

Exception (3):
epc1=0x4010010e epc2=0x00000000 epc3=0x00000000 excvaddr=0x40007469 depc=0x00000000

ctx: sys 
sp: 3fffeb50 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffecf0:  3fff0eac 000002cf 000002cf 4010020c  
3fffed00:  3fff2d3c 00000005 3fffed40 401006dc  
3fffed10:  3fff2d3c 00000005 3fff0cb0 4020a54c  
3fffed20:  3ffe90a8 00000005 3ffef378 4020a568  
3fffed30:  3fff2c0c 00000005 3ffef378 40208c11  
3fffed40:  00000000 00000000 00000000 00000000  
3fffed50:  3fff0eac 00002bb2 00002bb2 4010020c  
3fffed60:  3fff0ba8 3fff2d3c 3fff2d7c 401006dc  
3fffed70:  3fff0ba8 00000000 3fff2d3c 4020a9ec  
3fffed80:  3fff0ba8 00000000 3fff2d3c 402063c2  
3fffed90:  3fff1d44 3fff2d3c 3fff2d3c 4020c030  
3fffeda0:  3fff0eac 3fff2d3c 3fff2d3c 402030ad  
3fffedb0:  3fff1d44 00000001 00000000 402070a0  
3fffedc0:  3ffef2a8 00000000 3fff0cb0 4020a54c  
3fffedd0:  3ffe9078 00000001 3ffef368 3ffef2a8  
3fffede0:  3fff1d44 00000001 3ffef368 40208b78  
3fffedf0:  00000000 00000000 00000000 00000000  
3fffee00:  00000000 00000000 00000000 00000000  
...`

simone1999, 2019-07-26:
first of all it is verry strange that the Internet connection has annything to do with the Network performing. Except you connect to a MQTT Server via the Internet and not over your local Network.
please provide how exactly the Mesh is connected to the MQTT Server, then I may be able to help you further.

dlangamer, 2019-07-26:
I was using an external mqtt server. I had noticed that when there is an internal server mqtt the nodes are stable. But when I use external mqtt server and by chance this connection is abruptly terminated (maybe a network cable disconnected from the router), the nodes start to generate these errors.

It is not every time. Sometimes the gateway node simply gives the disconnect message and tries to reconnect to mqtt server. But most of the time, it generates this error as soon as disconnection occurs ([onMqttDisconnect] Disconnected from MQTT: 0)

simone1999, 2019-07-28:
thanks,
first of all this is a VERRY nice and cload picture, thank you.
There's only one little thing wrong with the Picture, the Nodes themselves can't directly communicate with each other, the only way for them to communicate is to send a mqtt message over the gateway into the cloud mqtt broker and then from there the called node gets the Message back from the cloud MQTT Server.

And then the watchdog timer is used to restart the Node so it is clean. Like you can see in your logs the node disconnects from the wifi itself and disconnects every connected client. Afterwards it just completly restarts itself to ensure that the connection Process is as simple and clean as possible. Since every Connection is closed it is nearly no difference if the Node restarts or retries to connect to the WIFI and MQTT. the Watchdog Error accoured also in my System, it defently is not as great as it could be, but since it restarts afterwards this behaviour should not harm the System in anny way.
correct me if sth. is wrong.

have a verry nice day and hopefully lots of fun with the MESH!

wilson-wsn, 2020-07-04:
Hello everyone, I can compile the code on the Arduino IDE. But after recording the sketch, NodeMcu is stuck. The messages on the serial port are:

* ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024520f depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd70 end: 3fffffc0 offset: 0190
3fffff00:  3ffe8ceb 3fffff3c 4024520f 4020af7f  
3fffff10:  4020cb5d 00000010 3ffeeab8 40204d6d  
3fffff20:  0000001c 0001c200 00000000 00000000  
3fffff30:  007a1200 943e371f ffffff00 0000005b  
3fffff40:  401052c5 8104f40c 3fff0600 00000000  
3fffff50:  80fee4d0 3fff06c8 000003e8 40201a44  
3fffff60:  0000002f 3ffe866d 00000000 00000000  
3fffff70:  40201024 3fff06c8 3fff0668 3fff0668  
3fffff80:  3fffdad0 00000000 3ffeeab8 40201119  
3fffff90:  40201024 feefeffe 4020d4e8 4020d4d0  
3fffffa0:  feefeffe feefeffe 3fff0628 4020b810  
3fffffb0:  feefeffe feefeffe 3ffe851c 40100d09  
<<<stack<<<

Does anyone know where the error is?