PhracturedBlue/ESP8266MQTTMesh #64: Node Friendly Name and Is Node Alive?

hubshi, 2019-02-06:
Make sure you have the latest version of ESP8266MQTTMesh before reporting an issue.

### Please Provide the following information:
- ESP8266MQTTMesh version: 1.0.4
- AsyncMQTTClient version:  0.5.0
- ESPAsyncTCP version: 1.2.0
- ESP8266 Core version: 2.4.1

**If you are you using platformio or Arduino, which one?**
Arduino

**Description of problem:**

Thank you for this wonderful piece of software which helped me to build my dream network of home automation. I have 2 questions:

1. Is it possible to assign Nodes a friendly name like "Kitchen Temperature" etc. ? so instead of using the device id which become difficult sometimes to understand . 

2. Is it possible to track if a particular Node is Online or Not.

I am using Node-Red for routing and visualization and Internal Mosquito MQTT Broker.

Best Regards.

simone1999, 2019-07-26:
hi,
for your Name Issues i would suggest you to take Node Red and just "map" the device ID to a more meaningful Name, you could even only subscribe to the normal Nodes with your "mapper" Flow and resend the Packets with the newly mapped Names, and just subscribe to this retransmitted Packages with the rest of your Node Red Stuff.

by the Way i am quite interrested in what home Automation stuff you are doing with this System, how manny Nodes you have and if you are only using Node Red itself or other software together with Node Red as well.

simone1999, 2019-07-26:
oh, and my fork of this Project (will be largely updatet the next days) implements an ping option, so you can ping the Nodes and receive a msg back if it is reachable. I also used this in my Node Red and IOBroker Configuration to track when a node turns off , which sadly happens way to often at the moment for me, but hopefully will be fixed in the next days in my fork.
(Link to my Fork: [)](https://github.com/simone1999/ESP8266MQTTMesh)

eporocrail, 2019-07-27:
@simone19999
Would you be so kind to let us know when your fork is updated.
Where I am going to publish my model railroad system I would prefer to exclude that nodes of the network are closing down as much as possible.
Regards
Ellard

simone1999, 2019-07-28:
hi,
my fork is now updated, manny Stability Issues are solved now, and from my homenetwork i can ensure you that it helped a lot stability wise!
the ping feature was also added a long time ago, yust isn't documented yet (kind of all changes i made till now).
To use the Ping, you simply have to send the Node anny payload to its Topic with the subtopic "Ping" and it will respond i think with the outtopic and suptopic "Ping" and as a payload it has a 1. To track every node just send the "Ping" to "esp8266-in/broadcast/Ping" so every node receives it and answers. Then you can build routines (maybe in Node-Red) to track if every Node respondet to your ping.

Hope my Fork helps you, and most importantly:
have Fun!!!

eporocrail, 2019-07-28:
Hi,

Bootsequence looks much cleaner