PhracturedBlue/ESP8266MQTTMesh #67: Array ota_info.md5 out of boundaries on OTA when calling base64_decode

miseryofdevice, 2019-02-12:
Make sure you have the latest version of ESP8266MQTTMesh before reporting an issue.

### Please Provide the following information:
- ESP8266MQTTMesh version:
- AsyncMQTTClient version: 0.8.2
- ESPAsyncTCP version: 1.2.0
- ESP8266 Core version: commit 3745054f94f698555e8421334addeead1cdf8fbe

**If you are you using platformio or Arduino, which one?**
PlatformIO: version 4.0.0a3
/.platformio/platforms/espressif8266 commit: 7b360716161fa31175ba9faca7f4e7d6960a584c

**Description of problem:**
Probable overflow of `ota_info.md5` on 
https://github.com/PhracturedBlue/ESP8266MQTTMesh/blob/master/src/ESP8266MQTTMesh.cpp#L754
in function base64_decode(..)
https://github.com/PhracturedBlue/ESP8266MQTTMesh/blob/master/src/Base64.cpp#L59

I just tested it and, according to my tests, `decLen` goes up to 16. Thus, `ota_info.md5` goes out of it boundaries.

simone1999, 2019-07-26:
looks like you are absolutely right, I think the Problem is that the decoded message is in fact 16 byte long, which is absolutely right, but then at the End of the Function the String Terminator (\0) gets addet, which is the 17th byte and like you said this is overflowing.

simone1999, 2019-07-26:
one verry interesting thing is also, that the Function base64_decode also gets used to decode the whole Firmware sent over the OTA function. This means that every Block of Memmory has a String Terminator (B00000000) at the End, maybe the Instruction (B00000000) doesn't do anything (maybe the NOP Instruction) but it defently gets written in the Flash! I'm thinking about just removing the String Terminator in the base64_decode, because if you really want to use String Functions on the decoded CharArray, you could just add the Terminator after you called base64_decode. besides the only 2 times the Function is cadded is for decoding the MD5 sum and decoding the OTA binary packets. And base64_encode isn't called at all