PhracturedBlue/ESP8266MQTTMesh #69: Unique unsigned int value as Node id instead of chip-id.

eporocrail, 2019-03-19:
I managed to get the mesh working.

I would like to use the mesh as a carrier for the MQTT messages from and to the model railroad control software "Rocrail". To facilitate this Rocrail requires a "node id". This node id is an integer unique for each node.
My question is if you would be able and willing to implement a node-id with an unsigned int value as id instead of the chip id used in the existing library.

simone1999, 2019-03-22:
Hi, my Implementation for this Problem would be to not change the Libary, but just "map" the Messages from your adressing way to the libary Adressing way, you could do this for example verry easy with Node-Red. So you send from your rail Software, but the only subscriber is node red, which itself has an Function to map all Rail Adresses to the Mesh Id's and then sends the same msg it received, but on the new topic. If this sounds solid for you, but you don't know how to use node Red, and have no other comfortable way of dooing it, feel free to ask me, i can help you with this.

eporocrail, 2019-03-22:
Thanks for your reply. The mapping idea is very appealing. Never thought of that. I think Node-Red with Raspi is an overkill. I will start with a Wemos D1 mini module. If that proves to be not fast enough or something else, I have to look into Node-Red. I will keep you informed.

eporocrail, 2019-04-12:
As promised an update about the "mapping" idea.
I succeeded to create a gateway which is performing the mapping.
Messages going out of the mesh are routed to the gateway. The gateway has a translation table which is filled automatically during the boot process of the mesh and the gateway. No user intervention is required.
I also added time-stamping with milliseconds and message logging on SD card.
The processing time of the gateway is within 2 mSec.
The last step will be sending messages the other way around.

Have fun

simone1999, 2019-07-26:
Sry for not responding this long, I'm verry happy that it worked hope it now does for you, what you need it for!
have a nice Time building your railroad!

closed 2019-07-27 by eporocrail