PhracturedBlue/ESP8266MQTTMesh #71: Disconnect Reason: 8

kcbleeker, 2019-06-19:
Make sure you have the latest version of ESP8266MQTTMesh before reporting an issue.

### Please Provide the following information:
- ESP8266MQTTMesh version: 1.0.4
- AsyncMQTTClient version: 0.8.2
- ESPAsyncTCP version: 1.2.0
- ESP8266 Core version: 2.4.0-rc2

**If you are you using platformio or Arduino, which one?** 
Arduino

**Description of problem:**
I have only 2 nodes: 
- "Relay" node which is connecting to my WiFi AP
- Mesh node with no AP's configured

Everything works, *something* happens so the node with no access starts reporting disconnected from the other node with reason 8.

My "relay" node is a Wemos D1 Mini, the other is a NodeMCU 1.0

These are the closest similar issues I can find, but I have no idea what could be the cause. 

https://github.com/letscontrolit/ESPEasy/issues/1447
and https://github.com/PhracturedBlue/ESP8266MQTTMesh/issues/3

My temporary and awful solution is just for all my nodes to periodically restart, and then they seem able to connect again.

Any ideas?

kcbleeker, 2019-06-28:
@PhracturedBlue are you still maintaining this project?

simone1999, 2019-07-25:
hi, is it possible, that you use Delays in your Code? For me it seems like your relay Node is not able to respond in time to the message from your "Mesh node". This could be caused by using Delays or manny Interrupts in your Code. But that's only a guess.

kcbleeker, 2019-07-25:
Hi!

No delays, no. Just yesterday I added a "setErrorCallback" to the library, and I just check the error code and ESP.restart() if it's this disconnect reason. Annoying, but seems to allow it to connect again after that.

simone1999, 2019-07-28:
ok, in my fork of the Project i have fixed lots of stability issues (and also added verry little features).Maybe you can try out my Fork and report back if everything starts woring.
my Fork:
[ESP8266MQTTMesh Fork](https://github.com/simone1999/ESP8266MQTTMesh)

JungDev, 2019-12-30:
I have same issue, and I think that is caused when router changing channel. The mesh node lost connection and it can't establish again

JungDev, 2019-12-31:
> Hi!
> 
> No delays, no. Just yesterday I added a "setErrorCallback" to the library, and I just check the error code and ESP.restart() if it's this disconnect reason. Annoying, but seems to allow it to connect again after that.

@kcbleeker Do you can share this workaround, please?

simone1999, 2019-12-31:
> 
> 
> I have same issue, and I think that is caused when router changing channel. The mesh node lost connection and it can't establish again

sadly I also have this issue. but I'm quite sure it doesn't come from the Router changing the Channels... My router, or to be preceice wireless access point is configured to a fixed channel, so it shouldn't cause this issue.
Also I have a Network of around 15 ESP's where around half is connected to the Router directly and the other half is connected over the Mesh. the Nodes fail from time to time, but defently not all at once. In my fork of this Project I have added multiple Functions, also a ping Function. So I am pinging my whole Mesh system every 5 seconds and check if all the Nodes respond. the Result is visualized in a nice Graph via Iobroker and Node-Red. In this Graph I can see that sometimes all Nodes respont to every Ping for multiple Days and sometimes single Nodes are constantly crashing, reconnecting every 2 Minutes or so.

[image]
here it is quite nice and all the common drops are coming from me playing with the Ping System itself or restarting the router. Also the last 3 ESP's have other Problems, so don't look at them

[image]
here again the common Drops are from chanching the System itself, but the drops in the Blue Line and the Red Line are from the ESP restarting. the first half of the red line where the ESP is longer down than reachable should now be fixed in my Fork, but the later half of red and the blue ripples are still there sometimes.

I thought this behaviour could come from an Buffer Overflow in the Lib, so i rewrote the Lib to be more save from Buffer Overflows and this seemed to make it a bit more stable, but sadly not stable...

Also I send the reason why the ESP reconnected. Nearly everytime this jitter comes the ESP doesn't disconnect because a restart or a exception, but just loses it's connection and reconnects without restarting. But sadly from that I don't have a screenshot at the moment.

If you find some more Informations about this I would love to hear them and could implement them.