PhracturedBlue/ESP8266MQTTMesh #73: Reduce transmission level

eporocrail, 2019-08-02:
As you know I am building a switching and signalling system for model railroad.
Development is proceeding quite well.
On my test layout I am employing 14 decoders(network nodes) in a network. The aim is to have some 40 odd decoders in a single network. These decoders are transmitting all the time.
The decoders are placed in one room. 
Would it be possible to reduce transmission power to bring the total power requirement down as far as possible.

simone1999, 2019-08-02:
hm, to be honest I really don't know much about the ESP8266 Wifi. I defently know that you are able to controll the Transmission power, but I'm not sure if you could configure the WIFI Settings in your sketch or if you have to manipulate the wifi object itself, which the library uses internally.

simone1999, 2019-08-02:
oh, never mind look what i have found:
`void ESP8266WiFiGenericClass::setOutputPower(float dBm) {

if(dBm > 20.5) {
dBm = 20.5;
} else if(dBm < 0) {
dBm = 0;
}

uint8_t val = (dBm*4.0f);
system_phy_set_max_tpw(val);
} `

with this you should be able to just set the txpower by yourselve and since (as far as i saw) the Libary doesn't set this value mannualy it should be enougth to set this Value on Startup and that's it.

Jazzman2006, 2019-08-07:
Try this code:

#define WIFI_TX_POWER 20 // TX power of ESP module (0 -> 0.25dBm) (0...85)
#ifdef ESP8266
extern "C" {
#include "ets_sys.h"
#include "os_type.h"
#include "osapi.h"
#include "mem.h"
#include "user_interface.h"
#include "cont.h"
}
#endif

setup()
{
system_phy_set_max_tpw(WIFI_TX_POWER);
}

closed 2019-09-01 by eporocrail