PhracturedBlue/ESP8266MQTTMesh #75: add me as Collaborator

simone1999, 2019-09-05:
@PhracturedBlue 
hi, if you could add me as Collaborator, I could help manage this Repo a bit more.
I think afterwards i also can close Issues, which is annoying me a bit, because here are quite manny Issues, which I alredy answered, which aren't closed yet.
Also maybe we could link this Repo to my Fork, which at the moment is quite a bit more stable and has added some little Features to it (which i still have to document in the readme :sweat_smile: ).

the Collaborator function is under Settings/Collaborators.

also verry god work on this Project, I use it to automate my whole House with over 30 ESP8266's

PhracturedBlue, 2020-04-15:
Are you still interested in this?  I don't have resources to work on this project right now, but would be willing to take on some help to keep it moving.
The same offer goes for anyone else who has the skills to help.

bfruiz04, 2020-04-16:
sure we can help to you, this project is awesome,!!!

simone1999, 2020-04-21:
Hi, I'm sure willing. I also don't have that much time, but I think I could at least contribute a bit to this Project. I'm heavily using it in my own house, so I'm really interesstet in keeping it moving.

bfruiz04, 2020-04-22:
great @simone1999 , i am looking your repo with fork to it, i can watch many fixes, great job

Arijeet96, 2020-11-18:
@PhracturedBlue Hey, I am working on your program, I need to ask you how do you put 30 esp8266 in your home with controlling using painless mesh concept. can you show me which code u were using?it might be a great help for me.

alexmogavero, 2024-01-06:
Hi I could also give my contribution. I have added a pull request #84