PhracturedBlue/ESP8266MQTTMesh #78: Soft Coding: Getting build parameters over UART.

kazimyuksel, 2020-05-05:
- ESP8266MQTTMesh version: 1.0.4
- AsyncMQTTClient version: 0.8.2
- ESPAsyncTCP version: 1.2.2
- ESP8266 Core version: 2.4.0

**If you are you using platformio or Arduino, which one?**
Arduino

**Description of problem:**
I started feeling like i am spamming the issues panel but i really have a frustrating problem and i think couple people also encountered it therefore i feel obligated to ask.

I need a soft coded system where i can send SSID and PASS over UART (Bluetooth or Computer).
As far as i can see, library does not allow me to re-set values given in Builder constructer.

Is there a way to bypass the Builder? Or a more general question in const char really necessary for the library, can it work without it? If it can i can rewrite constructer in order to work with String input from UART. But if there is a currently available solution, it is much more appreciated.

Thank you!

kazimyuksel, 2020-05-05:
**SUMMARY**

How can initialize ESP8266MQTTMesh as global without any parameters and assign parameters at setup()

**or**

How can initialize ESP8266MQTTMesh as global with dummy parameters and change parameters at setup()

PhracturedBlue, 2020-05-05:
if you define the builder at runtime instead of at compile time, you should be able to use setMeshSSID and setMeshPassword to define these values at runtime.  You'll need to handle storing them in EEPROM and fetching and applying at runtime.  You then use buildptr() instead of build() to create the mesh object

You don't need to use the builder.  You can directly call ESP8266MQTTMesh() though there are a lot of parameters, so it is generally a hassle to do that.

closed 2020-05-11 by kazimyuksel