PhracturedBlue/ESP8266MQTTMesh #79: Connect to wifi problem on ESP32

Tim97Egl, 2020-05-26:
I have a similar problem like some other users mentioned. 
Using the library on an ESP32 the programm stops at:
[connect] Connecting to SSID: 

I havent found any solutions to this problem in previous issues. Has anyone used the lib on an ESP32?

simone1999, 2020-07-07:
Hi, have you tried out my fork of the Project?

Tim97Egl, 2020-08-27:
Yes, but it didnt work either. Can you explain, what you have done differently?

eporocrail, 2021-05-01:
In ESP822MQTTMesh.h three lines have to be uncommented and one has to be adapted:

#ifdef ESP32
  #include <AsyncTCP.h>
  //#include <ESP32Ticker.h>
  #include <Ticker.h>
  #define USE_WIFI_ONEVENT
  #include "WiFiCompat.h"
#else
  #include <ESP8266WiFi.h>
  #include <ESPAsyncTCP.h>
  #include <Ticker.h>
#endif

In ESP8266.cpp the following changes have to be made:

//os_sprintf(id, "ChipID:%06X FirmwareID:%04X v%s IP:%s %s", _chipID, firmware_id, firmware_ver, WiFi.localIP().toString().c_str(), meshConnect ? "mesh" : "");

    sprintf(id, "ChipID:%06X FirmwareID:%04X v%s IP:%s %s", _chipID, firmware_id, firmware_ver, WiFi.localIP().toString().c_str(), meshConnect ? "mesh" : "");

I remember that another error message occurred but I can not find the spot in the source code.
I just commented the lines out and it is working fine.

alexmogavero, 2024-01-06:
Please have a try also of my fork, see also my pull request #84